gbionics/idyntree #630: Build failure on macOS using YARP

lrapetti, 2020-01-15:
I am experiencing a compilation failure when compiling i`DynTreePlotter` on macOS both with YARP `master` and `devel` branches:
```
In file included from /Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/src/tools/plotter/src/main.cpp:2:
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/src/tools/plotter/include/ChartsManager.h:5:10: fatal error: 
      'ChartsService.h' file not found
#include <ChartsService.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
```

The problem similar to https://github.com/robotology/idyntree/issues/624, and is fixed by reverting the changes that were fixing that problem (https://github.com/robotology/idyntree/pull/625). So with:
```
#include "thrifts/ChartsService.h"
#include "thrifts/RealTimeStreamData.h"
```
the problem is not happening

lrapetti, 2020-01-15:
> So, by adding `thrifts/` in:
> 
> ```
> #include "thrifts/ChartsService.h"
> #include "thrifts/RealTimeStreamData.h"
> ```
> 
> the problem is not happening and I am able to compile it.

I opened a PR at https://github.com/robotology/idyntree/pull/631, we can see if there is any compilation failure in CI.

traversaro, 2020-01-15:
Which version of YARP are you using? Can you try with a clean build?

lrapetti, 2020-01-16:
> Which version (commit) of YARP are you using? 
```
YARP version 3.3.100+20200114.16+gitd62a3903a
```
I am on the current HEAD. May be that the problem is due to leftovers of previous build?

traversaro, 2020-01-16:
> 
> 
> > Which version (commit) of YARP are you using?
> 
> ```
> YARP version 3.3.100+20200114.16+gitd62a3903a
> ```
> 
> I am on the current HEAD. May be that the problem is due to leftovers of previous build?

This is possible, that is why I suggested to try with a clean build. @pattacini also experienced similar problems, that he solved by enabling the `ALLOW_IDL_GENERATION` CMake option. This should not be necessary, and is probably due to some problems when migrating from `yarp_idl_to_dir` macro without parameters names to the one with parameter names. However, I cannot think it is useful to add more code in iDynTree's build system to handle this if everything works fine with a clean build.

lrapetti, 2020-01-16:
With a clean build of `iDynTree` it actually worked fine, thanks I think we can close then since there is no plan to change iDynTree build system.

closed 2020-01-16 by lrapetti