gbionics/idyntree #632: Generate bindings for inverse-kinematics class

lrapetti, 2020-01-16:
Today we tried to generate the bindings also for the `inverse-kinematics` class. 
Following the [guidelines](https://github.com/robotology/idyntree/blob/master/doc/generating-idyntree-matlab-bindings.md), we modified the `iDynTree.i` file by adding:
```
#include "iDynTree/InverseKinematics.h"
```
and
```
%include "iDynTree/InverseKinematics.h"
```

But the generation of the binding failed with the following error:
```
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:84083:5: warning: delete called on 'iDynTree::DiscreteExtendedKalmanFilterHelper' that is abstract but has non-virtual
      destructor [-Wdelete-non-virtual-dtor]
    delete arg1;
    ^
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:97006:3: error: unknown type name 'Polygon2D'; did you mean 'iDynTree::Polygon2D'?
  Polygon2D *arg2 = 0 ;
  ^~~~~~~~~
  iDynTree::Polygon2D
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/src/inverse-kinematics/include/iDynTree/ConvexHullHelpers.h:85:11: note: 'iDynTree::Polygon2D' declared here
    class Polygon2D
          ^
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:97029:28: error: unknown type name 'Polygon2D'; did you mean 'iDynTree::Polygon2D'?
  arg2 = reinterpret_cast< Polygon2D * >(argp2);
                           ^~~~~~~~~
                           iDynTree::Polygon2D
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/src/inverse-kinematics/include/iDynTree/ConvexHullHelpers.h:85:11: note: 'iDynTree::Polygon2D' declared here
    class Polygon2D
          ^
1 warning and 2 errors generated.
```
the error seems related to [`Polygon2D` class](https://github.com/robotology/idyntree/blob/master/src/inverse-kinematics/include/iDynTree/ConvexHullHelpers.h#L85) definition.

---

@traversaro answered
> I remember having similar problems in the past, that are due to the C++ parser used by SWIG to generate the code. I remember fixing them by adding `iDynTree::` to some declarations. However, to be more constructive, I suggest you to add:
> 
> ```
> #include "iDynTree/ConvexHullHelpers.h"
> ```
> 
> and
> 
> ```
> %include "iDynTree/ConvexHullHelpers.h"
> ```
> 
> to `iDynTree.i` as well, as this header is included by `InverseKinematics.h` and so ensuring that it is wrapper correctly could help.

lrapetti, 2020-01-16:
> I suggest you to add:
> 
> ```
> #include "iDynTree/ConvexHullHelpers.h"
> ```
> 
> and
> 
> ```
> %include "iDynTree/ConvexHullHelpers.h"
> ```
> 
> to `iDynTree.i` as well, as this header is included by `InverseKinematics.h` and so ensuring that it is wrapper correctly could help.

Adding `ConvexHullHelpers.h` I no longer experience the error and I was able to generate the bindings.

lrapetti, 2020-01-16:
PR opened at https://github.com/robotology/idyntree/pull/633

traversaro, 2020-01-16:
There was also issue https://github.com/robotology/idyntree/issues/498 that tracked the same problem.

lrapetti, 2020-01-27:
> PR opened at #633

merged.

closed 2020-01-27 by lrapetti