gbionics/idyntree #634: add vendored libxml2

ahoarau, 2020-01-16:
Now that we have a nice CMakeLists available at https://github.com/roboticslibrary/libxml2
What about adding libxml2 in the "extern" dir and build a private static lib ?

traversaro, 2020-01-18:
I was thinking of adding an option of grabbing Eigen3 and LibXml2 via  FetchContent, would it be ok also that?

ahoarau, 2020-01-18:
My dreams would come true !

ahoarau, 2020-01-27:
iDyntree self contained for 1.2 ?

traversaro, 2020-01-27:
> iDyntree self contained for 1.2 ?

Let's see, it is not difficult but I don't know when we will be working on it.

ahoarau, 2020-01-27:
If neither eigen nor libxml2 appears in the public header, having it build by itself would be a great addition. Easier to install on embedded systems

closed 2023-09-10 by ahoarau