gbionics/idyntree #635: Switch to use Python 3 by default

traversaro, 2020-01-17:
Python 2 is officially out of support, so we should switch to support Python 3 by default, for exampling by `IDYNTREE_USES_PYTHON_VERSION` to 3 by default, see https://github.com/robotology/gym-ignition/issues/120 .

traversaro, 2020-01-17:
cc  @paolo-viceconte

diegoferigo, 2020-09-10:
I think https://github.com/robotology/idyntree/pull/726 was enough to deprecate the usage of Python 2. Handling `IDYNTREE_USES_PYTHON_VERSION` is still missing, should we just delete this option?

traversaro, 2020-09-11:
Unless you have some other preference I would delete it.

diegoferigo, 2020-09-11:
I forgot that this was done already in #726 as also commented in https://github.com/robotology/idyntree/pull/726/commits/ff8b936069f47978306a0090d06f8dc87bfa4d51#r479650740. I'm closing this issue.

closed 2020-09-11 by diegoferigo