gbionics/idyntree #648: Unable to visualize the equation in the doxygen documentation

GiulioRomualdi, 2020-02-13:
When I load the Doxygen documentation the equations are not rendered
[image]

The problem is related to mathjax` indeed the chrome console displays the following error 
```
Mixed Content: The page at 'https://robotology.github.io/idyntree/devel/classiDynTree_1_1optimalcontrol_1_1integrators_1_1RK4.html' was loaded over HTTPS, but requested an insecure script 'http://cdn.mathjax.org/mathjax/latest/MathJax.js'. This request has been blocked; the content must be served over HTTPS.
```

The problem can be easily fixed updating the `Doxygen.in` file as
```patch
 doc/Doxyfile.in | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/doc/Doxyfile.in b/doc/Doxyfile.in
index 1b2d527f5..4c126db5e 100644
--- a/doc/Doxyfile.in
+++ b/doc/Doxyfile.in
@@ -1188,7 +1188,7 @@ USE_MATHJAX            = YES
 # However, it is strongly recommended to install a local
 # copy of MathJax from http://www.mathjax.org before deployment.
 
-MATHJAX_RELPATH        = http://cdn.mathjax.org/mathjax/latest
+MATHJAX_RELPATH        = https://cdn.mathjax.org/mathjax/latest
 
 # The MATHJAX_EXTENSIONS tag can be used to specify one or MathJax extension
 # names that should be enabled during MathJax rendering.
```

GiulioRomualdi, 2020-02-13:
I can propose the PR

traversaro, 2020-02-13:
Thanks a lot!

PR #649 (2020-02-13): Fix mathjax in doxygen documentation
This PR fixes #648
closed 2020-02-13 by traversaro