gbionics/idyntree #652: [Bindings] baseFrame is not changed by loadReducedModel

lrapetti, 2020-03-13:
I am trying to load a model using `loadReducedModel` and passing as input a  chosen `baseFrame`.
However, when I try to set the robot configuration with `setRobotState`, the computed pose of the chosen `baseFrame` doesn't seem to reflect the one given as input.

An example code can be seen in https://gist.github.com/lrapetti/21f6c28415f1243d1732b6b6f3235f45 (in order to run it change `Config.Model.modelPath` to your path to the iCub model). In particular in that code, what is observed is that:
- the [selected `baseLink` is `l_sole`](https://gist.github.com/lrapetti/21f6c28415f1243d1732b6b6f3235f45#file-gistfile1-txt-L9) and the chosen base link transform is the [identity](https://gist.github.com/lrapetti/21f6c28415f1243d1732b6b6f3235f45#file-gistfile1-txt-L30-L35)
- after setting the robot state for kindyn computation, the `l_sole` transform is different from the identity
- the `root_link` transform instead correspond to the transform passed as base transform `w_H_b_init `

gabrielenava, 2020-03-13:
I tried to run the code with `iCubGazeboV2_5`,  but I got an error:

```
[ERROR] Model :: getLinkIndex : Impossible to find link l_sole in the Model
[ERROR] Model :: computeFullTreeTraversal : requested traversalBase is out of bounds
[loadReducedModel]: loaded model: ./model.urdf, number of joints: 25
```
It seems `iCubGazeboV2_5` does not have `l_sole` link. I took the model inside my `robotology-superbuild/install`folder. @lrapetti do you have the same error?

Note that the model is anyway loaded.

lrapetti, 2020-03-13:
> It seems `iCubGazeboV2_5` does not have `l_sole` link. I took the model inside my `robotology-superbuild/install`folder. @lrapetti do you have the same error?

No, I don't see that error, I see the `l_sole` link on `iCubGazeboV2_5` model. Did you change the path at https://gist.github.com/lrapetti/21f6c28415f1243d1732b6b6f3235f45#file-gistfile1-txt-L10 ?

gabrielenava, 2020-03-13:
That's strange, it seems the link is in my model:

[image]

but I get the error anyways. I changed the name for the model to `model_test` to be sure I am loading the correct one, but it seems so.

Furthermore, if I try to repeat your test with a different link (e.g. the `chest`), it seems the chosen base frame is loaded correctly:

```
Config.Model.baseLinkName = 'chest';

iDynTreeWrappers.getWorldTransform(KinDynModel,'chest')

ans =

     1     0     0     0
     0     1     0     0
     0     0     1     0
     0     0     0     1

>> iDynTreeWrappers.getWorldTransform(KinDynModel,'base_link')

ans =

    0.0000   -1.0000         0    0.0000
    0.0523    0.0000    0.9986   -0.0820
   -0.9986   -0.0000    0.0523   -0.0055
         0         0         0    1.0000
```

If I repeat the test with the `l_sole` as chosen base frame:

```
Config.Model.baseLinkName = 'l_sole';
```

I get this errror and then the frame is not set despite the model is loaded, and what happens is that the transformation is not correct:

```
[ERROR] Model :: getLinkIndex : Impossible to find link l_sole in the Model
[ERROR] Model :: computeFullTreeTraversal : requested traversalBase is out of bounds
[loadReducedModel]: loaded model: ./model_test.urdf, number of joints: 25
>> iDynTreeWrappers.getWorldTransform(KinDynModel,'l_sole')

ans =

   -0.9979    0.0126    0.0634   -0.0522
   -0.0127   -0.9999   -0.0009   -0.1078
    0.0633   -0.0017    0.9980   -0.6126
         0         0         0    1.0000

```

I have to say I never chose `l_sole` as base frame before. But I am interested in understanding this error better.

cc @traversaro

traversaro, 2020-03-13:
Unfortunately `l_sole` is not a link, but rather an additional frame of the link `l_foot`. This happens because during the URDF parsing, the "fake links" of URDF are converted to additional frames in iDynTree, see https://github.com/robotology/idyntree/blob/d5bbceab9e8ae52943b31199f6e024d12ff52159/src/model/include/iDynTree/Model/ModelTransformers.h#L50, https://github.com/robotology/idyntree/blob/d5bbceab9e8ae52943b31199f6e024d12ff52159/src/model_io/urdf/src/URDFDocument.cpp#L123 and https://robotology.github.io/idyntree/master/group__iDynTreeModelIO.html#ga7fc5d09c1c301c6d1d92ad4dc4a775df .

Supporting the possibility of setting additional frames as base frame of the dynamics is doable, see https://github.com/robotology/idyntree/issues/422 . If you are interested in working on this, please let me know and I can provide you additional details.

lrapetti, 2020-03-13:
> No, I don't see that error, I see the `l_sole` link on `iCubGazeboV2_5` model. Did you change the path at https://gist.github.com/lrapetti/21f6c28415f1243d1732b6b6f3235f45#file-gistfile1-txt-L10 ?

my bad @gabrielenava, the error was there, but I missed the fact it was not causing the application to fail. Thanks.

> Supporting the possibility of setting additional frames as base frame of the dynamics is doable, see #422 . If you are interested in working on this, please let me know and I can provide you additional details.

Thanks @traversaro for the explanation. If it can be helpful I might check how to handle this, so please, if you can, send me more details (we might continue the discussion in #422)