gbionics/idyntree #654: [Visualizer] Cannot visualize models in two separate Visualizer objects

prashanthr05, 2020-03-20:
Here, I tried to add the same model to two instantiated objects of the iDynTree visualizer. I set camera targets and positions.
And when I call draw() on both the objects, only one of the object gets updated.

[image]

Am I missing something here? Or is it not possible to have multiple instances of the Visualizer due to some underlying shared SceneManager ?

DanielePucci, 2020-03-20:
CC @traversaro @gabrielenava @S-Dafarra

traversaro, 2020-03-20:
As discussed on Teams, in theory the two Visualizers  should be  completely independent, but it is possible there are some bugs in iDynTree or in Irrlicht, I personally do not remember if I have tried to do something like that.