gbionics/idyntree #656: [Bindings] Can not access the visual objects in Matlab

fjandrad, 2020-03-23:
In matlab when executing the following code :+1: 

```
KinDynModel = iDynTreeWrappers.loadReducedModel(cJoints,'torso_link',path,file,false);
   
model=KinDynModel.kinDynComp.model 

visual= model.visualSolidShapes % http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1Model.html

linkSolids=visual.linkSolidShapes % http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1ModelSolidShapes.html
```
 
linkSolids would be std::vector< std::vector<SolidShape *> > iDynTree::ModelSolidShapes::linkSolidShapes , but we seem unable to access the individual SolidShape variables. 

When attempting to get the size of the vector we get:
```
linkSolids.size()
Unrecognized method, property, or field 'size' for class 'SwigRef'. 
   Error in SwigRef/subsref (line 33)
        [varargout{1:nargout}] = builtin('subsref',self,s);
```

It seems the bindings are not generating the code for those kinds of variables. We should update the bindings to achieve this. 
An example ( courtesy of @traversaro )  of the a similar case can be seen in https://github.com/robotology/idyntree/blob/master/bindings/iDynTree.i#L9 what we do for std::vector<std::string>

fjandrad, 2020-03-25:
It seems that we are missing a template for `std::vector< std::vector<SolidShape *>>`. This is a vector of vectors of pointers of Solidshapes.

The solution is then to add something like :

 `%template(StringVector) std::vector<std::string>;`

Probably something like 

 `%template(linksSolidshapesVector) `std::vector< std::vector<SolidShape *>>;`

traversaro, 2020-03-25:
I guess you need to esplicitly instantiate and give a name to both `std::vector< std::vector<SolidShape *>>` and `std::vector<SolidShape *>`.

fjandrad, 2020-03-25:
Should I define one before the other? 
Something like 
1. `%template(SolidshapesVector) std::vector<SolidShape *>`
1. `%template(linksSolidshapesVector) std::vector< std::vector<SolidShape *>>`

or 

1. `%template(SolidshapesVector) std::vector<SolidShape *>`
1. `%template(linksSolidshapesVector) std::vector< SolidshapesVector>`

Does the order matters?

traversaro, 2020-03-25:
> 
> 
> Should I define one before the other?
> Something like
> 
>     1. `%template(SolidshapesVector) std::vector<SolidShape *>`
> 
>     2. `%template(linksSolidshapesVector) std::vector< std::vector<SolidShape *>>`
> 
> 
> or
> 
>     1. `%template(SolidshapesVector) std::vector<SolidShape *>`
> 
>     2. `%template(linksSolidshapesVector) std::vector< SolidshapesVector>`
> 
> 
> Does the order matters?

I do not know.

fjandrad, 2020-03-25:
I'll give it a try to the second case first and see if it works. If not I'll try the first.

fjandrad, 2020-03-25:
The addition to the binding is failing for debugging purposes I tried to generate bindings alone and it is working. 

Then I tried to both and it failed.

 So I tried the following cases:
## Errors in the autogeneration step

### `%template(SolidshapesVector) std::vector<SolidShape *>`

The error I get in this case is in this file:
[cmake_output.txt](https://github.com/robotology/idyntree/files/4382428/cmake_output.txt)

The last output of cmake is :
```
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:1818:76: error: ‘type_name’ is not a member of ‘swig::traits<int>’
     return traits<typename noconst_traits<Type >::noconst_type >::type_name();
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
make[2]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/autogenerated/iDynTreeMATLAB_wrap.cxx.o] Error 1
make[1]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/all] Error 2
make: *** [all] Error 2

```

###  `%template(SolidshapesVector) std::vector<SolidShape>`
Same error if I take out the * 

@traversaro does this mean I need to change the iDyntree source code to make it const?

fjandrad, 2020-03-25:
Trying alone the SolishapesVector compiles with no error but when trying to include the second vector of vectors I get the following errors.

## Trying without SolishapesVector
### Trying `%template(linksSolidshapesVector) std::vector< std::vector<SolidShape *>>` alone

The output file is
[cmake_output.txt](https://github.com/robotology/idyntree/files/4382591/cmake_output.txt)

The error is : 
```
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:1816:76: error: ‘type_name’ is not a member of ‘swig::traits<iDynTree::SolidShape>’
     return traits<typename noconst_traits<Type >::noconst_type >::type_name();
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
bindings/matlab/CMakeFiles/iDynTreeMEX.dir/build.make:70: recipe for target 'bindings/matlab/CMakeFiles/iDynTreeMEX.dir/autogenerated/iDynTreeMATLAB_wrap.cxx.o' failed
make[2]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/autogenerated/iDynTreeMATLAB_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:1352: recipe for target 'bindings/matlab/CMakeFiles/iDynTreeMEX.dir/all' failed
make[1]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

## Trying with SolishapesVector 
### Trying  `%template(linksSolidshapesVector) std::vector< SolidshapesVector>`
The output file is
[cmake_output.txt](https://github.com/robotology/idyntree/files/4382610/cmake_output.txt)

Some of the errors are:

```
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:63565:16: error: ‘SolidshapesVector’ was not declared in this scope
   std::vector< SolidshapesVector > *arg1 = (std::vector< SolidshapesVector > *) 0 ;
                ^~~~~~~~~~~~~~~~~
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:63565:16: note: suggested alternative: ‘_wrap_new_SolidshapesVector’
   std::vector< SolidshapesVector > *arg1 = (std::vector< SolidshapesVector > *) 0 ;
                ^~~~~~~~~~~~~~~~~
                **_wrap_new_SolidshapesVector**
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:63565:34: error: template argument 1 is invalid
   std::vector< SolidshapesVector > *arg1 = (std::vector< SolidshapesVector > *) 0 ;
```
It makes a suggestion to use `_wrap_new_SolidshapesVector` instead of `SolidshapesVector`
The last error is : 
```
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:1820:76: error: ‘type_name’ is not a member of ‘swig::traits<int>’
     return traits<typename noconst_traits<Type >::noconst_type >::type_name();
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
bindings/matlab/CMakeFiles/iDynTreeMEX.dir/build.make:70: recipe for target 'bindings/matlab/CMakeFiles/iDynTreeMEX.dir/autogenerated/iDynTreeMATLAB_wrap.cxx.o' failed
make[2]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/autogenerated/iDynTreeMATLAB_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:1352: recipe for target 'bindings/matlab/CMakeFiles/iDynTreeMEX.dir/all' failed
make[1]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

### Trying the suggestion   `%template(linksSolidshapesVector) std::vector< _wrap_new_SolidshapesVector>`
The output file is:
[cmake_output.txt](https://github.com/robotology/idyntree/files/4382649/cmake_output.txt)

The errors are:
```
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:63579:69: note:   expected a type, got ‘_wrap_new_SolidshapesVector’
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:63579:69: error: template argument 2 is invalid
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:63579:71: error: expected ‘>’ before ‘*’ token
  arg1 = reinterpret_cast< std::vector< _wrap_new_SolidshapesVector > * >(argp1);
             
```        

The last error:
```
/home/iiticublap138/shared-folder/dev/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:1820:76: error: ‘type_name’ is not a member of ‘swig::traits<int>’
     return traits<typename noconst_traits<Type >::noconst_type >::type_name();
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
bindings/matlab/CMakeFiles/iDynTreeMEX.dir/build.make:70: recipe for target 'bindings/matlab/CMakeFiles/iDynTreeMEX.dir/autogenerated/iDynTreeMATLAB_wrap.cxx.o' failed
make[2]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/autogenerated/iDynTreeMATLAB_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:1352: recipe for target 'bindings/matlab/CMakeFiles/iDynTreeMEX.dir/all' failed
make[1]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```
###  Trying `%template(linksSolidshapesVector) std::vector< std::vector<SolidShape *>>`
**Compiles without error. Now we can move on to test.**

fjandrad, 2020-03-26:
For testing we use the following code:

```
%
icubModelPath=[ getenv('ROBOTOLOGY_SUPERBUILD_SOURCE_DIR') '/build-linux/install/share/iCub/robots/iCubGazeboV2_5/'];
fileName='model.urdf';
jointOrder={
    'r_hip_pitch';
    'r_hip_roll';
    'r_hip_yaw';
    'r_knee';
    'r_ankle_pitch';
    'r_ankle_roll';
    'l_hip_pitch';
    'l_hip_roll';
    'l_hip_yaw';
    'l_knee';
    'l_ankle_pitch';
    'l_ankle_roll';
    'torso_pitch';
    'torso_roll';
    'torso_yaw';
    'r_shoulder_pitch';
    'r_shoulder_roll';
    'r_shoulder_yaw';
    'r_elbow';
    'r_wrist_prosup';
    'r_wrist_pitch';
    'r_wrist_yaw';
    'l_shoulder_pitch';
    'l_shoulder_roll';
    'l_shoulder_yaw';
    'l_elbow';
    'l_wrist_prosup';
    'l_wrist_pitch';
    'l_wrist_yaw';
    'neck_pitch';
    'neck_roll';
    'neck_yaw'
    };
%
KinDynModel = iDynTreeWrappers.loadReducedModel(jointOrder,'root_link',icubModelPath,fileName,false);
model=KinDynModel.kinDynComp.model;
visual=model.visualSolidShapes;
linkSolidShapesV=visual.linkSolidShapes;
% get number of links
numberOfLinks=linkSolidShapesV.size;
% get pointer to beggining of vector
iterator=linkSolidShapesV.begin;

% iterate getting the name of the mesh and the name of the link
for links=1:numberOfLinks
    solidarray=iterator.next
    if solidarray{1}.isExternalMesh
        meshName=split(solidarray{1}.asExternalMesh.filename,':')
        meshFiles{links}=meshName{2};
        linkNames{links}=model.getLinkName(links-1)        
    end
end
```

The output is 

[image]

Therefore we can consider it a **success**.

fjandrad, 2020-03-26:
Changes in iDyntree.i are in 0bddc1d6bf9b2bd66990a68dd42eb581b0f53695
Updated bindings accordingly are in 1331cf3f99139b5c72c33ccd0388b572809f1f11

fjandrad, 2020-04-01:
PR merged #658

closed 2020-04-01 by fjandrad