gbionics/idyntree #657: Fix path in the read me to matlab binding modifications

fjandrad, 2020-03-23:
In the faq there is a link to https://github.com/robotology/idyntree#matlab-bindings-modifications , but it seems that path is not working as intended since it just sends you to the idyntree main page. 

The link in the faq should be changed to https://github.com/robotology/idyntree#matlaboctave-bindings-modifications

fjandrad, 2020-03-23:
For really small modifications like this, how do you suggest to proceed @traversaro pull request? simple commit?

traversaro, 2020-03-23:
> 
> 
> For really small modifications like this, how do you suggest to proceed @traversaro pull request? simple commit?

Pull request via the Web UI. It is really simple and it saves a lot of time.

fjandrad, 2020-03-23:
Done here a177b6d7908a87b4b28caf8391efa5b453282e8a, I'm afraid I did it in master by mistake... @traversaro  should I revert?

traversaro, 2020-03-23:
> 
> 
> Done here [a177b6d](https://github.com/robotology/idyntree/commit/a177b6d7908a87b4b28caf8391efa5b453282e8a), I'm afraid I did it in master by mistake... @traversaro should I revert?

No problem, but for the future please pay attention before commiting to master, thanks!

closed 2020-03-23 by traversaro