gbionics/idyntree #674: Document semantics difference between Pinocchio quantities and iDynTree KinDynComputations quantities

traversaro, 2020-04-20:
Thi

traversaro, 2020-04-20:
cc @GiulioRomualdi

traversaro, 2020-04-20:
Related issues: 
* https://github.com/stack-of-tasks/pinocchio/issues/1171
* https://github.com/stack-of-tasks/pinocchio/issues/1140

GiulioRomualdi, 2020-07-07:
Before I forgot I want to drop here this table of conversion:

|           `iDynTree`          |     `Pinocchio`     |
|:-----------------------------:|:-------------------:|
|      MIXED_REPRESENTATION     | LOCAL_WORLD_ALIGNED |
| INERTIAL_FIXED_REPRESENTATION |        WORLD        |
|   BODY_FIXED_REPRESENTATION   |        LOCAL        |

DanielePucci, 2020-07-11:
CC @robotology/iit-dynamic-interaction-control

traversaro, 2022-03-26:
> Before I forgot I want to drop here this table of conversion:

Updated version to be self-contained. Assumptions:
* we have two frame a and e, and the transform between this two frames is represented by ᵃRₑ ∈ SO(3)  ᵃoₑ ∈ ℝ³ 
*  (.)ᵛ represents the mapping between the skew-symmetrix matrix and the related vector in ℝ³, i.e. such that if u,v ∈ ℝ³, (u)ᵛv = u × v  
*  the dot represent the usual mathematical analysis dot operator (time derivative) and not the Featherstone one then the different velocities are:

|           `iDynTree`          |     `Pinocchio`     |   Math      | 
|:-----------------------------:|:-------------------:|:-----------:|
|      MIXED_REPRESENTATION     | LOCAL_WORLD_ALIGNED | Trans: ᵃȯₑ . Ang:  ( ᵃṘₑ   ᵃRₑᵀ )ᵛ  .  |
| INERTIAL_FIXED_REPRESENTATION |        WORLD        | Trans: ᵃȯₑ - (ᵃṘₑ   ᵃRₑᵀ) ᵃoₑ . Ang:  ( ᵃṘₑ   ᵃRₑᵀ )ᵛ  . |
|   BODY_FIXED_REPRESENTATION   |        LOCAL        | Trans: ᵃRₑᵀ ᵃȯₑ . Ang:  ( ᵃRₑᵀ  ᵃṘₑ )ᵛ  . |

Note: pinocchio uses angular-translational, iDynTree translational-angular .
Sources and derivation for the equations:
* `MIXED_REPRESENTATION` Eq. 35 in https://research.tue.nl/en/publications/multibody-dynamics-notation-version-2 
* `INERTIAL_FIXED_REPRESENTATION` Eq. 23 in https://research.tue.nl/en/publications/multibody-dynamics-notation-version-2 
* `BODY_FIXED_REPRESENTATION` Eq. 18 in https://research.tue.nl/en/publications/multibody-dynamics-notation-version-2 

Related Pinocchio issues:
*