gbionics/idyntree #675: Build devel branch fails on MacOS when USES_MATLAB is ON

lrapetti, 2020-04-21:
At first it failed with the same error experienced in https://github.com/robotology/idyntree/issues/630, fixed by recompiling a clean build.

However, after turning `IDYNTREE_USES_MATLAB` `ON`, it was failing with the following message:
```
[ 99%] Building CXX object bindings/matlab/CMakeFiles/iDynTreeMEX.dir/autogenerated/iDynTreeMATLAB_wrap.cxx.o
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:4549:30: error: non-constant-expression cannot be narrowed from type 'int' to 'mwSize' (aka 'unsigned long') in initializer list [-Wc++11-narrowing]
        const mwSize dims[]={nMatrices,nRows,nCols};
                             ^~~~~~~~~
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:4549:30: note: insert an explicit cast to silence this issue
        const mwSize dims[]={nMatrices,nRows,nCols};
                             ^~~~~~~~~
                             static_cast<mwSize>( )
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:4549:40: error: non-constant-expression cannot be narrowed from type 'int' to 'mwSize' (aka 'unsigned long') in initializer list [-Wc++11-narrowing]
        const mwSize dims[]={nMatrices,nRows,nCols};
                                       ^~~~~
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:4549:40: note: insert an explicit cast to silence this issue
        const mwSize dims[]={nMatrices,nRows,nCols};
                                       ^~~~~
                                       static_cast<mwSize>( )
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:4549:46: error: non-constant-expression cannot be narrowed from type 'int' to 'mwSize' (aka 'unsigned long') in initializer list [-Wc++11-narrowing]
        const mwSize dims[]={nMatrices,nRows,nCols};
                                             ^~~~~
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:4549:46: note: insert an explicit cast to silence this issue
        const mwSize dims[]={nMatrices,nRows,nCols};
                                             ^~~~~
                                             static_cast<mwSize>( )
/Users/lorenzorapetti/Software/robotology-superbuild/robotology/iDynTree/bindings/matlab/autogenerated/iDynTreeMATLAB_wrap.cxx:88975:5: warning: delete called on 'iDynTree::DiscreteExtendedKalmanFilterHelper' that is abstract but has non-virtual destructor [-Wdelete-non-virtual-dtor]
    delete arg1;
    ^
1 warning and 3 errors generated.
make[2]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/autogenerated/iDynTreeMATLAB_wrap.cxx.o] Error 1
make[1]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/all] Error 2
make: *** [all] Error 2

```

lrapetti, 2020-04-21:
CC @claudia-lat

claudia-lat, 2020-04-21:
I just tried to build it and it failed in the very same way.

lrapetti, 2020-04-21:
The problem was introduced in https://github.com/robotology/idyntree/pull/668

traversaro, 2020-04-21:
So, there are two main problems here: 
* The actual bug in https://github.com/robotology/idyntree/blob/devel/bindings/matlab/matlab_mat4x4vec.i#L10 , where the arguments `nMatrices`, `nRows` and `nCols` should be declared as `mwSize`, and the bindings regenerated. @fjandrad as I  guess you already have the bindings machinery ready could you provide a PR for that? Otherwise if someone else wants to tackle it,  you need to follow the instructions in  https://github.com/robotology/idyntree#matlaboctave-bindings-modifications to get the proper SWIG for matlab generation. 
* The  fact that this was not catched by the CI. Probably to fix this we just need to add `brew install octave` and pass `-DIDYNTREE_USES_OCTAVE:BOOL=ON` to the macOS build.

traversaro, 2020-04-21:
Let me  know if no one is able to handle the fix discussed in the previous comment, then I can handle it. @fjandrad @lrapetti @claudia-lat

lrapetti, 2020-04-21:
> Let me know if no one is able to handle the fix discussed in the previous comment, then I can handle it. @fjandrad @lrapetti @claudia-lat

I can take care of regenerating the bindings

traversaro, 2020-04-21:
Great, thanks! Remember to separate the commit that modifies the `.i` file and the one that add the generated bindings (as documented in https://github.com/robotology/idyntree/blob/master/doc/dev/faqs.md#how-to-add-wrap-a-new-class-or-function-with-swig).  Instead you don't need to update the Changelog, as this fix is fixing a bug that was never included in a released version of iDynTree.

lrapetti, 2020-04-21:
PR opened at https://github.com/robotology/idyntree/pull/676

claudia-lat, 2020-04-21:
The PR has been merged. Now it works! Thanks @lrapetti @traversaro

lrapetti, 2020-04-21:
> * The  fact that this was not catched by the CI. Probably to fix this we just need to add `brew install octave` and pass `-DIDYNTREE_USES_OCTAVE:BOOL=ON` to the macOS build.

I have opened a PR also for adding the test in https://github.com/robotology/idyntree/pull/677

traversaro, 2020-04-21:
> 
> 
> > * The  fact that this was not catched by the CI. Probably to fix this we just need to add `brew install octave` and pass `-DIDYNTREE_USES_OCTAVE:BOOL=ON` to the macOS build.
> 
> I have opened a PR also for adding the test in #677

Thanks!

lrapetti, 2020-04-22:
Both #676 and #677 have been merged. We can close this issue.

closed 2020-04-22 by lrapetti

traversaro, 2020-04-22:
Thanks @lrapetti !

fjandrad, 2020-04-23:
Sorry for the late response I was without a computer for two days. I'm glad is fixed. Why does it only affect MACOS?

traversaro, 2020-04-23:
I guess it is something related to the default settings of Clang (used in macOS) that are different from the one in GCC (used in Linux).