gbionics/idyntree #679: Fix solidshape grouping error

fjandrad, 2020-04-22:
When using certain urdfs the solidshapes are all included in the first link and then some of the next spaces in the vector appear empty. This renders the visualization to have a mismatch between the number of links in the model and the number of transforms that are generated for the visualization.

fjandrad, 2020-04-22:
cc @claudia-lat

fjandrad, 2020-04-22:
The reason seems to be a faulty assumption. When updating with the string idyntree vector we were asking for the number of links transforms, but in fact, the list of names contains only the links that have visuals. So we exceed the index by setting the end condition of the for to the number of links in the model.

Loading the full model gives the following links have visuals:
| Geometry/Mesh | Link Name       |
|---------------|-----------------|
| 'box'         | 'Pelvis'        |
| 'cylinder'    | 'LeftUpperLeg'  |
| 'cylinder'    | 'LeftLowerLeg'  |
| 'box'         | 'LeftFoot'      |
| 'box'         | 'LeftToe'       |
| 'box'         | 'L5'            |
| 'box'         | 'L3'            |
| 'box'         | 'T12'           |
| 'box'         | 'T8'            |
| 'cylinder'    | 'RightShoulder' |
| 'cylinder'    | 'LeftShoulder'  |
| 'cylinder'    | 'LeftUpperArm'  |
| 'cylinder'    | 'LeftForeArm'   |
| 'box'         | 'LeftHand'      |
| 'cylinder'    | 'Neck'          |
| 'sphere'      | 'Head'          |
| 'cylinder'    | 'RightUpperArm' |
| 'cylinder'    | 'RightForeArm'  |
| 'box'         | 'RightHand'     |
| 'cylinder'    | 'RightUpperLeg' |
| 'cylinder'    | 'RightLowerLeg' |
| 'box'         | 'RightFoot'     |
| 'box'         | 'RightToe'      |

But the full list of links seen by iDyntree is:
 Links: 
         [0] Pelvis
         [1] RightUpperLeg_f1
         [2] L5_f1
         [3] LeftUpperLeg_f1
         [4] LeftUpperLeg_f2
         [5] LeftUpperLeg
         [6] LeftLowerLeg_f1
         [7] LeftLowerLeg
         [8] LeftFoot_f1
         [9] LeftFoot_f2
         [10] LeftFoot
         [11] LeftToe
         [12] L5
         [13] L3_f1
         [14] L3
         [15] T12_f1
         [16] T12
         [17] T8_f1
         [18] T8_f2
         [19] T8
         [20] RightShoulder
         [21] Neck_f1
         [22] LeftShoulder
         [23] LeftUpperArm_f1
         [24] LeftUpperArm_f2
         [25] LeftUpperArm
         [26] LeftForeArm_f1
         [27] LeftForeArm
         [28] LeftHand_f1
         [29] LeftHand
         [30] Neck_f2
         [31] Neck
         [32] Head_f1
         [33] Head
         [34] RightUpperArm_f1
         [35] RightUpperArm_f2
         [36] RightUpperArm
         [37] RightForeArm_f1
         [38] RightForeArm
         [39] RightHand_f1
         [40] RightHand
         [41] RightUpperLeg_f2
         [42] RightUpperLeg
         [43] RightLowerLeg_f1
         [44] RightLowerLeg
         [45] RightFoot_f1
         [46] RightFoot_f2
         [47] RightFoot
         [48] RightToe

If we take a look at the urdf we see for example:
```xml
<link name="L5_f1">
        <inertial>
            <mass value="0"/>
            <origin xyz="0 0 0" rpy="0 0 0" />
            <inertia ixx="0" iyy="0" izz="0" ixy="0" ixz="0" iyz="0"/>
        </inertial>
    </link>
	<link name="L5">
        <inertial>
            <mass value="6.3444"/>
			<!--COM origin wrt jL5S1-->
            <origin xyz="0           0    0.047552" rpy="0 0 0" />
            <inertia ixx="0.018868" iyy="0.014872" izz="0.024177" ixy="0" ixz="0" iyz="0"/>
        </inertial>

        <visual>
			<!--box origin origin wrt jL5S1-->
            <origin xyz="0           0    0.047552" rpy="0 0 0" />
            <geometry>
                <box size="0.13815     0.16323    0.095104"/>
            </geometry>
        </visual>

        <collision>
			<!--box origin origin wrt jL5S1-->
            <origin xyz="0           0    0.047552" rpy="0 0 0" />
            <geometry>
                <box size="0.13815     0.16323    0.095104"/>
            </geometry>
        </collision>
    </link>
```

As it can be seen the _f1 or _f2 links in the model have no visuals and therefore the number of links is no thte number of visual objects.

fjandrad, 2020-04-23:
Solved in dde66f5e2d231e8a7fe92c36565747fdebb0d8e4

closed 2020-04-23 by fjandrad

traversaro, 2020-04-23:
Hi @fjandrad , I suggest to close issues when the problem has been fixed  in `devel` or `master` branch, i.e. when the related PR has been merged. Otherwise is a bit misleading to see that an issue, but then download the latest version and find that the issue has not been fixed.

PR #682 (2020-04-24): Add new features to the MATLAB visualizer: added world frame, optional ground and modification of visual aspects using variable inputs
In this PR we allow to receive variable inputs for:

-  custom view point
-  change color of the robot
-  change level of transparency
-  have a shinny surface or dull
-  change style of display: full mesh , wireframe or invisible

It can be done from the variable inputs in the prepareVisualization function or using the modifyLinksVisual after the figure creation.

In addition at the preparation stage it can add

-  add something to represent the floor
-  add a [... cut]