gbionics/idyntree #680: #binding python, error in /iDynTreePYTHON_wrap.cxx

hustfzt2, 2020-04-23:
/home/kingsfu/DynaCalib/idyntree/build/bindings/python/iDynTreePYTHON_wrap.cxx:47388:111: error: macro "SWIG_as_voidptr" passed 2 arguments, but takes just 1
   pyobj = SWIG_NewPointerObj(SWIG_as_voidptr(&iDynTree::Span< double,-1 >::extent), SWIGTYPE_p_index_type,  0 );
                                                                                                               ^
/home/kingsfu/DynaCalib/idyntree/build/bindings/python/iDynTreePYTHON_wrap.cxx: In function ‘PyObject* Swig_var_DynamicSpan_extent_get()’:
/home/kingsfu/DynaCalib/idyntree/build/bindings/python/iDynTreePYTHON_wrap.cxx:47388:30: error: ‘SWIG_as_voidptr’ was not declared in this scope
   pyobj = SWIG_NewPointerObj(SWIG_as_voidptr(&iDynTree::Span< double,-1 >::extent), SWIGTYPE_p_index_type,  0 );
                              ^
/home/kingsfu/DynaCalib/idyntree/build/bindings/python/iDynTreePYTHON_wrap.cxx:1189:89: note: in definition of macro ‘SWIG_NewPointerObj’
 #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)

traversaro, 2020-04-23:
Hi @hustfzt2 , 
  it would be useful if you could provide a bit more of information regarding you setup:
* Which version of iDynTree are you using? With which options? 
* Which operating system are you using ? Which compiler? Which  CMake version? Which SWIG version? 

Thanks.

hustfzt2, 2020-04-23:
Hi, thanks for your reply.
The basic info. are : iDynTree-master, Ubuntu 18.04, pycharm 2019, CMake 3.5.1,  swig 2.0

traversaro, 2020-04-23:
I am  afraid that at least SWIG 3.0 is required (that is the default one that you can install on Ubuntu 18.04 with  `sudo apt install swig`). There is any reason why you are using SWIG 2?

hustfzt2, 2020-04-23:
The SWIG version is now updgraded to 3.0.8-0ubuntu3, however, the issue is still not solved. I think the iDynTreePYTHON_wrap.cxx file may have some syntax errors.

traversaro, 2020-04-23:
> The SWIG version is now updgraded to 3.0.8-0ubuntu3, however, the issue is still not solved. I think the iDynTreePYTHON_wrap.cxx file may have some syntax errors.

The `iDynTreePYTHON_wrap.cxx` file is generated  by SWIG, and at least in our Continuous Integration tests  it is working correctly. As it is  possible that that file has not been regenerated after you upgraded from SWIG  2.0 to SWIG 3.0, can you try again on a clean build?

hustfzt2, 2020-04-23:
the same issue occurs after clean build.

traversaro, 2020-04-23:
As everything is working fine in our testing infrastructure (see https://github.com/robotology/idyntree/runs/592576033) I guess there  is something different on your machine. If you could  try to reproduce the  problem  in a clean Ubuntu 18.04 environment (for example via  Docker or virtual machine) it would be ideal, but also pasting the output of the `env` command and the exact compilation command that fails (via `make VERBOSE=1`) specifically could help understand what is  going different in your enviroment.

hustfzt2, 2020-04-24:
However, my friends and I have tested it on basis of your suggestions, however, it still fails with the same errors.

traversaro, 2020-04-24:
> However, my friends and I have tested it on basis of your suggestions, however, it still fails with the same errors.

You tried in a clean Ubuntu 18.04 enviroment and it still fails? Are you sure it is vanilla Ubuntu and there are not other libraries such as ROS or Conda already installed? In that case, can you provide the details of the configuration and the exact commands that you used? Also the exact compilation command via `make VERBOSE=1` and `env` as mentioned before can be helpful to debug this kind of problems.

traversaro, 2020-04-24:
To get a clean Ubuntu 18.04, you can use for example https://www.gitpod.io/ or similar systems.

traversaro, 2021-01-26:
Hi @hustfzt2 , if you still have problems feel free to re-open this issue or open new issue, thanks!