gbionics/idyntree #684: Understand why iDynTree inverse kinematics crashes in blockfactory

bemilio, 2020-04-28:
I have encountered a problem in the iDynTree inverse kinematics (based on IPOPT) defined [here](https://github.com/robotology/idyntree/blob/master/src/inverse-kinematics/src/InverseKinematicsData.cpp).

I wrapped it in a blockfactory-based simulink block. When calling the [solve](https://github.com/robotology/idyntree/blob/83f306bf521a2e89c594017747bda95fe786e346/src/inverse-kinematics/include/iDynTree/InverseKinematics.h#L766) method, it crashes with final output (note that IPOPT is at maximum verbosity)

```
******************************************************************************
This program contains Ipopt, a library for large-scale nonlinear optimization.
 Ipopt is released as open source code under the Eclipse Public License (EPL).
         For more information visit http://projects.coin-or.org/Ipopt
******************************************************************************

This is Ipopt version 3.11, running with linear solver mumps.
NOTE: Other linear solvers might be more efficient (see Ipopt documentation).

Number of nonzeros in equality constraint Jacobian...:      126
Number of nonzeros in inequality constraint Jacobian.:       93
Number of nonzeros in Lagrangian Hessian.............:        0

Hessian approximation will be done in the space of all 31 x variables.

```

Running the debugger shows a segmentation fault at that line. Here is the stack trace before crashing:

```
1  mkl_blas_avx2_xdcopy                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                         0x7f0bb1cce362 
2  mkl_blas_dcopy                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               0x7f0bb0e79e45 
3  __kmp_invoke_microtask                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       0x7f0dc33ed7d3 
4  __kmp_invoke_task_func                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       0x7f0dc33b454a 
5  __kmp_fork_call                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                              0x7f0dc33b5d47 
6  __kmpc_fork_call                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             0x7f0dc3375100 
7  mkl_blas_dcopy                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               0x7f0bb0e79c39 
8  mkl_blas.dcopy                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               0x7f0bb0c6960a 
9  Ipopt::IpBlasDcopy(int, double const *, int, double *, int)                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  0x7f0b93917ef1 
10 Ipopt::TNLPAdapter::GetStartingPoint(Ipopt::SmartPtr<Ipopt::Vector>, bool, Ipopt::SmartPtr<Ipopt::Vector>, bool, Ipopt::SmartPtr<Ipopt::Vector>, bool, Ipopt::SmartPtr<Ipopt::Vector>, bool, Ipopt::SmartPtr<Ipopt::Vector>, bool)                                                                                                                                                                                                                                                                                                                                           0x7f0b937d5ebe 
11 Ipopt::GradientScaling::DetermineScalingParametersImpl(Ipopt::SmartPtr<Ipopt::VectorSpace const>, Ipopt::SmartPtr<Ipopt::VectorSpace const>, Ipopt::SmartPtr<Ipopt::VectorSpace const>, Ipopt::SmartPtr<Ipopt::MatrixSpace const>, Ipopt::SmartPtr<Ipopt::MatrixSpace const>, Ipopt::SmartPtr<Ipopt::SymMatrixSpace const>, Ipopt::Matrix const&, Ipopt::Vector const&, Ipopt::Matrix const&, Ipopt::Vector const&, double&, Ipopt::SmartPtr<Ipopt::Vector>&, Ipopt::SmartPtr<Ipopt::Vector>&, Ipopt::SmartPtr<Ipopt::Vector>&)                                              0x7f0b9383ab5a 
12 Ipopt::StandardScalingBase::DetermineScaling(Ipopt::SmartPtr<Ipopt::VectorSpace const>, Ipopt::SmartPtr<Ipopt::VectorSpace const>, Ipopt::SmartPtr<Ipopt::VectorSpace const>, Ipopt::SmartPtr<Ipopt::MatrixSpace const>, Ipopt::SmartPtr<Ipopt::MatrixSpace const>, Ipopt::SmartPtr<Ipopt::SymMatrixSpace const>, Ipopt::SmartPtr<Ipopt::MatrixSpace const>&, Ipopt::SmartPtr<Ipopt::MatrixSpace const>&, Ipopt::SmartPtr<Ipopt::SymMatrixSpace const>&, Ipopt::Matrix const&, Ipopt::Vector const&, Ipopt::Matrix const&, Ipopt::Vector const&)                             0x7f0b9388a429 
[... cut ...]
68 clone                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            clone.S                 95  0x7f0de0ad988f 

```

traversaro, 2020-04-28:
Note, this was happening on Ubuntu 18.04 with the system installed ipopt. 

This issues seems related to https://projects.coin-or.org/Ipopt/ticket/285, that was migrated to https://github.com/coin-or/Ipopt/issues/285 . The issue was closed on the basis that the matlab interface is now maintained at https://github.com/ebertolazzi/mexIPOPT, but  I now suspect that probably you would the same issue even using `mexIPOPT`. 

The key observation is that the program is crashing while calling `mkl_blas_avx2_xdcopy`, that is a function of the MKL-specific BLAS implementation optimized for `avx2` architectures. Clearly the system ipopt (when executed outside of matlab) does not use that function.

In theory different [BLAS implementations](https://en.wikipedia.org/wiki/Basic_Linear_Algebra_Subprograms) should be ABI compatible. However, given that the system is crashing at the `mkl_blas_avx2_xdcopy`, what I suspect is happening is that the mkl BLAS used by MATLAB has some constraint on the alignment of the passed memory, and the allocator used in walking-controllers/IPOPT has no such constraint. If that is indeed the problem, we would need to force IPOPT to call the system  blas that it usually use, instead of  calling the internal mkl blas of Matlab.

I think we should try avoid to use the avx2 blas functions in Matlab with the following strategies: 
1. Set the `MKL_CBWR` env variable to `SSE2` (see https://software.intel.com/en-us/mkl-linux-developer-guide-setting-the-environment-variable-for-conditional-numerical-reproducibility and https://software.intel.com/en-us/onemkl-linux-developer-guide-specifying-code-branches#D66A7E19-D6C4-4A02-A8CE-DEE306C1CA3F) in the shell were you launch Matlab 
2. If 1. does not work, try to set the env variables `BLAS_VERSION` to `/usr/lib/x86_64-linux-gnu/liblapack.so.3` and `LAPACK_VERSION` to `/usr/lib/x86_64-linux-gnu/liblapack.so.3` in a shell and start Matlab in that shell

In both cases, even if the program continues to crash, it would be interesting to run in the matlab shell the commands `version -blas` and `version -lapack`.

traversaro, 2020-04-29:
Previous  related  comment that provides a bit more background:

> By definition, all BLAS and LAPACK implementation share the same headers as they have the same ABI, so compiling against a different implementation does not change anything. In  particular, the signature of the dcopy function (see http://www.netlib.org/lapack/explore-html/da/d6c/dcopy_8f.html) will always be the same. The problem is that I guess is that if `mkl_blas_avx2_xdcopy` requires 32 byte alignment (see https://software.intel.com/en-us/forums/intel-math-kernel-library/topic/278730) and the allocator used by IPOPT/iDynTree does not guarantee this alignment, then the problem arise. 
>
> However, I think that changing `BLAS_VERSION`, `LAPACK_VERSION` or even `MKL_CBWR` is the a quite easy attempt, as it just involves setting an environmental variable. 
>
> In the long run if the alignment is actually the problem, using a custom `malloc` implementation that ensures the required alignment.
> 
> One thing that I realized yesterday night is that interesting this issue was not experienced by the people that recently used iDynTree MATLAB bindings for inverse kinematics (see https://github.com/robotology/idyntree/pull/633), perhaps they were just lucky with the alignment (if that is the problem). @lrapetti @claudia-lat