gbionics/idyntree #685: Collect features for next release in matlab Visualizer

fjandrad, 2020-04-29:
Here we will collect the features we want to add into the  matlab Visualizer.

- Avoiding external functions as much as possible : replace mesh2tri for surf2patch
- Exploit a bit better the figure strucutres in matlab: adding tags to the transforms
- Support .dae ( collada ) files
- Adding objects acting on a frame

fjandrad, 2020-04-29:
### Add tags to transforms can be done by adding in the set of the matrix to transform object

`set(transform,'Matrix',w_H_link,'Tag',linkMeshInfo.linkName);`

traversaro, 2021-01-26:
@fjandrad if you do not plan to work on this anymore, can we close this issue and just keep the issue for the individual improvements? Thanks!

fjandrad, 2021-01-26:
Sure, you are right.

closed 2021-01-26 by fjandrad