gbionics/idyntree #686: iDynTree inverse kinematics crashes in MATLAB on Ubuntu 18.04 LTS

gabrielenava, 2020-04-29:
With @claudia-lat we tried to run the iDynTree inverse kinematics in MATLAB using the MATLAB bindings on Ubuntu. MATLAB crashes right after IPOPT is opened:

```
******************************************************************************

This program contains Ipopt, a library for large-scale nonlinear optimization.

 Ipopt is released as open source code under the Eclipse Public License (EPL).

         For more information visit http://projects.coin-or.org/Ipopt

******************************************************************************

--------------------------------------------------------------------------------

       Segmentation violation detected at mer apr 29 12:23:44 2020 +0200

--------------------------------------------------------------------------------

Configuration:

  Crash Decoding           : Disabled - No sandbox or build area path

  Crash Mode               : continue (default)

  Default Encoding         : UTF-8

  Deployed                 : false

  GNU C Library            : 2.27 stable

  Graphics Driver          : Unknown software 

  Graphics card 1          : 0x10de ( 0x10de ) 0x1b06 Version 435.21.0.0 (0-0-0)

  Java Version             : Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode

  MATLAB Architecture      : glnxa64

  MATLAB Entitlement ID    : 6829974

  MATLAB Root              : /usr/local/MATLAB/R2019b

  MATLAB Version           : 9.7.0.1296695 (R2019b) Update 4

  OpenGL                   : software

  Operating System         : Ubuntu 18.04.3 LTS

  Process ID               : 21823

  Processor ID             : x86 Family 6 Model 85 Stepping 4, GenuineIntel

  Session Key              : 657523bf-e38c-4e7d-b020-bc18a66b64c1

  Static TLS mitigation    : Disabled: Unnecessary

  Window System            : No active display

Fault Count: 12

Abnormal termination:

Segmentation violation

Register State (from fault):

  RAX = 5555d62a4a7d8870  RBX = 578286ffeaaaaab1

  RCX = 4852bfe2f5284d00  RDX = 046677ff55555548

  RSP = 00007fe4eddb4cc0  RBP = 00007fe4eddb4db0

  RSI = 3b1f57e2f5284d28  RDI = df4877e2f5284d08

   R8 = a4299fe2f5284d20   R9 = 5555d6324a7d8878

  R10 = 0000000000000008  R11 = d2150fe2f5284d30

  R12 = 0000000800000008  R13 = 690ac7e2f5284d38

  R14 = 0000000000000000  R15 = 5555d6024a7d8848

  RIP = 00007fe4efe5613e  EFL = 0000000000010202

  CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):

[  0] 0x00007fe4efe5613e        /usr/local/MATLAB/R2019b/bin/glnxa64/mkl.so+20001086 mkl_blas_avx512_xdcopy+00000398

[  1] 0x00007fe4ef167e45        /usr/local/MATLAB/R2019b/bin/glnxa64/mkl.so+06442565 mkl_blas_dcopy+00000949

[  2] 0x00007fe63047e7d3 /usr/local/MATLAB/R2019b/bin/glnxa64/../../sys/os/glnxa64/libiomp5.so+00931795 __kmp_invoke_microtask+00000147

[  3] 0x00007fe63044554a /usr/local/MATLAB/R2019b/bin/glnxa64/../../sys/os/glnxa64/libiomp5.so+00697674

[  4] 0x00007fe630444c0b /usr/local/MATLAB/R2019b/bin/glnxa64/../../sys/os/glnxa64/libiomp5.so+00695307

[  5] 0x00007fe63047ec30 /usr/local/MATLAB/R2019b/bin/glnxa64/../../sys/os/glnxa64/libiomp5.so+00932912

[  6] 0x00007fe64ac266db              /lib/x86_64-linux-gnu/libpthread.so.0+00030427

[  7] 0x00007fe64a3b788f                    /lib/x86_64-linux-gnu/libc.so.6+01185935 clone+00000063

[  8] 0x0000000000000000                                   <unknown-module>+00000000

** This crash report has been saved to disk as /home/icub/matlab_crash_dump.21823-2 **

MATLAB is exiting because of fatal error

Killed

```

The error is reproducible on UBUNTU 18.04 LTS with iDynTree in the latest `devel` branch. Looking at the logger, it seems there is a problem with the `mkl.so` library. In particular I've found [this comment](https://www.mathworks.com/matlabcentral/answers/223469-how-to-revert-matlab-back-to-it-s-default-blas-and-lapack-libraries-after-setting-environment-variab) online.

I think it is related to the issue we are experiencing, but I don't know how to properly fix the problem.

cc @lrapetti @bemilio

bemilio, 2020-04-29:
related https://github.com/robotology/idyntree/issues/684 

CC @traversaro @S-Dafarra

gabrielenava, 2020-04-29:
Side note: we tested the code on the DIC shared computer `monster`. I also tested on my personal laptop and I had the same issue, but in my  curiously I can use standalone IPOPT bindings without issues. I've created the standalone bindings following the instructions here: https://github.com/ebertolazzi/mexIPOPT

traversaro, 2020-04-29:
As suggested by @bemilio, I think it is worth trying the workaround mentioned in https://github.com/robotology/idyntree/issues/684 . `mexIPOPT` probably works because it uses its own static version of the used blas library, see https://github.com/ebertolazzi/mexIPOPT/blob/master/README_linux_how_to_compile_ipopt.md .

traversaro, 2020-04-29:
By the way, I suggest to continue to post in https://github.com/robotology/idyntree/issues/684 (as the problem is the same) and mark this issue as duplicated.

gabrielenava, 2020-04-29:
Closing the issue as duplicated of #684.

closed 2020-04-29 by gabrielenava