gbionics/idyntree #702: Broken link to floating base conventions documentation.

xEnVrE, 2020-06-19:
In [this](https://robotology.github.io/idyntree/master/classiDynTree_1_1KinDynComputations.html) page of the documentation at some point it is written that:

_This class supports three possible convention to express the floating base information : the inertial, the body-fixed and the mixed convention. To get more info on this three conventions, check section II.C of On the Base Frame Choice in Free-Floating Mechanical Systems and its Connection to Centroidal Dynamics https://traversaro.github.io/preprints/changebase.pdf_

Unfortunately, the link seems to be broken.

traversaro, 2020-06-19:
For reference, the linked document was https://www.researchgate.net/profile/Daniele_Pucci/publication/312200239_A_Unified_View_of_the_Equations_of_Motion_used_for_Control_Design_of_Humanoid_Robots/links/5878dd9708aed3826ae1d3a3/A-Unified-View-of-the-Equations-of-Motion-used-for-Control-Design-of-Humanoid-Robots.pdf  , but probably you can find more on this in Chapter 3 of my PhD thesis: https://traversaro.github.io/phd-thesis/traversaro-phd-thesis.pdf .

traversaro, 2020-06-19:
Another useful document on  notation may be https://research.tue.nl/en/publications/multibody-dynamics-notation-version-2 .