gbionics/idyntree #714: Broken  compilation on devel branch  with CMake 3.16

traversaro, 2020-07-08:
Originally reported by @AlbertoRemusIIT : 

```
In file included from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Utils.h:40:0,
                 from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/VectorFixSize.h:14,
                 from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Direction.h:16,
                 from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Axis.h:16,
                 from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/src/Axis.cpp:11:
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:2:17: warning: extra tokens at end of #ifndef directive
 #ifndef IDYNTREE-CORE_EXPORT_H
                 ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:3:17: warning: ISO C++11 requires whitespace after the macro name
 #define IDYNTREE-CORE_EXPORT_H
                 ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:5:16: warning: extra tokens at end of #ifdef directive
 #ifdef IDYNTREE-CORE_STATIC_DEFINE
                ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:6:19: warning: ISO C++11 requires whitespace after the macro name
 #  define IDYNTREE-CORE_EXPORT
                   ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:6:0: warning: "IDYNTREE" redefined
 #  define IDYNTREE-CORE_EXPORT
 
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:3:0: note: this is the location of the previous definition
 #define IDYNTREE-CORE_EXPORT_H
 
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:7:19: warning: ISO C++11 requires whitespace after the macro name
 #  define IDYNTREE-CORE_NO_EXPORT
                   ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:7:0: warning: "IDYNTREE" redefined
 #  define IDYNTREE-CORE_NO_EXPORT
 
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:6:0: note: this is the location of the previous definition
 #  define IDYNTREE-CORE_EXPORT
 
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:24:17: warning: extra tokens at end of #ifndef directive
 #ifndef IDYNTREE-CORE_DEPRECATED
                 ^
INFO:__main__:Preprocessing /home/albertoremusiit/idjl-software/build/_deps/open62541-src/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
[ 19%] Building CXX object src/libraries/idjl-ellipse-fitting/CMakeFiles/idjl_ellipse_fitting.dir/idjl_ellipse_fitting.cpp.o
In file included from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Utils.h:40:0,
                 from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/MatrixFixSize.h:15,
                 from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/ArticulatedBodyInertia.h:14,
                 from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/src/ArticulatedBodyInertia.cpp:12:
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:2:17: warning: extra tokens at end of #ifndef directive
 #ifndef IDYNTREE-CORE_EXPORT_H
                 ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:3:17: warning: ISO C++11 requires whitespace after the macro name
 #define IDYNTREE-CORE_EXPORT_H
                 ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:5:16: warning: extra tokens at end of #ifdef directive
 #ifdef IDYNTREE-CORE_STATIC_DEFINE
                ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:6:19: warning: ISO C++11 requires whitespace after the macro name
 #  define IDYNTREE-CORE_EXPORT
                   ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:6:0: warning: "IDYNTREE" redefined
 #  define IDYNTREE-CORE_EXPORT
 
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:3:0: note: this is the location of the previous definition
 #define IDYNTREE-CORE_EXPORT_H
 
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:7:19: warning: ISO C++11 requires whitespace after the macro name
 #  define IDYNTREE-CORE_NO_EXPORT
                   ^
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:7:0: warning: "IDYNTREE" redefined
 #  define IDYNTREE-CORE_NO_EXPORT
 
/home/albertoremusiit/idjl-software/build/_deps/idyntree-build/src/core/CoreExport.h:6:0: note: this is the location of the previous definition
 #  define IDYNTREE-CORE_EXPORT
[... 81 lines cut ...]
/home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Utils.h:46:5: error: ‘IDYNTREE_CORE_EXPORT’ does not name a type; did you mean ‘IDYNTREE_CONSTEXPR’?
     IDYNTREE_CORE_EXPORT extern int UNKNOWN;
     ^~~~~~~~~~~~~~~~~~~~
     IDYNTREE_CONSTEXPR
/home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Utils.h:49:5: error: ‘IDYNTREE_CORE_EXPORT’ does not name a type; did you mean ‘IDYNTREE_CONSTEXPR’?
     IDYNTREE_CORE_EXPORT extern double DEFAULT_TOL;
     ^~~~~~~~~~~~~~~~~~~~
     IDYNTREE_CONSTEXPR
/home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Utils.h:132:86: error: ‘DEFAULT_TOL’ was not declared in this scope
     bool checkDoublesAreEqual(const double & val1, const double & val2, double tol = DEFAULT_TOL);
                                                                                      ^~~~~~~~~~~
/home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Utils.h:132:86: note: suggested alternative: ‘EFAULT’
     bool checkDoublesAreEqual(const double & val1, const double & val2, double tol = DEFAULT_TOL);
                                                                                      ^~~~~~~~~~~
                                                                                      EFAULT
In file included from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Axis.h:16:0,
                 from /home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/src/Axis.cpp:11:
/home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Direction.h:68:35: error: ‘DEFAULT_TOL’ was not declared in this scope
         void Normalize(double tol=DEFAULT_TOL);
                                   ^~~~~~~~~~~
/home/albertoremusiit/idjl-software/build/_deps/idyntree-src/src/core/include/iDynTree/Core/Direction.h:68:35: note: suggested alternative: ‘EFAULT’
         void Normalize(double tol=DEFAULT_TOL);
                                   ^~~~~~~~~~~
                                   EFAULT
```

traversaro, 2020-07-08:
Apparently the main build works fine, but it breaks if iDynTree is included in a bigger project via `add_subdirectory` .

traversaro, 2020-07-08:
> 
> 
> Apparently the main build works fine, but it breaks if iDynTree is included in a bigger project via `add_subdirectory` .

The CMake  version does not matter, what creates the problem is the invocation of `find_package(VTK)` when VTK 6 is found.

GiulioRomualdi, 2020-07-08:
That's really complex to spot!

traversaro, 2020-07-08:
> 
> 
> That's really complex to spot!

I was  lucky as I wanted to understand  what was going on in the `GenerateExportHeader` file in CMake, so I added some good old debug prints in it (even if it is a system file, but you only live once). In the failing case, no print were happening, so it was clear that some other `GenerateExportHeader` module was being included. As the include path of the `include` command is given by the `CMAKE_MODULE_PATH`, I printed the  `CMAKE_MODULE_PATH` and I found  that the only not expected one was a vtk-related path. Searching on google vtk + GenerateExportHeader gave as first result https://gitlab.kitware.com/vtk/vtk/commit/44dafc3e2b9287d74978d18d79fc4e6f1f068342 and solved the mistery.

traversaro, 2024-01-09:
We do not support anymore compilation with Ubuntu 18.04 apt dependendencies, so it should be safe to close this issue.

closed 2024-01-09 by traversaro