gbionics/idyntree #722: Coefficient not recomputed in CubicSpline interpolation when boundary conditions are set

GiulioRomualdi, 2020-08-12:
I was checking the code of the cubic spline interpolation and I realized that the coefficients are not recomputed when the `setInitialConditons()` and `setFinalConditions()` functions are called. 

https://github.com/robotology/idyntree/blob/6712307e378f84fd8b98276b057e8c5cf82855bd/src/core/src/CubicSpline.cpp#L163-L173

However, the initial and final velocity conditions are required to compute the coefficients
https://github.com/robotology/idyntree/blob/6712307e378f84fd8b98276b057e8c5cf82855bd/src/core/src/CubicSpline.cpp#L77-L80

Once the `CubicSpline` object is created the velocity and the acceleration are reset to zero.
https://github.com/robotology/idyntree/blob/6712307e378f84fd8b98276b057e8c5cf82855bd/src/core/src/CubicSpline.cpp#L19-L23
 This means that that if one sets the boundary condition after calling `setData()` the new velocities (and accelerations) are not taken into account in the spline.

Unfortunately, this is not documented in the documentation and it may lead to error for the final user. 

The test was not failing because the velocities and acceleration are set before calling `setData()`
https://github.com/robotology/idyntree/blob/6712307e378f84fd8b98276b057e8c5cf82855bd/src/core/tests/CubicSplineUnitTest.cpp#L33-L40

Applying this patch to the test
```patch
diff --git a/src/core/tests/CubicSplineUnitTest.cpp b/src/core/tests/CubicSplineUnitTest.cpp
index 50fab7d8..179d1db6 100644
--- a/src/core/tests/CubicSplineUnitTest.cpp
+++ b/src/core/tests/CubicSplineUnitTest.cpp
@@ -30,6 +30,7 @@ bool setNpoints(size_t n, double initialTime, double finalTime, const Vector4& p
         yVec(i) = parameters(0) + parameters(1)*tVec(i) + parameters(2)*pow(tVec(i),2) + parameters(3)*pow(tVec(i),3);
     }
 
+    assertTrue(spline.setData(tVec, yVec));
     double initialVelocity = parameters(1) + 2*parameters(2)*initialTime + 3*parameters(3)*pow(initialTime,2);
     double initialAcceleration = 2*parameters(2) + 6*parameters(3)*initialTime;
     spline.setInitialConditions(initialVelocity, initialAcceleration);
@@ -37,7 +38,6 @@ bool setNpoints(size_t n, double initialTime, double finalTime, const Vector4& p
     double finalAcceleration = 2*parameters(2) + 6*parameters(3)*finalTime;
     spline.setFinalConditions(finalVelocity, finalAcceleration);
 
-    assertTrue(spline.setData(tVec, yVec));
     return true;
 }
 ```
the test fails with the following error is thrown
```
./bin/CubicSplineUnitTest 
Vel i = :0 : assertDoubleAreEqual failure: val1 is 4.3559 while val2 is 0
```

The following patch completely solves the bug
```patch
diff --git a/src/core/include/iDynTree/Core/CubicSpline.h b/src/core/include/iDynTree/Core/CubicSpline.h
index 14a9cb29..6ce1ac96 100644
--- a/src/core/include/iDynTree/Core/CubicSpline.h
+++ b/src/core/include/iDynTree/Core/CubicSpline.h
@@ -41,8 +41,8 @@ namespace iDynTree
         CubicSpline(unsigned int buffersDimension);
         
         bool setData(const iDynTree::VectorDynSize& time, const iDynTree::VectorDynSize& yData);
-        void setInitialConditions(double initialVelocity, double initialAcceleration);
-        void setFinalConditions(double finalVelocity, double finalAcceleration);
+        bool setInitialConditions(double initialVelocity, double initialAcceleration);
+        bool setFinalConditions(double finalVelocity, double finalAcceleration);
         double evaluatePoint(double t);
         double evaluatePoint(double t, double& velocity, double& acceleration);
     };
diff --git a/src/core/src/CubicSpline.cpp b/src/core/src/CubicSpline.cpp
index 8cd103e2..fd71e751 100644
--- a/src/core/src/CubicSpline.cpp
+++ b/src/core/src/CubicSpline.cpp
@@ -160,16 +160,20 @@ bool iDynTree::CubicSpline::computePhasesDuration()
     return true;
 }
 
-void iDynTree::CubicSpline::setInitialConditions(double initialVelocity, double initialAcceleration)
+bool iDynTree::CubicSpline::setInitialConditions(double initialVelocity, double initialAcceleration)
 {
     m_v0 = initialVelocity;
     m_a0 = initialAcceleration;
+
+    return this->computeCoefficients();
 }
 
-void iDynTree::CubicSpline::setFinalConditions(double finalVelocity, double finalAcceleration)
+bool iDynTree::CubicSpline::setFinalConditions(double finalVelocity, double finalAcceleration)
 {
     m_vf = finalVelocity;
     m_af = finalAcceleration;
+
+    return this->computeCoefficients();
 }
 
 double iDynTree::CubicSpline::evaluatePoint(double t)
```

@robotology/iit-dynamic-interaction-control

GiulioRomualdi, 2020-08-12:
https://github.com/robotology/idyntree/pull/723 fixes the bug

S-Dafarra, 2020-08-17:
@GiulioRomualdi thanks for dealing with this.

traversaro, 2020-09-07:
Fixed in https://github.com/robotology/idyntree/pull/723 by @GiulioRomualdi .

PR #723 (2020-08-14): Change the logic used to update the Coefficients in the CubicSpline class
This PR fixed #722 

@traversaro I opened the PR on `devel`. Let me know if you prefer to merge it on `master`. I will update the [`changelog.md`](https://github.com/robotology/idyntree/blob/master/CHANGELOG.md) accordingly
closed 2020-09-07 by traversaro