gbionics/idyntree #727: Master branch causes configuration failures of packages depending on model_io on Windows

S-Dafarra, 2020-09-02:
When trying to compile the ``robotology-superbuild``, several packages that depend on ``model_io`` fail to configure with the error:
```
40>CMake Error in src/RetargetingHelper/CMakeLists.txt:
40>  Imported target "iDynTree::idyntree-modelio-xml" includes non-existent path
40>
40>    "C:/robotology-superbuild/robotology/iDynTree/src/model_io/xml/include/private"
40>
40>  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
40>
40>  * The path was deleted, renamed, or moved to another location.
40>
40>  * An install or uninstall procedure did not complete successfully.
40>
40>  * The installation package was faulty and references files it does not
40>  provide.
```
I am trying to compile on Windows 10 using Visual Studio 16 2019
Here a portion of the log (the initial part got cut by Visual Studio) 
[superbuild-error.txt](https://github.com/robotology/idyntree/files/5160709/superbuild-error.txt)

Note that ``iDynTree`` is compiling correctly. I also tried removing the ``install`` folder.

traversaro, 2020-09-02:
Can you copy/paste or attach the files in iDynTreeTargets-Release.cmake or with similar names that should be in `install/lib/cmake/iDynTree` ?

traversaro, 2020-09-02:
Can you double check that iDyntTree commit is indeed the latest commit in master ( https://github.com/robotology/idyntree/commit/6dcb2c8cddf85cf8d76f15f7fbdf14ebc9045d0c )?

S-Dafarra, 2020-09-02:
> Can you double check that iDyntTree commit is indeed the latest commit in master ( 6dcb2c8 )?

Yes, I tried to pull first.

I simply removed https://github.com/robotology/idyntree/blob/master/src/model_io/xml/CMakeLists.txt#L37 and it compiled. I can try to restore it and attach here the content of ``iDynTreeTargets-Release.cmake`` if you like.

S-Dafarra, 2020-09-02:
> Can you copy/paste or attach the files in iDynTreeTargets-Release.cmake or with similar names that should be in install/lib/cmake/iDynTree ?

[iDynTreeTargets.txt](https://github.com/robotology/idyntree/files/5160836/iDynTreeTargets.txt)
[iDynTreeTargets-release.txt](https://github.com/robotology/idyntree/files/5160837/iDynTreeTargets-release.txt)

(I have changed the extensions to upload them here)

traversaro, 2020-09-02:
> [iDynTreeTargets.txt](https://github.com/robotology/idyntree/files/5160836/iDynTreeTargets.txt) [...]

This is the version that was working or not?

S-Dafarra, 2020-09-02:
> This is the version that was working or not?

That version in **not** we working.

traversaro, 2020-09-02:
This is quite strange, as in those files do not contain any kind of reference to `C:/robotology-superbuild/robotology/iDynTree/src/model_io/xml/include/private`.

S-Dafarra, 2020-09-02:
I thought that too. Where the ``INTERFACE_INCLUDE_DIRECTORIES`` variable of a target is supposed to set? :thinking:

traversaro, 2020-09-02:
It is set in `iDynTreeTargets.txt`, and it does not contain any reference to that. Now that I think about it, I think I know what is happening: for some reason the `walking-retargeting` CMake project is finding iDynTree from the build directory, rather then from the install. You can check this by looking at the `iDynTree_DIR` variable in the walking-retargeting project.

traversaro, 2020-09-02:
So, indeed there was a problem in using iDynTree from the build directory, that will be fixed in https://github.com/robotology/idyntree/pull/728, but the big problem here seems that for some reason iDynTree is found from the build instead that from the install. Are you sure that there is not any strange set `iDynTree_DIR` env variable in your enviroment?

S-Dafarra, 2020-09-02:
Nice catch, iDynTree is found on the build folder, but I did not see anything strange on the path. I used the script provided by the ``robotology-superbuild`` to set it up. I double checked the env variables and none point to the build folder :thinking:

S-Dafarra, 2020-09-02:
Do you have any other idea on why the build folder was found? If it is not because of env variables, can it be due to some ``superbuild`` machinery?

traversaro, 2020-09-02:
`find_package` is a complex beast, and its search procedure is https://cmake.org/cmake/help/latest/command/find_package.html#search-procedure . If it is continuing to find the iDynTree build even after you delete the build of walking-teleoperation, then I think it may be possible to investigate the issue, instead if from a clean build nothing happens, it would probably be difficult to investigate what happened.

traversaro, 2020-09-02:
To debug a bit, I guess it would be useful to: 
* Set [`CMAKE_FIND_DEBUG_MODE`](https://cmake.org/cmake/help/latest/variable/CMAKE_FIND_DEBUG_MODE.html) before a `find_package(iDynTree REQUIRED)` that is finding iDynTree in the build, and report the output. 
* Report the current windows environment (it is sufficient to execute `env` in a Git Bash and copy&paste the output)

S-Dafarra, 2020-09-02:
> * Report the current windows environment (it is sufficient to execute `env` in a Git Bash and copy&paste the output)

```
icub@DESKTOP-T0NQNLN MINGW64 /c/robotology-superbuild/build/robotology/UnicyclePlanner (master)
$ env
USERDOMAIN=DESKTOP-T0NQNLN
OS=Windows_NT
COMMONPROGRAMFILES=C:\Program Files\Common Files
OculusBase=C:\Program Files\Oculus\
PROCESSOR_LEVEL=6
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
CommonProgramW6432=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
LANG=en_US.UTF-8
MSYSTEM_CARCH=x86_64
DISPLAY=needs-to-be-defined
HOSTNAME=DESKTOP-T0NQNLN
PUBLIC=C:\Users\Public
CONFIG_SITE=/mingw64/etc/config.site
EXEPATH=C:\Program Files\Git
MSYSTEM_CHOST=x86_64-w64-mingw32
USERNAME=icub
ROBOTOLOGY_SUPERBUILD_SOURCE_DIR=C:/robotology-superbuild
LOGONSERVER=\\DESKTOP-T0NQNLN
PROCESSOR_ARCHITECTURE=AMD64
OculusSDK_ROOT=C:\OculusSDK
LOCALAPPDATA=C:\Users\icub\AppData\Local
COMPUTERNAME=DESKTOP-T0NQNLN
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
!::=::\
SYSTEMDRIVE=C:
USERPROFILE=C:\Users\icub
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
SYSTEMROOT=C:\WINDOWS
USERDOMAIN_ROAMINGPROFILE=DESKTOP-T0NQNLN
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 158 Stepping 10, GenuineIntel
MINGW_PACKAGE_PREFIX=mingw-w64-x86_64
PWD=/c/robotology-superbuild/build/robotology/UnicyclePlanner
SSH_ASKPASS=/mingw64/libexec/git-core/git-gui--askpass
HOME=/c/Users/icub
TMP=/tmp
CMAKE_PREFIX_PATH=C:\robotology\vcpkg\installed\x64-windows;C:\robotology\vcpkg\installed\x64-windows\debug;C:/robotology-superbuild/build/install
TERM_PROGRAM=mintty
TERM_PROGRAM_VERSION=3.1.6
MSYSTEM_PREFIX=/mingw64
CMAKE_PROGRAM_PATH=C:\robotology\vcpkg\installed\x64-windows\tools\protobuf
OneDrive=C:\Users\icub\OneDrive
CybSDK_DIR=C:\CybSDK_Cpp
PROCESSOR_REVISION=9e0a
FPS_BROWSER_USER_PROFILE_STRING=Default
TMPDIR=/tmp
ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX=C:/robotology-superbuild/build/install
NUMBER_OF_PROCESSORS=12
ProgramW6432=C:\Program Files
COMSPEC=C:\WINDOWS\system32\cmd.exe
APPDATA=C:\Users\icub\AppData\Roaming
SHELL=/usr/bin/bash
TERM=xterm
YARP_DATA_DIRS=C:/robotology-superbuild/build/install/share/yarp;C:/robotology-superbuild/build/install/share/iCub;C:/robotology-superbuild/build/install/share/ICUBcontrib
WINDIR=C:\WINDOWS
MINGW_CHOST=x86_64-w64-mingw32
ProgramData=C:\ProgramData
SHLVL=1
PLINK_PROTOCOL=ssh
ACLOCAL_PATH=/mingw64/share/aclocal:/usr/share/aclocal
PROGRAMFILES=C:\Program Files
MANPATH=/mingw64/local/man:/mingw64/share/man:/usr/local/man:/usr/share/man:/usr/man:/share/man
ORIGINAL_TEMP=/tmp
ORIGINAL_TMP=/tmp
ALLUSERSPROFILE=C:\ProgramData
TEMP=/tmp
DriverData=C:\Windows\System32\Drivers\DriverData
MSYSTEM=MINGW64
MINGW_PREFIX=/mingw64
SESSIONNAME=Console
ProgramFiles(x86)=C:\Program Files (x86)
PATH=/c/Users/icub/bin:/mingw64/bin:/usr/local/bin:/usr/bin:/bin:/mingw64/bin:/usr/bin:/c/Users/icub/bin:/c/Program Files/Oculus/Support/oculus-runtime:/c/WINDOWS/system32:/c/WINDOWS:/c/WINDOWS/System32/Wbem:/c/WINDOWS/System32/WindowsPowerShell/v1.0:/c/WINDOWS/System32/OpenSSH:/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/DAL:/c/Program Files/Intel/Intel(R) Management Engine Components/DAL:/c/Program Files/Microsoft SQL Server/130/Tools/Binn:/c/Program Files/CMake/bin:/cmd:/c/Users/icub/AppData/Local/Microsoft/WindowsApps:/c/robotology/vcpkg/installed/x64-windows/bin:/c/robotology/vcpkg/installed/x64-windows/debug/bin:/c/robotology-superbuild/build/install/bin:/usr/bin/vendor_perl:/usr/bin/core_perl
PS1=\[\033]0;$TITLEPREFIX:$PWD\007\]\n\[\033[32m\]\u@\h \[\033[35m\]$MSYSTEM \[\033[33m\]\w\[\033[36m\]`__git_ps1`\[\033[0m\]\n$
HOMEDRIVE=C:
PKG_CONFIG_PATH=/mingw64/lib/pkgconfig:/mingw64/share/pkgconfig
INFOPATH=/usr/local/info:/usr/share/info:/usr/info:/share/info
HOMEPATH=\Users\icub
ORIGINAL_PATH=/mingw64/bin:/usr/bin:/c/Users/icub/bin:/c/Program Files/Oculus/Support/oculus-runtime:/c/WINDOWS/system32:/c/WINDOWS:/c/WINDOWS/System32/Wbem:/c/WINDOWS/System32/WindowsPowerShell/v1.0:/c/WINDOWS/System32/OpenSSH:/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/DAL:/c/Program Files/Intel/Intel(R) Management Engine Components/DAL:/c/Program Files/Microsoft SQL Server/130/Tools/Binn:/c/Program Files/CMake/bin:/cmd:/c/Users/icub/AppData/Local/Microsoft/WindowsApps:/c/robotology/vcpkg/installed/x64-windows/bin:/c/robotology/vcpkg/installed/x64-windows/debug/bin:/c/robotology-superbuild/build/install/bin
BLOCKFACTORY_PLUGIN_PATH=C:/robotology-superbuild/build/install/lib/blockfactory
_=/usr/bin/env
```

S-Dafarra, 2020-09-02:
> * Set [`CMAKE_FIND_DEBUG_MODE`](https://cmake.org/cmake/help/latest/variable/CMAKE_FIND_DEBUG_MODE.html) before a `find_package(iDynTree REQUIRED)` that is finding iDynTree in the build, and report the output. Alternatively, pass the `--debug-find` option to CMake, see https://cmake.org/cmake/help/v3.17/manual/cmake.1.html#manual:cmake(1) . Note that both tools required CMake >= 3.17 .

Interestingly, only ``iDynTree`` seems to be found in the build folder: 
[image]

Here the output of cmake when setting ``FIND_DEBUG_MODE``: 
[unicycle_planner_idyntree_debug.txt](https://github.com/robotology/idyntree/files/5162020/unicycle_planner_idyntree_debug.txt)

traversaro, 2020-09-02:
Looking at the debug outputs, I have no idea what is going on.

S-Dafarra, 2020-09-02:
I think these are the interesting lines:
```
CMake Debug Log at C:/vcpkg/scripts/buildsystems/vcpkg.cmake:439 (_find_package):
  find_package considered the following paths for iDynTree.cmake

    C:/robotology-superbuild/robotology/UnicyclePlanner/cmake/FindiDynTree.cmake
    C:/Program Files/CMake/share/cmake-3.18/Modules/FindiDynTree.cmake

  The file was not found.

    C:/robotology-superbuild/build/robotology/iDynTree/iDynTreeConfig.cmake

Call Stack (most recent call first):
  CMakeLists.txt:55 (find_package)
```
All the rest seems to be generated by calls from within ``C:/robotology-superbuild/build/robotology/iDynTree/iDynTreeConfig.cmake``, but I have no idea on how it found the ``C:/robotology-superbuild/build/robotology/iDynTree/iDynTreeConfig.cmake`` file in the first place.

S-Dafarra, 2020-09-02:
After deleting the build and fixing the dependencies (I had two ``vcpkg`` around), this problem disappeared. Closing.

closed 2020-09-02 by S-Dafarra