gbionics/idyntree #729: Possible bug in iDynTree::Span

GiulioRomualdi, 2020-09-06:
If I'm not mistaken here
https://github.com/robotology/idyntree/blob/f9fff5998c459dacf869629c3fe629d5831f240f/src/core/include/iDynTree/Core/Span.h#L498-L499

`double` should be substituted with `reference`.

traversaro, 2020-09-07:
Good catch! Are you sure that `reference` is the correct fix? At a first glance I would say that `value_type` may be more in line with the semantics of the method.

traversaro, 2020-10-11:
Friendly ping @GiulioRomualdi , I suspect that `value_type` may be indeed a better fix. As in any case it would be an API/ABI change, I suggest to do it in the `devel`  branch,

traversaro, 2020-11-17:
Fixed in https://github.com/robotology/idyntree/pull/749 .

PR #749 (2020-10-28): Bugfix in Span class
This PR should fix #729
closed 2020-11-17 by traversaro