gbionics/idyntree #743: Broken compilation on devel branch with IDYNTREE_USES_MATLAB=ON

gabrielenava, 2020-10-06:
I have a problem compiling iDynTree in `devel` branch with  IDYNTREE_USES_MATLAB=ON. It seems there is a problem with the `FindMatlab.cmake` file:

```[cmake]
-- Found YARP: /usr/local/src/robot/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.2+43-20200411.3+gita33cf54b1")
-- Found YARP: /usr/local/src/robot/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.2+43-20200411.3+gita33cf54b1")
-- Using iCub from install
-- Found YARP: /usr/local/src/robot/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.2+43-20200411.3+gita33cf54b1")
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
CMake Error at cmake/FindMatlab.cmake:973 (message):
  [MATLAB] The MEX target name cannot be empty
Call Stack (most recent call first):
  bindings/matlab/CMakeLists.txt:27 (matlab_add_mex)
  bindings/matlab/CMakeLists.txt:94 (swig_compile_module_mex)
```

cc @diegoferigo @CarlottaSartore

traversaro, 2020-10-07:
It is probably something related to https://github.com/robotology/idyntree/commit/166f8e2eb8753e6849daf27fd071ebae0169bd45#diff-b49daac0aeab812d486fce0b390dec13 . Just to check, can you try if everything was working correctly on https://github.com/robotology/idyntree/commit/4c1ec4e32bd824de6b155d9e7fa482027782bcf7#diff-b49daac0aeab812d486fce0b390dec13 .

Giulero, 2020-10-09:
@traversaro I tried https://github.com/robotology/idyntree/commit/4c1ec4e32bd824de6b155d9e7fa482027782bcf7#diff-b49daac0aeab812d486fce0b390dec13 and it didn't work.

I went back to 582b942fa1bf17487ad49acfaad092215b8daff5 and it works.

Giulero, 2020-10-09:
Oh! Thanks @diegoferigo!

traversaro, 2020-10-11:
Fixed by https://github.com/robotology/idyntree/pull/746 .

closed 2020-10-11 by traversaro