gbionics/idyntree #758: Make iDynTree::Span compliant with the std::span implementation

GiulioRomualdi, 2020-11-02:
`std::span` has been introduced in C++20. Since `iDynTree` is C++11 compliant a custom implementation of `span` is required. DIscussing with @traversaro I noticed that our current implementation contains methods that are not defined by the standard, e.g. 

https://github.com/robotology/idyntree/blob/a97aa8407359b12cccc95313b00c654a0d444508/src/core/include/iDynTree/Core/Span.h#L507-L508

https://github.com/robotology/idyntree/blob/a97aa8407359b12cccc95313b00c654a0d444508/src/core/include/iDynTree/Core/Span.h#L515-L516

https://github.com/robotology/idyntree/blob/a97aa8407359b12cccc95313b00c654a0d444508/src/core/include/iDynTree/Core/Span.h#L499-L500

Furthermore, we introduced also introduced some custom constructors. Please check [here](https://en.cppreference.com/w/cpp/container/span/span) the `std::span` constructors.
e.g. 

https://github.com/robotology/idyntree/blob/a97aa8407359b12cccc95313b00c654a0d444508/src/core/include/iDynTree/Core/Span.h#L402-L420

As far as I understood this constructor should be substituted with something like
```cpp
template< class R >
explicit(extent != std::dynamic_extent)
constexpr span( R&& r );
```
indeed the `span` can be constructed using the `ranges` library. https://en.cppreference.com/w/cpp/ranges. Unfortunately the `ranges` library has been introduced with C++20. So we cannot use it as it is. 

The [above constructors](https://github.com/robotology/idyntree/blob/a97aa8407359b12cccc95313b00c654a0d444508/src/core/include/iDynTree/Core/Span.h#L402-L420)  are not compatible with Eigen objects.  
Indeed 
```cpp
Eigen::Vector3d v;
iDynTree::Span <double> s(v)
``` 
will not compile because `Eigen::Dense` does not defined `Container::pointer` typedef. 

On the other hand, Eigen 3.4 will provide STL compatible iterators. Please check [here](https://eigen.tuxfamily.org/dox-devel/group__TutorialSTL.html). So I hope it will also be compatible with `std::ranges`. (Probably we should double-check on this)

If `Eigen` will support `std::ranges` the following code will compile
```cpp
#include <span>
#include <Eigen/Dense>

int main ()
{
    Eigen::Vector3d m;
    std::span<double> s(m);

    return 0;
}
```

<details><summary> Click here to check the error </summary>
<p>

```
[ 50%] Building CXX object CMakeFiles/eigen_test.dir/span_eigen.cpp.o
/home/gromualdi/robot-code/test_eigen/span_eigen.cpp: In function ‘int main()’:
/home/gromualdi/robot-code/test_eigen/span_eigen.cpp:9:26: error: no matching function for call to ‘std::span<double>::span(Eigen::Vector3d&)’
    9 |     std::span<double> s(m);
      |                          ^
In file included from /home/gromualdi/robot-code/test_eigen/span_eigen.cpp:2:
/usr/include/c++/10/span:234:2: note: candidate: ‘template<class _OType, long unsigned int _OExtent>  requires (_Extent == std::dynamic_extent || _OExtent == std::dynamic_extent || _Extent == _OExtent) && (std::__is_array_convertible<_Type, _Tp>::value) constexpr std::span<_Type, _Extent>::span(const std::span<_OType, _OExtent>&) [with _OType = _OType; long unsigned int _OExtent = _OExtent; _Type = double; long unsigned int _Extent = 18446744073709551615]’
  234 |  span(const span<_OType, _OExtent>& __s) noexcept
      |  ^~~~
/usr/include/c++/10/span:234:2: note:   template argument deduction/substitution failed:
/home/gromualdi/robot-code/test_eigen/span_eigen.cpp:9:26: note:   ‘Eigen::Vector3d’ {aka ‘Eigen::Matrix<double, 3, 1>’} is not derived from ‘const std::span<_Tp, _Num>’
    9 |     std::span<double> s(m);
      |                          ^
In file included from /home/gromualdi/robot-code/test_eigen/span_eigen.cpp:2:
/usr/include/c++/10/span:226:7: note: candidate: ‘constexpr std::span<_Type, _Extent>::span(const std::span<_Type, _Extent>&) [with _Type = double; long unsigned int _Extent = 18446744073709551615]’
  226 |       span(const span&) noexcept = default;
      |       ^~~~
/usr/include/c++/10/span:226:12: note:   no known conversion for argument 1 from ‘Eigen::Vector3d’ {aka ‘Eigen::Matrix<double, 3, 1>’} to ‘const std::span<double>&’
  226 |       span(const span&) noexcept = default;
      |            ^~~~~~~~~~~
/usr/include/c++/10/span:214:2: note: candidate: ‘constexpr std::span<_Type, _Extent>::span(_Range&&) [with _Range = Eigen::Matrix<double, 3, 1>&; _Type = double; long unsigned int _Extent = 18446744073709551615]’
[... 47 lines cut ...]
/usr/include/c++/10/span:174:2: note: candidate: ‘template<class _It, class _End>  requires (contiguous_iterator<_It>) && (sized_sentinel_for<_End, _It>) && ((std::__is_compatible_ref<decltype(*(declval<_It&>)())>::value) && !(is_convertible_v<_End, std::span<_Type, _Extent>::size_type>)) constexpr std::span<_Type, _Extent>::span(_It, _End) [with _It = _It; _End = _End; _Type = double; long unsigned int _Extent = 18446744073709551615]’
  174 |  span(_It __first, _End __last)
      |  ^~~~
/usr/include/c++/10/span:174:2: note:   template argument deduction/substitution failed:
/home/gromualdi/robot-code/test_eigen/span_eigen.cpp:9:26: note:   candidate expects 2 arguments, 1 provided
    9 |     std::span<double> s(m);
      |                          ^
In file included from /home/gromualdi/robot-code/test_eigen/span_eigen.cpp:2:
/usr/include/c++/10/span:160:2: note: candidate: ‘template<class _It>  requires (contiguous_iterator<_It>) && (std::__is_compatible_ref<decltype(*(declval<_It&>)())>::value) constexpr std::span<_Type, _Extent>::span(_It, std::span<_Type, _Extent>::size_type) [with _It = _It; _Type = double; long unsigned int _Extent = 18446744073709551615]’
  160 |  span(_It __first, size_type __count)
      |  ^~~~
/usr/include/c++/10/span:160:2: note:   template argument deduction/substitution failed:
/home/gromualdi/robot-code/test_eigen/span_eigen.cpp:9:26: note:   candidate expects 2 arguments, 1 provided
    9 |     std::span<double> s(m);
      |                          ^
In file included from /home/gromualdi/robot-code/test_eigen/span_eigen.cpp:2:
/usr/include/c++/10/span:152:7: note: candidate: ‘constexpr std::span<_Type, _Extent>::span() requires  _Extent + 1 <= 1 [with _Type = double; long unsigned int _Extent = 18446744073709551615]’
  152 |       span() noexcept
      |       ^~~~
/usr/include/c++/10/span:152:7: note:   candidate expects 0 arguments, 1 provided
```
</p>
</details>

So after double-checking the future compatibility of `Eigen` with `std::span` we can think to modified the constructors in order to work also with Eigen objects.

S-Dafarra, 2020-11-02:
Rather than modifying the constructor, IMHO it would be better to define custom ``make_span`` functions (that are not present in the standard either), that can remain even after moving to the standard ``span``. These function may chack if the ``data`` and ``size`` methods are defined in a SFINAE fashion. I did something similar in ``bipedal-locomotion-framework``. For example, in [this ``static_assert``](https://github.com/dic-iit/bipedal-locomotion-framework/blob/be1e0a2ea0077e3ed745ec6891aac3e71d35824a/src/GenericContainer/include/BipedalLocomotion/GenericContainer/Vector.h#L701-L702) I check if ``span`` is constructible given the inputs, or if the ``data`` and ``size`` methods are defined. According to the case, I construct the span in different ways.

traversaro, 2020-11-02:
Any solution is fine for me, even because I think in any case we will never do:
~~~
namespace iDynTree 
{ 
    typedef std::span Span;
}
~~~
so even not having 100% std::span compatibility is ok.

traversaro, 2020-11-18:
Using Eigen with Span constructor should be solved by https://github.com/robotology/idyntree/pull/766, but I am not sure if this issue was covering also something else.

GiulioRomualdi, 2020-11-19:
In theory yes

closed 2020-11-19 by traversaro