gbionics/idyntree #761: Matlab visualization in broken on ``devel``

S-Dafarra, 2020-11-12:
By running [``visualizeRobot.m``](https://github.com/robotology/idyntree/blob/master/examples/iDynTreeWrappers/visualizeRobot.m), it exits with the following error:
```
Unrecognized method, property, or field 'linkSolidShapes' for class 'iDynTree.ModelSolidShapes'.

Error in SwigRef/subsref (line 26)
            [varargout{1}] = builtin('subsref',self,substruct('.',s.subs,'()',{}));

Error in iDynTreeWrappers.getMeshes (line 25)
linkSolidShapesV=visual.linkSolidShapes;

Error in iDynTreeWrappers.prepareVisualization (line 74)
[linkMeshInfo,map]=iDynTreeWrappers.getMeshes(model,meshFilePrefix);

Error in visualizeRobot (line 84)
[visualizer,objects]=iDynTreeWrappers.prepareVisualization(KinDynModel,meshFilePrefix,...
 
```

I guess this is due to #751 
@nunoguedelha @gabrielenava @Giulero @CarlottaSartore

Giulero, 2020-11-12:
Hi @S-Dafarra! I faced the same issue. It seems that in devel some methods acquired the `get` prefix. 
Changing from `linkSolidShape` to `getLinkSolidShape` should work.

S-Dafarra, 2020-11-12:
> Hi @S-Dafarra! I faced the same issue. It seems that in devel some methods acquired the `get` prefix.
> Changing from `linkSolidShape` to `getLinkSolidShape` should work.

Ah great! If you could fill in a PR for that, it would be great

Giulero, 2020-11-12:
> Ah great! If you could fill in a PR for that, it would be great
I was planning it :D

traversaro, 2020-11-12:
> I guess this is due to #751

Exactly, the method were deprecated in https://github.com/robotology/idyntree/pull/739, but unfortunately the deprecation are not easy to detect on MATLAB.

traversaro, 2020-11-13:
Fixed by @Giulero  in https://github.com/robotology/idyntree/pull/762 .

closed 2020-11-13 by traversaro