gbionics/idyntree #772: Document how to install idyntree from pip, and how to upload a new release to PyPI

traversaro, 2020-11-19:
@diegoferigo already shared with me this two piece of info, but I keep forgetting where I have them.

diegoferigo, 2020-11-19:
#### Installation

```bash
# Optional: create and enable first a virtualenv, then:
sudo apt install libxml2-dev libassimp-dev libeigen3-dev coinor-libipopt-dev swig
pip3 install idyntree
```

Note: if a virtualenv is enabled, you can also use `pip` and `python` instead of `pip3` and `python3`. They are symlinks.

#### Packaging

From the folder of the cloned repo:

```bash
# Create the source distribution
python3 setup.py sdist

# Install twine
pip3 install twine

# Upload to PyPI
twine upload dist/idyntree-<ver>.tar.gz
```

diegoferigo, 2020-11-20:
After https://github.com/robotology/idyntree/pull/774, the supported platforms where we distribute wheels no longer need to install the system dependencies since all the necessary files are included in the wheel.

Currently only linux distributions with python 3.8 are supported #774.