gbionics/idyntree #788: Fail to compile when MATLAB is ON on MacOS

lrapetti, 2020-12-02:
After updating `iDynTree` to the current `master` branch, the compilation with `USES_MATLAB` ON is failing as follow:
```
...
[100%] Linking CXX shared module ../../lib/iDynTreeMEX.mexmaci64
Undefined symbols for architecture x86_64:
  "_mexAtExit_800", referenced from:
      SWIG_Matlab_LoadModule() in iDynTreeMATLAB_wrap.cxx.o
  "_mexCallMATLABWithTrap_800", referenced from:
      SWIG_Matlab_NewPointerObj(void*, swig_type_info*, int) in iDynTreeMATLAB_wrap.cxx.o
  "_mexCallMATLAB_800", referenced from:
      SWIG_Matlab_isOwned(mxArray_tag*) in iDynTreeMATLAB_wrap.cxx.o
      SWIG_Matlab_ConvertPtrAndOwn(mxArray_tag*, void**, swig_type_info*, int, int*) in iDynTreeMATLAB_wrap.cxx.o
      SWIG_Matlab_NewPointerObj(void*, swig_type_info*, int) in iDynTreeMATLAB_wrap.cxx.o
      SWIG_Matlab_LoadModule() in iDynTreeMATLAB_wrap.cxx.o
      _mexFunction in iDynTreeMATLAB_wrap.cxx.o
  "_mexErrMsgIdAndTxt_800", referenced from:
      SWIG_Matlab_ConvertPtrAndOwn(mxArray_tag*, void**, swig_type_info*, int, int*) in iDynTreeMATLAB_wrap.cxx.o
      SWIG_Matlab_NewPointerObj(void*, swig_type_info*, int) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_MatlabSwigIterator_distance(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_MatlabSwigIterator_equal(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_StringVector_pop(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_StringVector_brace(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_StringVector_front(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      ...
  "_mexErrMsgTxt_800", referenced from:
      _mexFunction in iDynTreeMATLAB_wrap.cxx.o
  "_mexLock_800", referenced from:
      SWIG_Matlab_LoadModule() in iDynTreeMATLAB_wrap.cxx.o
  "_mexPrintf_800", referenced from:
      swig::SWIG_Matlab_streambuf::xsputn(char const*, long) in iDynTreeMATLAB_wrap.cxx.o
      swig::SWIG_Matlab_streambuf::overflow(int) in iDynTreeMATLAB_wrap.cxx.o
  "_mexWarnMsgIdAndTxt_800", referenced from:
      SWIG_Matlab_NewPointerObj(void*, swig_type_info*, int) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_MatrixDynSize_fromMatlab(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_SparseMatrixRowMajor_fromMatlab(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_SparseMatrixColMajor_fromMatlab(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_VectorDynSize_fromMatlab(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
  "_mxAddField_800", referenced from:
      SWIG_Matlab_LoadModule() in iDynTreeMATLAB_wrap.cxx.o
  "_mxCreateCellMatrix_800", referenced from:
      _wrap_LinksSolidShapesVector_pop(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_LinksSolidShapesVector_brace(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_LinksSolidShapesVector_front(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_LinksSolidShapesVector_back(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      swig::MatlabSwigIteratorOpen_T<std::__1::__wrap_iter<std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> >*>, std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> >, swig::from_oper<std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> > > >::value() const in iDynTreeMATLAB_wrap.cxx.o
      swig::MatlabSwigIteratorOpen_T<std::__1::reverse_iterator<std::__1::__wrap_iter<std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> >*> >, std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> >, swig::from_oper<std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> > > >::value() const in iDynTreeMATLAB_wrap.cxx.o
      swig::traits_from_stdseq<std::__1::vector<iDynTree::BerdySensor, std::__1::allocator<iDynTree::BerdySensor> >, iDynTree::BerdySensor>::from(std::__1::vector<iDynTree::BerdySensor, std::__1::allocator<iDynTree::BerdySensor> > const&) in iDynTreeMATLAB_wrap.cxx.o
      ...
  "_mxCreateCharArray_800", referenced from:
      _wrap_StringVector_pop(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_StringVector_brace(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_StringVector_front(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_StringVector_back(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_MatrixDynSize_toString(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_MatrixDynSize_display(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_SparseMatrixRowMajor_description__SWIG_0(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      ...
  "_mxCreateDoubleMatrix_800", referenced from:
      _wrap_MatrixDynSize_toMatlab(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_SparseMatrixRowMajor_toMatlabDense(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_SparseMatrixColMajor_toMatlabDense(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_VectorDynSize_toMatlab(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_Matrix1x6_toMatlab(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_Matrix2x3_toMatlab(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_Matrix3x3_toMatlab(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      ...
  "_mxCreateDoubleScalar_800", referenced from:
      _wrap_MatrixDynSize_paren__SWIG_0(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_MatrixDynSize_getVal(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_SparseMatrixRowMajor_paren__SWIG_0(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
[... 200 lines cut ...]
      ...
  "_mxIsStruct_800", referenced from:
      SWIG_Matlab_LoadModule() in iDynTreeMATLAB_wrap.cxx.o
  "_mxSetCell_800", referenced from:
      _wrap_LinksSolidShapesVector_pop(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_LinksSolidShapesVector_brace(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_LinksSolidShapesVector_front(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      _wrap_LinksSolidShapesVector_back(int, mxArray_tag**, int, mxArray_tag**) in iDynTreeMATLAB_wrap.cxx.o
      swig::MatlabSwigIteratorOpen_T<std::__1::__wrap_iter<std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> >*>, std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> >, swig::from_oper<std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> > > >::value() const in iDynTreeMATLAB_wrap.cxx.o
      swig::MatlabSwigIteratorOpen_T<std::__1::reverse_iterator<std::__1::__wrap_iter<std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> >*> >, std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> >, swig::from_oper<std::__1::vector<iDynTree::SolidShape*, std::__1::allocator<iDynTree::SolidShape*> > > >::value() const in iDynTreeMATLAB_wrap.cxx.o
      swig::traits_from_stdseq<std::__1::vector<iDynTree::BerdySensor, std::__1::allocator<iDynTree::BerdySensor> >, iDynTree::BerdySensor>::from(std::__1::vector<iDynTree::BerdySensor, std::__1::allocator<iDynTree::BerdySensor> > const&) in iDynTreeMATLAB_wrap.cxx.o
      ...
  "_mxSetFieldByNumber_800", referenced from:
      SWIG_Matlab_LoadModule() in iDynTreeMATLAB_wrap.cxx.o
  "_mxSetField_800", referenced from:
      SWIG_Matlab_LoadModule() in iDynTreeMATLAB_wrap.cxx.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/iDynTreeMEX.mexmaci64] Error 1
make[1]: *** [bindings/matlab/CMakeFiles/iDynTreeMEX.dir/all] Error 2
make: *** [all] Error 2

```

I did some tests to understand when the problem came out, and I observed that I was able to compile the code with the tag `v1.2.1` , but it is failing since the tag `v2.0.0`

traversaro, 2020-12-02:
> I did some tests to understand when the problem came out, and I observed that I was able to compile the code with the tag `v1.2.1` , but it is failing since the tag `v2.0.0`

The v2.0.0 was tagged after merging devel in master, I guess that you did never used the devel branch in the last year?

lrapetti, 2020-12-02:
> If you could give a try to [c268082](https://github.com/robotology/idyntree/commit/c2680825ba16fad9fcca2e20559085667609708c) and that works,

On this commit the cmake configuration for `iDynTree` is failing with
```
 $ cmake .
-- Found YARP: /Users/lorenzorapetti/Software/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.4.1+6-20201002.1+git8b157884f")
-- Found YARP: /Users/lorenzorapetti/Software/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.4.1+6-20201002.1+git8b157884f")
-- Found YARP: /Users/lorenzorapetti/Software/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.4.1+6-20201002.1+git8b157884f")
-- Found ICUB 1.17.0 (Components: iCubDev ctrlLib skinDynLib iKin iDyn learningMachine perceptiveModels actionPrimitives optimization)
CMake Error at cmake/FindMatlab.cmake:973 (message):
  [MATLAB] The MEX target name cannot be empty
Call Stack (most recent call first):
  bindings/matlab/CMakeLists.txt:27 (matlab_add_mex)
  bindings/matlab/CMakeLists.txt:94 (swig_compile_module_mex)

-- Configuring incomplete, errors occurred!
See also "/Users/lorenzorapetti/Software/robotology-superbuild/build/robotology/iDynTree/CMakeFiles/CMakeOutput.log".
```

traversaro, 2020-12-03:
I reproduced the same problem on Windows.

closed 2020-12-03 by traversaro

lrapetti, 2020-12-03:
Thanks @traversaro for fixing it, the problem is now solved, I was able to compile and use the bindings.