gbionics/idyntree #795: Using idyntree with Matlab

simeoncapy, 2020-12-10:
Hello,  
I followed the guidelines to install idyntree according the readme file. My goal is to use it with Matlab. I finished the *build* section with no errors. But then, I'm a bit lost.

I guess the begining of the *link* part is to link the iDynTree lib to my project for compiling. (Does it matter for Matlab?). I also created the `CMAKE_PREFIX_PATH` environment variable.

And then for the *bindings* part, I'm completely lost. I understood I have to activate the `IDYNTREE_USES_MATLAB` option in the `bindings\CMakeLists.txt` file. But then, I don't know what to do, should I build again? Moreover, I don't get where and what is the `<prefix>/mex` directory.

Thanks in advance for your help,
regards,
Siméon.

hu-yue, 2020-12-10:
@traversaro We actually managed to install everything on Windows 10, with the Matlab bindings enabled, and added the `mex` folder to Matlab path.
When our student here at GVLab tried to run the example code, though, this is the error he gets:

```
無効な MEX ファイル 'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64': 依存する共有ライブラリがありません:
'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64' で必要とされる 'idyntree-estimationd.dll'
'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64' で必要とされる 'idyntree-solid-shapesd.dll'
'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64' で必要とされる 'idyntree-inverse-kinematicsd.dll'
'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64' で必要とされる 'idyntree-visualizationd.dll'
'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64' で必要とされる 'idyntree-high-leveld.dll'
'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64' で必要とされる 'idyntree-modelio-urdfd.dll'
'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64' で必要とされる 'idyntree-sensorsd.dll'
'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64' で必要とされる 'idyntree-modeld.dll'
'C:\src\idyn\out\install\x64-Debug (default)\mex\iDynTreeMEX.mexw64' で必要とされる 'idyntree-cored.dll'。
```

Sorry for the Japanese, it basically says that he  `.dll` files cannot be found.
Thank you!

traversaro, 2020-12-10:
It seems that it is not finding the Debug iDynTree dll. I guess you need to add the `C:\src\idyn\out\install\x64-Debug (default)\bin` directory to the [PATH environment variable](https://en.wikipedia.org/wiki/PATH_(variable)#DOS,_OS/2,_and_Windows). You can do that with the [Rapid Environment Editor](https://www.rapidee.com/en/about), but you may also that just in a Command Prompt (actually I dot not know how to deal with the spaces in the path, pay attention to that):
~~~
set PATH=%PATH%;"C:\src\idyn\out\install\x64-Debug (default)\bin"
~~~
and then launch MATLAB. This may be better to avoid polluting your system with global modifications.

hu-yue, 2020-12-10:
We tried to add the path in the environment variable but it didn't work somehow.
We are also a bit confused about the situation.  
Maybe we can try with the editor you suggested and see again.

traversaro, 2020-12-10:
> We tried to add the path in the environment variable but it didn't work somehow.

How are you launching MATLAB? It is possible that it did not get the new version of the PATH. A good idea to check may be to print `getenv('PATH')`  in MATLAB, using https://www.mathworks.com/help/matlab/ref/getenv.html, to check if it has the expected value. Another possible thing that is going wrong is that the dll of dependencies (ipopt, libxml2) are the one that are not found, for debugging that you could use https://github.com/lucasg/Dependencies and load into it the `iDynTreeMEX.mexw64` .

hu-yue, 2020-12-10:
Now it is working. We removed and added again the environmental variables in `PATH`.
I guess we were doing something wrong with the environmental variables before.
Thanks for the hints!

traversaro, 2020-12-10:
> 
> 
> Now it is working. We removed and added again the environmental variables in `PATH`.
> I guess we were doing something wrong with the environmental variables before.
> Thanks for the hints!

Thanks, hopefully 2021 will bring binary installation also MATLAB bindings, let's see how that works out, this should remove a lot of hassle from the installation process. If you don't have other question, I will close the issue.

closed 2020-12-10 by traversaro