gbionics/idyntree #808: Find a way of running visualizer-tests in CI

traversaro, 2021-01-26:
I guess you need to set DISPLAY or something similar.

S-Dafarra, 2021-01-29:
I guess that the problem is the OpenGL version. In #806 the OpenGL version was
```
Irrlicht Engine version 1.8.4
 (Build 9200)
Using renderer: OpenGL 1.1.0
GDI Generic: Microsoft Corporation
OpenGL driver version is not 1.2 or better.
Failed to load OpenGL's multitexture extension, proceeding without.
Warning: OpenGL device only has one texture unit. Disabling multitexturing.
GLSL not available.
Resizing window (800 600)
```

On my PC OpenGL is 4.6.0

S-Dafarra, 2021-01-29:
Relevant issue https://github.community/t/i-need-opengl-version-2-0-whereas-windows-lastest-has-opengl-version-1-1-0/17810

S-Dafarra, 2024-03-25:
Related comment: https://github.com/conda-forge/idyntree-feedstock/pull/81#issuecomment-2018151627