gbionics/idyntree #814: Discuss the possibility to change the color palette of the iDynTree IRRLICHT visualizer

GiulioRomualdi, 2021-02-01:
I opened this issue out the possibility to change the color palette of the iDynTree visualizer. 

**Disclaimer: colors are really subjective. The colors that I like are probably terrible for you** :smile: 

During my latest work, I had the opportunity to play with the colors of the visualizer and I was able to mimic di aspect of meshcat. 

This is an example:

| Original | MeshCat like |
|:--------:|:------------:|
| [image] | [image] |

What do you think? How can we make visualizer more appealing?

traversaro, 2021-02-01:
I am afraid the new color is too cool and too 2020s for iDynTree. iDynTree vibe is much more "old stuff that works" that is reflected by the 1980s [TRON-like](https://en.wikipedia.org/wiki/Tron) current palette. :D 

Jokes apart, I think the new palette seems nice, it is totally ok for me to switch to it. Just to understand, is this issue about changing the default palette or also changing some parameters that can't be changed back by the user?

GiulioRomualdi, 2021-02-01:
>  Just to understand, is this issue about changing the default palette or also changing some parameters that can't be changed back by the user?

We may think to define different palets.
1. Meshcat
2. gazebo
3. ...

traversaro, 2021-02-01:
I definitely don't feel strong on this, so whatever you or other users of the visualizer prefer is fine for me.

S-Dafarra, 2021-02-01:
As far as I understood, the main interest is in changing the background and the ground grid color. They both are in the environment, so we could add a method like ``setTheme`` in ``IEnvironment`` with an enum or a string. Then, each tag will correspond to a specific set of colors. The naming then is up to who implements it :grin:

traversaro, 2021-02-01:
> As far as I understood, the main interest is in changing the background and the ground grid color. They both are in the environment, so we could add a method like `setTheme` in `IEnvironment` with [...]

Can't we just have a setBackgroundColor or setGroundGridColor or similar methods?

S-Dafarra, 2021-02-01:
> > As far as I understood, the main interest is in changing the background and the ground grid color. They both are in the environment, so we could add a method like `setTheme` in `IEnvironment` with an enum or a string. Then, each tag will correspond to a specific set of colors. The naming then is up to who implements it grin
> 
> Can't we just have a setBackgroundColor or setGroundGridColor or similar methods?

They are there already. The problem is that setting the RBGA values by hand is not very intuitive. I think that the idea behind this discussion is to have a no-brainer for quickly changing the both the ground grid and the background color.

traversaro, 2021-02-01:
Ok! Changing the default or adding theme are both ok options for me.

GiulioRomualdi, 2021-02-01:
> Ok! Changing the default or adding theme are both ok options for me.

We may do both of them :) The problem is which colors do you (@S-Dafarra  and @traversaro ) like?

traversaro, 2021-02-01:
Any palette is fine for me, the proposed meshcat one is great!

S-Dafarra, 2021-02-01:
> > Ok! Changing the default or adding theme are both ok options for me.
> 
> We may do both of them :) The problem is which colors do you (@S-Dafarra and @traversaro ) like?

I would keep the default as it is for continuity.