gbionics/idyntree #823: Things to do in iDynTree 4

traversaro, 2021-02-05:
https://github.com/robotology/idyntree/pull/822/commits/5a578123147858a83585f9f28629735d44427a3b

traversaro, 2021-07-09:
Anyone has any suggestion for breaking ABI change (but API compatible) or API deprecation that we should do in iDynTree 4? 

cc @robotology/iit-dynamic-interaction-control @Nicogene

Nicogene, 2021-07-09:
It is not an API/ABI breaking change, I would like to add the possibility to compute the reduced model passing the model and a lambda that defines the rule for defining the reduced model, maybe I can open a issue for this feature request

traversaro, 2021-07-09:
> 
> 
> It is not an API/ABI breaking change, I would like to add the possibility to compute the reduced model passing the model and a lambda that defines the rule for defining the reduced model, maybe I can open a issue for this feature request

Cool! Yes I think that can be done just by adding a function/method, so can be done without problems even after iDynTree 4 is released without the need to wait for a new major release.

traversaro, 2022-02-03:
> It is not an API/ABI breaking change, I would like to add the possibility to compute the reduced model passing the model and a lambda that defines the rule for defining the reduced model, maybe I can open a issue for this feature request

@Nicogene can you open a separate issue for this? This issue was solved by https://github.com/robotology/idyntree/pull/883, so I think we can close it, but if you are interested in that feature it is better to track it in a different issue.

closed 2022-02-03 by traversaro

Nicogene, 2022-02-03:
> @Nicogene can you open a separate issue for this? This issue was solved by https://github.com/robotology/idyntree/pull/883, so I think we can close it, but if you are interested in that feature it is better to track it in a different issue.

I did it, but I forgot to mention it :D See https://github.com/robotology/idyntree/issues/894