gbionics/idyntree #829: Add the foot_ft_sensor joints explicitly in the considered joints list even when removing the sensors from the URDF model

HosameldinMohamed, 2021-02-09:
When running `wholeBodyDyanmics` while removing the feet FT sensors from the URDF model. iDynTree considers only the non-fixed joints when loading the reduced model. See
https://github.com/robotology/idyntree/blob/master/src/estimation/src/ExtWrenchesAndJointTorquesEstimator.cpp#L151

As a result, the links `l_foot` and `r_foot` for example can't be found and we can't publish the estimated feet contact wrenches in those links.

In this [fork](https://github.com/HosameldinMohamed/idyntree/commit/04302a9ba52362d3b4d91060545cb33e69aa837d) the joints `l_foot_ft_sensor` and `r_foot_ft_sensor` are added explicitly to the reduced model which solves the issue.

The modification seem too specific though, maybe there's a way to make it more generic (doesn't depend on a certain joint name)?

CC @traversaro

traversaro, 2021-02-09:
Despite its name, the `consideredDOFs` argument of the `loadModelAndSensorsFromFileWithSpecifiedDOFs` method can also be used to pass the names of fixed joints that should be considered, and that is even considered as there is some logic to ensure that there are no problem if a fixed joint that correspond to an FT is added, as there is a logic to avoid that in that case the vector contains a duplicate element, see https://github.com/robotology/idyntree/blob/master/src/estimation/src/ExtWrenchesAndJointTorquesEstimator.cpp#L154 .

Having said that, from the iDynTree point of view you just need to add `l_foot_ft_sensor` and `r_foot_ft_sensor` to the `consideredDOFs` parameter. and the problem would be solved. If I recall correctly however you can't just add those joints to the `axesNames` parameter of wholeBodyDynamics, because that parameter is used both to set the consideredDOFs argument of `loadModelAndSensorsFromFileWithSpecifiedDOFs`, but also to specific which joints should be read from the YARP network, and clearly you can't read the joint position for a fixed joints. Probably the easiest fix is to add an additional parameters `additionalConsideredJoints` in wholebodydynamics (that by default would be an empty list, and in your case would contain `l_foot_ft_sensor` and `r_foot_ft_sensor`), that it would be append to joints already specified in axesNames so that in https://github.com/robotology/whole-body-estimators/blob/cc8ffb83375a2d410b225f4fa67aee4a29074b42/devices/wholeBodyDynamics/WholeBodyDynamicsDevice.cpp#L280 you pass the list of joints that you want, without interfering with the list of joints that are read and controller via YARP.

HosameldinMohamed, 2021-02-09:
Thanks! This is indeed better.

> Probably the easiest fix is to add an additional parameters additionalConsideredJoints in wholebodydynamics (that by default would be an empty list, and in your case would contain l_foot_ft_sensor and r_foot_ft_sensor),

Since there's nothing to do from iDynTree point of view, I close this issue and propose the suggested modifications in [`whole-body-estimators`](https://github.com/robotology/whole-body-estimators).

closed 2021-02-09 by HosameldinMohamed