gbionics/idyntree #846: Fix numbering of pre-release wheels

diegoferigo, 2021-04-06:
[Pre-release wheels](https://pypi.org/project/idyntree/#history) (built from `devel`) have the wrong `2.x` numbering, instead of `3.0.2.devXXX` (up to today).

Merging `master` into `devel` should suffice to propagate the latest tag to the devel branch. Let's remember that the version is computed automatically from the git repository in:

https://github.com/robotology/idyntree/blob/459f1cee58b2f9cd23772cf8f3e92370e38161a2/setup.py#L36

cc @traversaro

traversaro, 2021-04-06:
I just merged master in devel, is this sufficient to solve the issue?

diegoferigo, 2021-04-06:
Thanks! It should, let's wait [CI outcome](https://github.com/robotology/idyntree/actions/runs/723041637).

diegoferigo, 2021-04-06:
[`sdist`](https://github.com/robotology/idyntree/runs/2279660416?check_suite_focus=true#step:6:22) worked fine, I think we can close the issue:

> [...]
> creating idyntree-3.0.2.dev6
> [...]

closed 2021-04-06 by diegoferigo