gbionics/idyntree #865: iDynTree and CasADi MATLAB bindings install different files with the same name in the MATLAB path

traversaro, 2021-05-11:
iDynTree install the following files in the MATLAB path:
* `SwigGet.m`
* `SwigMem.m`
* `SwigRef.m`

Differently from the rest of MATLAB files that are installed as part of the project-specific `+iDynTree` namespace, these three files are not project specific, and so are installed with the same name by other projects, such as CasADi. This would not be a problem if the files installed by the different projects were exactly the same. Unfortunately, for iDynTree/YARP And CasADi this is not the case, as iDynTree/YARP  have a modified version of `SwigRef.m` since https://github.com/robotology-dependencies/swig/pull/2 . As it is not simple to make sure that the CasADi version of `SwigRef.m` does not shadows the one of iDynTree introducing subtle bugs, probably the easier option is to ensure that `iDynTree` creates a `iDynTreeSwigRef.m` file, to avoid any conflict.

CarlottaSartore, 2021-05-12:
I have done `which SwigRef.m` in my MATLAB env and I have got the following output
```
>> which SwigRef.m
/home/csartore/personal_ws/Casadi/casadi_install/casadi/SwigRef.m  % SwigRef constructor
```

So I would say I am using the Casadi one rather than the iDynTree. 

If I remove casadi from the matlab path I got instead the following output 
```
>> which SwigRef.m
/home/csartore/iit_ws/robotology-superbuild/build/install/mex/SwigRef.m  % SwigRef constructor
```

I have tried also to remove from the Path both Casadi and the mex folder of the robotology. I have then added back them to the path first Casadi and then mex and I have got the following output 

```
>> which SwigRef.m
/home/csartore/iit_ws/robotology-superbuild/build/install/mex/SwigRef.m  % SwigRef constructor
```

So I would say that in my configuration I use the Casadi SwigRef.m because I add to the path Casadi after robotology (Casadi is indeed automatically added to the path when I launch a Simulink model that depends on casadi )

@traversaro  Let me know if you need any other information form my environment!

traversaro, 2021-05-14:
Apparently, exactly this bug caused problem recently.

traversaro, 2021-05-19:
Fixed by https://github.com/robotology/idyntree/pull/868 and released in 3.2.0 for iDynTree .

Fixed by https://github.com/robotology/yarp-matlab-bindings/pull/65 and released in 3.4.4 fo yarp-matlab-bindings .

closed 2021-05-19 by traversaro