gbionics/idyntree #867: Unable to compile pybind11 bindings

GiulioRomualdi, 2021-05-12:
When I try to configure the cmake project with `IDYNTREE_USES_PYTHON_PYBIND11` enabled I get the following error
 

```
CMake Error at /usr/share/cmake-3.16/Modules/FindPython/Support.cmake:2341 (add_library):
   Cannot find source file:

     SYSTEM

   Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
   .hpp .hxx .in .txx
 Call Stack (most recent call first):
   /usr/share/cmake-3.16/Modules/FindPython3.cmake:304 (__Python3_add_library)

 /home/gromualdi/.local/lib/python3.8/site-packages/pybind11/share/cmake/pybind11/pybind11NewTools.cmak
 e:196 (python3_add_library)
   bindings/pybind11/CMakeLists.txt:1 (pybind11_add_module)

 CMake Error at /usr/share/cmake-3.16/Modules/FindPython/Support.cmake:2341 (add_library):
   No SOURCES given to target: pybind11_idyntree
 Call Stack (most recent call first):
   /usr/share/cmake-3.16/Modules/FindPython3.cmake:304 (__Python3_add_library)

 /home/gromualdi/.local/lib/python3.8/site-packages/pybind11/share/cmake/pybind11/pybind11NewTools.cmak
 e:196 (python3_add_library)
   bindings/pybind11/CMakeLists.txt:1 (pybind11_add_module)
```

`pip3 show pybind11` gives me the following information
```
Name: pybind11
Version: 2.6.2
Summary: Seamless operability between C++11 and Python
Home-page: https://github.com/pybind/pybind11
Author: Wenzel Jakob
Author-email: wenzel.jakob@epfl.ch
License: BSD
Location: /home/gromualdi/.local/lib/python3.8/site-packages
Requires: 
Required-by: 
```

traversaro, 2021-05-12:
Related issue: https://github.com/robotology/idyntree/issues/782 .

traversaro, 2021-05-12:
In any case, there is a specific reason for which you installed pybind11 via pip? In https://github.com/robotology/idyntree/issues/782, I think we can aim to support pybind11 installed via apt or the one installed via conda, but the mixed apt/pip installation seems to complicate to support effectively.

traversaro, 2021-11-07:
@GiulioRomualdi is this issue still present?

GiulioRomualdi, 2021-11-07:
Actually no, but I installed pybind11 from apt

traversaro, 2021-11-07:
> Actually no, but I installed pybind11 from apt

Ack, updated the title.