gbionics/idyntree #878: getRelativeTransform() not returning correct homogeneous transformation

nicholaspalomo, 2021-06-12:
Hello,

First off, thank you for your amazing work on iDynTree. I'm using it in a project right now and find it very cool.

I created a ROS2 project to command the Panda robot to different target locations. I'm using the IK solve to calculate joint angle targets, given an end effector target. The IK functionality works perfectly. Here is an example where I specified a setpoint of [x, y, z] = [0.4, 0.0, 0.8] and [r, p, yaw] = [0.0, 0.0, 0.0] for the manipulator. The arm goes to the specified location (or as close to it as possible), as expected.
[image]

However, I then periodically compose a ROS Odometry message (containing pose + twist information for the end effector) given the joint positions and joint velocities provided by ROS. The joint positions are given in radians and the velocities in radians / second. In the image above, the red-green-blue axes represent the target location at [0.4, 0.0, 0.8]. You can see on the left that the actual location of the end effector frame is at approximately [0.427, 0.057, 0.763], which is close enough. **However**, the purple arrow in the figure represents the end effector pose using values computed with forward kinematics and iDynTree. You can see that something here is quite wrong, as it does not agree with the ground truth values displayed in RViz.

I largely followed the code of one of your tutorials (`iDynTreeExampleInverseKinematics.cpp`). Here's how I'm trying to get the transform between the base frame and the end effector frame.

In the constructor of my class I create a `KinDynComputations` object:
```
self._fk = KinDynComputations()
self._fk.setFrameVelocityRepresentation(idt.INERTIAL_FIXED_REPRESENTATION)
self._fk.loadRobotModel(self._articulated_system)
```
where `self._articulated_system` is the robot model loaded with the `ModelLoader` class.

Then, directly after receiving the joint positions and joint velocities directly from ROS, I do:
```
self._fk.setJointPos(numpy.FromNumPy.to_idyntree_dyn_vector(array=np.array(joint_positions)))
end_effector_pose_in_base_frame = self._fk.getRelativeTransform(self.base_frame, self.end_effector_frame)
end_effector_position_in_base_frame = end_effector_pose_in_base_frame.getPosition().toNumPy()
end_effector_orientation_in_base_frame = end_effector_pose_in_base_frame.getRotation().toNumPy()
```
Where `base_frame` is `panda_link0` and `end_effector_frame` is `end_effector_frame` from the URDF, which I borrowed directly from your `gym-ignition` repo. Then I publish the information from this transform to a topic so I can visualize it in RViz.

My question is - am I missing some step here in using iDynTree for computing the forward kinematics for my robot, assuming that `getRelativeTransform` returns the correct result?

Many thanks!

traversaro, 2021-06-12:
Hi @nicholaspalomo, do you have a self-contained example to check? Preliminary, I guess something it may be worth to check is if the serialization you assume is the same used internally in iDynTree.

nicholaspalomo, 2021-06-12:
> Hi @nicholaspalomo, do you have a self-contained example to check? Also a link to an existing repo with the code you are mentioning would be helpful to check. Preliminarly, I guess something it may be worth to check is if the serialization you assume is the same used internally in iDynTree.

Hey, @traversaro. Thanks for the reply! Yes, my repo (and the instructions to run it) are here: https://github.com/nicholaspalomo/panda_ros2_gazebo. The snippet I posted in the issue comes from here:
https://github.com/nicholaspalomo/panda_ros2_gazebo/blob/master/panda_ros2_gazebo/scripts/models/panda.py#L138

I'll double check the joint serialization. My assumption was that I have it correct since the inverse kinematics seems to be working . Thanks for the suggestion.

traversaro, 2021-06-13:
Thanks for the detailed comment, I will check it. Minor comment: if you add new content to an issue, it is convenient to add at least a new comment, otherwise no one gets notifications for edits to old comments.

nicholaspalomo, 2021-06-13:
> Minor comment: if you add new content to an issue, it is convenient to add at least a new comment, otherwise no one gets notifications for edits to old comments.

Sure, will do. Thanks!

traversaro, 2021-06-14:
Thanks for providing the links! I looked into that and I could not find anything strange, however if you want to debug a bit more I suggest to try to compare the IK results and the FK results directly from the Python scripts without passing through the conversions and the RViz visualization, perhaps this could highlight what is going wrong, either at the iDynTree level or in the conversions.

nicholaspalomo, 2021-06-14:
Hey, @traversaro, I just figured it out and [pushed a commit a few seconds ago](https://github.com/nicholaspalomo/panda_ros2_gazebo/commit/bd633c0160534ee62567f7490a48588a8602e0ee)!

The problem was on my side.

Even though I specified in my [ROS control YAML](https://github.com/nicholaspalomo/panda_ros2_gazebo/blob/master/config/ros_control.yaml) file the same joint ordering as that provided by iDynTree, I noticed that when subscribing to the `/joint_states` topic, ROS rearranges the joint names and the joint measurements for some reason.

So, even though I specify the joints in this order:
```
- panda_joint1
- panda_joint2
- panda_joint3
- panda_joint4
- panda_joint5
- panda_joint6
- panda_joint7
- panda_finger_joint1
- panda_finger_joint2
```

ROS returns the measurements in this order:
```
- panda_finger_joint1
- panda_joint2
- panda_joint5
- panda_joint4
- panda_joint6
- panda_joint7
- panda_joint1
- panda_finger_joint2
- panda_joint3
```

Kind of perplexing, but anyway, I'm glad it's working now as intended.

Thank you very much again for the discussion and thanks for developing this great library. I will go ahead and close this issue.

closed 2021-06-14 by nicholaspalomo

traversaro, 2021-06-14:
Glad you figured this out! This kind of issues is exactly why we developed all the `ModelLoader::loadReducedModel***` methods, in which you can explicitly specify the joint serialization you want to use. Our typical workflow is that we only have a single source of truth for the serialization used, and all the software is based on that.

nicholaspalomo, 2021-06-14:
> Glad you figured this out! This kind of issues is exactly why we developed all the `ModelLoader::loadReducedModel***` methods, in which you can explicitly specify the joint serialization you want to [...]

Ah, ok. That makes sense. Thanks for the insight!