gbionics/idyntree #880: Some question for using in Python

MingshanHe, 2021-06-23:
I have successfully compile it as Cmake 5.18 and install it. But I can't use it in my python3.5 (Ubuntu 16.04), I have already move the package in the python3.5 site-packages and python3 dist-packages but not work. Would you like to help me? I will apreciate. Thanks

traversaro, 2021-06-23:
Hi @MingshanHe ! Can you provide more details on the exact command you used to compile iDynTree, which Python scripts are you trying to run that is not working, which version of iDynTree did you used and what do you mean by "I have already move the package in the python3.5 site-packages and python3 dist-packages but not work". Note that Ubuntu 16.04 is not tested anymore in CI, so it is possible that there are problems. 

However, if compilation works fine, it should be sufficient to appendthe `PYTHONPATH` env variable the directory `<install_prefix>/lib/python3.5/dist-packages`, and then you should be able to run an example such as https://github.com/robotology/idyntree/blob/master/examples/python/KinDynComputationsTutorial.py .

Note that furthermore idyntree is already available ready to use in conda and pip, and if that is ok for you you can install it via:
~~~
conda install -c conda-forge -c robotology idyntree
~~~

or:
~~~
pip install idyntree
~~~

MingshanHe, 2021-06-23:
I have download the it in Github which is the master branch(I think it is the lastest version) and compile it in the tutorial by cmake. It seems that I have compiled it successfully but I haven't the directory `<install_prefix>/lib/python3.5/dist-packages` but have the site-packages directory in python3.5, I want to know if it is possible for run the append command it any work directory in terminal? and How to solve the directory. Thanks.

traversaro, 2021-06-23:
> but have the site-packages directory in python3.5, 

What do you mean? Can you report the content of `<build_directory>/install_manifest.txt` so that it is clear which files are installed in your case?

MingshanHe, 2021-06-24:
`/usr/local/lib/libidyntree-core.so
/usr/local/include/iDynTree/Core/Axis.h
/usr/local/include/iDynTree/Core/ArticulatedBodyInertia.h
/usr/local/include/iDynTree/Core/ClassicalAcc.h
/usr/local/include/iDynTree/Core/Direction.h
/usr/local/include/iDynTree/Core/EigenSparseHelpers.h
/usr/local/include/iDynTree/Core/EigenMathHelpers.h
/usr/local/include/iDynTree/Core/EigenHelpers.h
/usr/local/include/iDynTree/Core/InertiaNonLinearParametrization.h
/usr/local/include/iDynTree/Core/MatrixDynSize.h
/usr/local/include/iDynTree/Core/MatrixFixSize.h
/usr/local/include/iDynTree/Core/Position.h
/usr/local/include/iDynTree/Core/PositionRaw.h
/usr/local/include/iDynTree/Core/Rotation.h
/usr/local/include/iDynTree/Core/RotationRaw.h
/usr/local/include/iDynTree/Core/RotationalInertiaRaw.h
/usr/local/include/iDynTree/Core/SpatialAcc.h
/usr/local/include/iDynTree/Core/SpatialForceVector.h
/usr/local/include/iDynTree/Core/SpatialInertiaRaw.h
/usr/local/include/iDynTree/Core/SpatialInertia.h
/usr/local/include/iDynTree/Core/SpatialMomentum.h
/usr/local/include/iDynTree/Core/SpatialMotionVector.h
/usr/local/include/iDynTree/Core/TestUtils.h
/usr/local/include/iDynTree/Core/Transform.h
/usr/local/include/iDynTree/Core/TransformDerivative.h
/usr/local/include/iDynTree/Core/Twist.h
/usr/local/include/iDynTree/Core/Utils.h
/usr/local/include/iDynTree/Core/VectorFixSize.h
/usr/local/include/iDynTree/Core/VectorDynSize.h
/usr/local/include/iDynTree/Core/Wrench.h
/usr/local/include/iDynTree/Core/PrivateUtils.h
/usr/local/include/iDynTree/Core/PrivatePreProcessorUtils.h
/usr/local/include/iDynTree/Core/GeomVector3.h
/usr/local/include/iDynTree/Core/SpatialVector.h
/usr/local/include/iDynTree/Core/SparseMatrix.h
/usr/local/include/iDynTree/Core/Triplets.h
/usr/local/include/iDynTree/Core/CubicSpline.h
/usr/local/include/iDynTree/Core/Span.h
/usr/local/include/iDynTree/Core/SO3Utils.h
/usr/local/include/iDynTree/Core/MatrixView.h
/usr/local/include/iDynTree/Core/AngularForceVector3.h
/usr/local/include/iDynTree/Core/AngularMotionVector3.h
/usr/local/include/iDynTree/Core/ForceVector3.h
/usr/local/include/iDynTree/Core/LinearForceVector3.h
/usr/local/include/iDynTree/Core/LinearMotionVector3.h
/usr/local/include/iDynTree/Core/MotionVector3.h
/usr/local/include/iDynTree/Core/CoreExport.h
/usr/local/lib/libidyntree-model.so
/usr/local/include/iDynTree/Model/ContactWrench.h
/usr/local/include/iDynTree/Model/DenavitHartenberg.h
/usr/local/include/iDynTree/Model/FixedJoint.h
/usr/local/include/iDynTree/Model/ForwardKinematics.h
/usr/local/include/iDynTree/Model/FreeFloatingState.h
/usr/local/include/iDynTree/Model/FreeFloatingMatrices.h
/usr/local/include/iDynTree/Model/IJoint.h
/usr/local/include/iDynTree/Model/Dynamics.h
/usr/local/include/iDynTree/Model/DynamicsLinearization.h
/usr/local/include/iDynTree/Model/DynamicsLinearizationHelpers.h
/usr/local/include/iDynTree/Model/Indices.h
/usr/local/include/iDynTree/Model/Jacobians.h
/usr/local/include/iDynTree/Model/JointState.h
/usr/local/include/iDynTree/Model/LinkTraversalsCache.h
/usr/local/include/iDynTree/Model/Link.h
/usr/local/include/iDynTree/Model/LinkState.h
/usr/local/include/iDynTree/Model/Model.h
/usr/local/include/iDynTree/Model/ModelTransformers.h
/usr/local/include/iDynTree/Model/MovableJointImpl.h
/usr/local/include/iDynTree/Model/RevoluteJoint.h
/usr/local/include/iDynTree/Model/PrismaticJoint.h
/usr/local/include/iDynTree/Model/SolidShapes.h
[... 74 lines cut ...]
/usr/local/include/iDynTree/Integrators/RK4.h
/usr/local/include/iDynTree/Integrators/ImplicitTrapezoidal.h
/usr/local/include/iDynTree/Integrators/ForwardEuler.h
/usr/local/include/iDynTree/Optimizers/IpoptInterface.h
/usr/local/include/iDynTree/Optimizers/OsqpInterface.h
/usr/local/include/iDynTree/Optimizers/AlglibInterface.h
/usr/local/include/iDynTree/Optimizers/WorhpInterface.h
/usr/local/include/iDynTree/OCSolvers/MultipleShootingSolver.h
/usr/local/include/iDynTree/yarp/YARPConversions.h
/usr/local/include/iDynTree/yarp/YARPEigenConversions.h
/usr/local/include/iDynTree/yarp/YARPConfigurationsLoader.h
/usr/local/include/iDynTree/yarp/YARPConversionsImplementation.h
/usr/local/include/iDynTree/yarp/YARPConfigurationsLoaderImplementation.h
/usr/local/include/iDynTree/iKinConversions.h
/usr/local/include/iDynTree/iKinConversionsImplementation.h
/usr/local/include/iDynTree/skinDynLibConversions.h
/usr/local/include/iDynTree/skinDynLibConversionsImplementation.h
/usr/local/lib/libidyntree-visualization.so
/usr/local/include/iDynTree/Visualizer.h
/usr/local/bin/idyntree-model-info
/usr/local/lib/cmake/iDynTree/iDynTreeConfigVersion.cmake
/usr/local/lib/cmake/iDynTree/iDynTreeConfig.cmake
/usr/local/lib/cmake/iDynTree/iDynTreeTargets.cmake
/usr/local/lib/cmake/iDynTree/iDynTreeTargets-release.cmake`
Above is my `<build_directory>/install_manifest.txt`, and you have mentioned that append the `PYTHON` env variable the directory for `python3.5/dist-packages` but in my PC I have not found out this directory but the `site-packages`.

traversaro, 2021-06-24:
> and you have mentioned that append the `PYTHON` env variable the directory for `python3.5/dist-packages` but in my PC I have not found out this directory but the `site-packages`.

The env variables is `PYTHONPATH`, not `PYTHON`, see https://docs.python.org/3/using/cmdline.html#envvar-PYTHONPATH . However, if you are installing in `/usr/local` (that is tipically discouraged unless you are installing in a Docker image or in any case not in your dev system), you should not need to modify `PYTHONPATH`. However, the problem here seems to be that the Python bindings are not compiled at all. You need to install `python3-dev` `python3-numpy` and `swig` apt packages, and then enabled the CMake option `IDYNTREE_USES_PYTHON` . However, note that we don't actively test on Ubuntu 16.04 anymore, so it is possible that the swig version provided in Ubuntu 16.04 is too old or to encounter other incompatibilities.

traversaro, 2021-07-08:
Hi @MingshanHe , were you able to solve your problems?

MingshanHe, 2021-07-08:
Yes, I think it is possible to close this issue

closed 2021-07-08 by traversaro