gbionics/idyntree #908: Model: add getName() method

Nicogene, 2021-07-22:
It would be nice to have a `getName()` method for getting the name stored in the `<robot name= "foo">` tag of the urdf

traversaro, 2021-07-22:
Probably it is sufficient to add a `name` attribute with `getName` `setName` accessors in https://github.com/robotology/idyntree/blob/a52e9240df154bcc752f438b73a67baabbebaaf9/src/model/include/iDynTree/Model/Model.h#L151 , and then use the `setName` in https://github.com/robotology/idyntree/blob/db312bfe9f490f991250603f05cea84a8c156899/src/model_io/urdf/src/RobotElement.cpp, as done for example for the `link`  tag in https://github.com/robotology/idyntree/blob/db312bfe9f490f991250603f05cea84a8c156899/src/model_io/urdf/src/LinkElement.cpp#L45 .

traversaro, 2023-04-08:
I guess @Nicogene @mfussi66 you could be soon hitting this.

traversaro, 2023-04-13:
Actually, at least for exporting URDF it is possible to use  `ModelExporterOptions::robotExportedName`.