gbionics/idyntree #926: iDynTree Irrlicht visualizer does not load TALOS model properly

GiulioRomualdi, 2021-10-31:
I try to visualize the model with the following command
```bash
idyntree-model-view -m $CONDA_PREFIX/share/example-robot-data/robots/talos_data/robots/talos_reduced.urdf
```
I got

[image]

On the other hand, if I load the model using `iDynTree-Meshcat` visualizer (after applying #925) the model is loaded correctly

[image]

<br/>
To replicate the error you can use the following conda environment
 (<code>conda list</code>):
<details>

```
conda list
# packages in environment at /home/gromualdi/mambaforge/envs/idyntree-env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                      1_llvm    conda-forge
ampl-mp                   3.1.0             h2cc385e_1005    conda-forge
argon2-cffi               21.1.0           py39h3811e60_0    conda-forge
assimp                    5.0.1                hedfc422_6    conda-forge
async_generator           1.10                       py_0    conda-forge
attrs                     21.2.0             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bleach                    4.1.0              pyhd8ed1ab_0    conda-forge
boost                     1.74.0           py39h5472131_3    conda-forge
boost-cpp                 1.74.0               h359cf19_5    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
cffi                      1.14.6           py39h4bc2ebd_1    conda-forge
console_bridge            1.0.1                h4bd325d_0    conda-forge
debugpy                   1.4.1            py39he80948d_0    conda-forge
decorator                 5.1.0              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
eigenpy                   2.6.9            py39h5aed9d1_0    conda-forge
entrypoints               0.3             py39hde42818_1002    conda-forge
example-robot-data        3.11.0           py39h1a9c180_0    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
hpp-fcl                   1.7.8            py39hbcdfc36_0    conda-forge
icu                       69.1                 h9c3ff4c_0    conda-forge
idyntree                  4.2.0            py39hf4c3102_0    conda-forge
importlib-metadata        4.8.1            py39hf3d152e_0    conda-forge
ipopt                     3.14.4               h7ede334_0    conda-forge
ipykernel                 6.4.2            py39hef51801_0    conda-forge
ipython                   7.29.0           py39hef51801_0    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
ipywidgets                7.6.5              pyhd8ed1ab_0    conda-forge
irrlicht                  1.8.4                hdaeae82_1    conda-forge
jbig                      2.1               h7f98852_2003    conda-forge
jedi                      0.18.0           py39hf3d152e_2    conda-forge
jinja2                    3.0.2              pyhd8ed1ab_0    conda-forge
jpeg                      9d                   h36c2ea0_0    conda-forge
jsonschema                4.1.2              pyhd8ed1ab_0    conda-forge
jupyter_client            7.0.6              pyhd8ed1ab_0    conda-forge
jupyter_core              4.9.1            py39hf3d152e_0    conda-forge
jupyterlab_pygments       0.1.2              pyh9f0ad1d_0    conda-forge
jupyterlab_widgets        1.0.2              pyhd8ed1ab_0    conda-forge
lcms2                     2.12                 hddcbb42_0    conda-forge
ld_impl_linux-64          2.36.1               hea4e1c9_2    conda-forge
lerc                      3.0                  h9c3ff4c_0    conda-forge
libblas                   3.9.0           12_linux64_openblas    conda-forge
[... 112 lines cut ...]
import os

def get_model_path(robot_name='talos_reduced'):
    conda_prefix = os.environ["CONDA_PREFIX"]
    robots_folder = os.path.join("share","example-robot-data","robots","talos_data","robots")
    
    return os.path.join(conda_prefix, robots_folder, robot_name +".urdf")

def get_joint_list():

    joint_list = ["torso_1_joint", "torso_2_joint",
                  "leg_left_1_joint", "leg_left_2_joint", "leg_left_3_joint", "leg_left_4_joint", "leg_left_5_joint", "leg_left_6_joint",
                  "leg_right_1_joint", "leg_right_2_joint", "leg_right_3_joint", "leg_right_4_joint", "leg_right_5_joint", "leg_right_6_joint",
                  "arm_left_1_joint", "arm_left_2_joint", "arm_left_3_joint", "arm_left_4_joint", "arm_left_5_joint", "arm_left_6_joint", "arm_left_7_joint",
                  "arm_right_1_joint", "arm_right_2_joint", "arm_right_3_joint", "arm_right_4_joint", "arm_right_5_joint", "arm_right_6_joint", "arm_right_7_joint"]

    return joint_list

if __name__ == "__main__":
   model_path = get_model_path()
   joint_list = get_joint_list()
   viz = MeshcatVisualizer()
   viz.set_model_from_file(model_path, joint_list)
   viz.load_model(color=[1, 1, 1, 0.8])
```

traversaro, 2021-10-31:
My guess is that the irrlicht visualizer is not taking properly taking into account the [`iDynTree::SolidShape::getLink_H_geometry`](https://github.com/robotology/idyntree/blob/db312bfe9f490f991250603f05cea84a8c156899/src/model/include/iDynTree/Model/SolidShapes.h#L76) data of visualization shapes.

traversaro, 2021-10-31:
> My guess is that the irrlicht visualizer is not taking properly taking into account the [...]

However, I quickly checked in https://github.com/Gepetto/example-robot-data/blob/v3.11.0/robots/talos_data/robots/talos_reduced.urdf and it seems that all the visual geometry elements have the form:
~~~
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
~~~

So this may not be the case.