gbionics/idyntree #960: Changing the Order of Link / Joint Indexing

Woolfrey, 2022-01-26:
I've loaded a model of the iCub3 in to iDynTree from `~/icub-models/iCub/robots/iCubGazeboV3/model.urdf`. When I print all the joint names they don't appear to be in a consistent order:
```
l_hip_pitch
torso_roll
r_hip_pitch
r_hip_roll
r_hip_yaw
r_knee
r_ankle_pitch
r_ankle_roll
torso_pitch
torso_yaw
r_shoulder_pitch
l_shoulder_pitch
neck_pitch
neck_roll
neck_yaw
l_shoulder_roll
l_shoulder_yaw
l_elbow
l_wrist_prosup
l_wrist_pitch
l_wrist_yaw
r_shoulder_roll
r_shoulder_yaw
r_elbow
r_wrist_prosup
r_wrist_pitch
r_wrist_yaw
l_hip_roll
l_hip_yaw
l_knee
l_ankle_pitch
l_ankle_roll
r_leg_ft_sensor
r_foot_rear_ft_sensor
r_foot_front_ft_sensor
neck_fixed_joint
l_arm_ft_sensor
r_arm_ft_sensor
l_leg_ft_sensor
l_foot_front_ft_sensor
l_foot_rear_ft_sensor
```
It starts with a left hip joint, then a torso joint, then a right hip joint. If I want to index through all the torso joints, for example, I'd have to code it manually. If they were listed in sequence it would be easier to access them in `for` loops.

Is there a way to change the order so that all joints of a single limb are grouped together?

traversaro, 2022-01-26:
We had some similar discussion back in time, as soon as I am on a PC I can link them. Just to understand, how do you define limbs in case of general model?

traversaro, 2022-01-26:
Anyhow iDynTree provides support for specifying any kind of joint order you like, and that is the way most people use it. I am on a phone, but I guess you can ask to @lrapetti or anyone else in AMI on how to do so.

Woolfrey, 2022-01-26:
I'm using the limb definitions in the iCub kinematics https://icub-tech-iit.github.io/documentation/icub_kinematics/icub-forward-kinematics/icub-forward-kinematics/, which I believe match the yarp ports of the icubSim.

lrapetti, 2022-01-26:
As @traversaro was mentioning, `iDynTree` loads models from `urdf` without specific ordering, since there is not a natural way to assign an order to tree structures.
Said that there are two way of proceeding, depending on what's your need:
- You can load the model with the [`loadReducedModel `](https://robotology.github.io/idyntree/classiDynTree_1_1ModelLoader.html#ab846e7becb8c98998cb2b05749a95ade) methods. This method accepts as an argument the list of selected joints, and the order of the joints in this list will then be used for ordering the joints. (Please read also the comment from @traversaro linked below regarding this point)
- retrieving the jointIndex trough the jointName using [`getJointIndex()`](https://robotology.github.io/idyntree/classiDynTree_1_1Model.html#a9e586e10ab288a30401b767c160181e9) method. In this way, you are independent from the ordering of the joints and you can directly iterate on the jointNames in the order that you prefer. (Note that I think most of the methods can accept as argument directly the joint name, so you don't even need to pass through the joint index)

Please have a look also at the explanation by @traversaro in https://github.com/robotology/human-dynamics-estimation/issues/222#issuecomment-734396124.

traversaro, 2022-01-26:
If you need a complete example, you can check also https://github.com/vvv-school/tutorial_impedance-control .

Woolfrey, 2022-01-26:
@traversaro , @lrapetti  thanks for the information! I assumed the way it generated the model was arbitrary, just wasn't sure if there was an option to re-order it manually. I will have a read of the links you sent me.

closed 2022-01-26 by Woolfrey

traversaro, 2022-01-27:
> If you need a complete example, you can check also https://github.com/vvv-school/tutorial_impedance-control .

In particular, at the YARP level the use of `remotecontrolboardremapper` YARP device permits in a lot of cases to avoid the need to have to deal with separate limbs at all.

Woolfrey, 2022-01-28:
Hi @traversaro, I saw this! Thanks again.

It could still be useful to group joints so they can be indexed through easier, even with the $remotecontrolboardremapper$.

traversaro, 2022-01-28:
> It could still be useful to group joints so they can be indexed through easier, even with the $remotecontrolboardremapper$.

And you can do that by passing a serialization in which the joint are grouped are you preferred, right?

Woolfrey, 2022-01-28:
> > It could still be useful to group joints so they can be indexed through easier, even with the $remotecontrolboardremapper$.
> 
> And you can do that by passing a serialization in which the joint are grouped are you preferred, right?

Yes, that's right. It looks like a combination of `loadReducedModel` and `remotecontrolboardremapper` will do what I need.