gbionics/idyntree #965: Visualizer::drawToFile method saves a black image

traversaro, 2022-02-07:
In a Python script that I can't share here, the `Visualizer::drawToFile` method is not working, and just saving a black image. It may be a good idea to investigate this.

traversaro, 2022-02-08:
As far I understood @CarlottaSartore is using it well (I guess on apt with irrlicht 1.8.4), so it may be indeed a Irrlicht 1.8.4 vs 1.8.5 regression.

CarlottaSartore, 2025-01-24:
I tired to use draw:toFile and I encountered the same error, I was using Irrlicht 1.8.4 

Here my conda list 

<details><summary>Conda list </summary>
<p>

```
# packages in environment at /home/carlotta/miniforge3/envs/comodo_jaxsim:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_x86_64-microarch-level   1                      2_x86_64    conda-forge
absl-py                   2.1.0              pyhd8ed1ab_1    conda-forge
ace                       8.0.2                hc8b94ec_0    conda-forge
adam-robotics             0.3.2              pyhe714708_0    conda-forge
aiohappyeyeballs          2.4.4              pyhd8ed1ab_1    conda-forge
aiohttp                   3.11.11         py310h89163eb_0    conda-forge
aiosignal                 1.3.2              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.13               hb9d3cd8_0    conda-forge
amo-urdf                  0.0.1                    pypi_0    pypi
ampl-asl                  1.0.0                h5888daf_2    conda-forge
anyio                     4.8.0              pyhd8ed1ab_0    conda-forge
aom                       3.9.1                hac33072_0    conda-forge
argon2-cffi               23.1.0             pyhd8ed1ab_1    conda-forge
argon2-cffi-bindings      21.2.0          py310ha75aee5_5    conda-forge
arrow                     1.3.0              pyhd8ed1ab_1    conda-forge
assimp                    5.4.3                h8943939_0    conda-forge
asttokens                 3.0.0              pyhd8ed1ab_1    conda-forge
async-lru                 2.0.4              pyhd8ed1ab_1    conda-forge
async-timeout             5.0.1              pyhd8ed1ab_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     24.3.0             pyh71513ae_0    conda-forge
babel                     2.16.0             pyhd8ed1ab_1    conda-forge
beautifulsoup4            4.12.3             pyha770c72_1    conda-forge
bipedal-locomotion-framework 0.19.0               ha770c72_8    conda-forge
bipedal-locomotion-framework-python 0.19.0          py310hc0ceca1_8    conda-forge
bleach                    6.2.0              pyhd8ed1ab_3    conda-forge
bleach-with-css           6.2.0                hd8ed1ab_3    conda-forge
blosc                     1.21.6               he440d0b_1    conda-forge
brotli                    1.1.0                hb9d3cd8_2    conda-forge
brotli-bin                1.1.0                hb9d3cd8_2    conda-forge
brotli-python             1.1.0           py310hf71b8c6_2    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.34.4               hb9d3cd8_0    conda-forge
ca-certificates           2024.12.14           hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.2               h3394656_1    conda-forge
casadi                    3.6.7           py310ha55d5d0_3    conda-forge
certifi                   2024.12.14         pyhd8ed1ab_0    conda-forge
cffi                      1.17.1          py310h8deb56e_0    conda-forge
cfgv                      3.3.1              pyhd8ed1ab_1    conda-forge
cftime                    1.6.4           py310hf462985_1    conda-forge
charset-normalizer        3.4.1              pyhd8ed1ab_0    conda-forge
chex                      0.1.88             pyhd8ed1ab_1    conda-forge
cli11                     2.4.2                h5888daf_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_1    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_4    conda-forge
comm                      0.2.2              pyhd8ed1ab_1    conda-forge
comodo                    0.0.1                     dev_0    <develop>
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.3.1           py310h3788b33_0    conda-forge
cppad                     20240000.7           h5888daf_1    conda-forge
cuda-cccl_linux-64        12.6.77              ha770c72_0    conda-forge
cuda-crt-dev_linux-64     12.6.85              ha770c72_0    conda-forge
cuda-crt-tools            12.6.85              ha770c72_0    conda-forge
cuda-cudart               12.6.77              h5888daf_0    conda-forge
cuda-cudart-dev_linux-64  12.6.77              h3f2d84a_0    conda-forge
[... 491 lines cut ...]
xorg-libxcursor           1.2.3                hb9d3cd8_0    conda-forge
xorg-libxdamage           1.1.6                hb9d3cd8_0    conda-forge
xorg-libxdmcp             1.1.5                hb9d3cd8_0    conda-forge
xorg-libxext              1.3.6                hb9d3cd8_0    conda-forge
xorg-libxfixes            6.0.1                hb9d3cd8_0    conda-forge
xorg-libxi                1.8.2                hb9d3cd8_0    conda-forge
xorg-libxinerama          1.1.5                h5888daf_1    conda-forge
xorg-libxrandr            1.5.4                hb9d3cd8_0    conda-forge
xorg-libxrender           0.9.12               hb9d3cd8_0    conda-forge
xorg-libxt                1.3.1                hb9d3cd8_0    conda-forge
xorg-libxtst              1.2.5                hb9d3cd8_3    conda-forge
[... cut ...]
yarl                      1.18.3          py310ha75aee5_0    conda-forge
ycm-cmake-modules         0.18.1               h5888daf_0    conda-forge
zeromq                    4.3.5                h3b0a872_7    conda-forge
zipp                      3.21.0             pyhd8ed1ab_1    conda-forge
zlib                      1.3.1                hb9d3cd8_2    conda-forge
zstandard                 0.23.0          py310ha39cb0e_1    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge

```