gbionics/idyntree #975: Matlab crashes when closing the Visualizer on Windows

antonellopaolino, 2022-02-22:
As reported in https://github.com/ami-iit/element_aerodynamics-control/issues/82#issuecomment-1047825701, when trying to close the iDynTree Visualizer from matlab (on Windows OS) it crashes.

<details>

<summary> Here the log file </summary>
MATLAB Log File: C:\Users\apaolino\AppData\Local\Temp\matlab_crash_dump.9036-1

------------------------------------------------
MATLAB Log File
------------------------------------------------ 

--------------------------------------------------------------------------------
                Assertion detected at 2022-02-22 14:51:31 +0100
--------------------------------------------------------------------------------

Configuration:
  Crash Decoding           : Disabled - No sandbox or build area path
  Crash Mode               : continue (default)
  Default Encoding         : windows-1252
  Deployed                 : false
  Graphics Driver          : Unknown hardware 
  Graphics card 1          : Intel Corporation ( 0x8086 ) Intel(R) UHD Graphics 630 Version 27.20.100.9664 (2021-6-1)
  Graphics card 2          : NVIDIA ( 0x10de ) NVIDIA Quadro RTX 3000 Version 27.21.14.6192 (2021-3-10)
  Java Version             : Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  MATLAB Architecture      : win64
  MATLAB Entitlement ID    : 6829974
  MATLAB Root              : C:\Program Files\MATLAB\R2021b
  MATLAB Version           : 9.11.0.1837725 (R2021b) Update 2
  OpenGL                   : hardware
  Operating System         : Microsoft Windows 10 Pro
  Process ID               : 9036
  Processor ID             : x86 Family 6 Model 158 Stepping 10, GenuineIntel
  Session Key              : a1ded017-5c14-4cd0-b358-dd55767b53ef
  Window System            : Version 10.0 (Build 19042)

Fault Count: 1

Assertion in bool __cdecl `anonymous-namespace'::UserEventCondition::timed_wait_impl(class std::chrono::duration<double,struct std::ratio<1,1000> > &&,const class mwboost::optional<struct services::system_events::PpeCondition::InputSource> &) at B:\matlab\src\iqm\UserEvents.cpp line 153:
Received WM_QUIT on the interpreter thread.

Current Thread: 'MCR 0 interpreter thread' id 13636

Register State (captured):
  RAX = 00007ffa817dd9d8  RBX = 00007ffa817dd9d8
  RCX = 0000008497cfce20  RDX = 0000000000000000
  RSP = 0000008497cfcd60  RBP = 0000000000000099
  RSI = 00007ffa4aa0e458  RDI = 00007ffa817dd9d8
 
   R8 = 0000000000000003   R9 = 0000008497cfccd8
  R10 = 0000000000000014  R11 = 00007ffa817cdaf6
  R12 = 0000008497cfda40  R13 = 0000025b65991250
  R14 = 00007ffa817cd918  R15 = 0000008497cfd450
 
  RIP = 00007ffa816b138a  EFL = 00000206
 
   CS = 0033   FS = 0053   GS = 002b

Stack Trace (captured):
[  0] 0x00007ffa816aa043 C:\Program Files\MATLAB\R2021b\bin\win64\libmwfl.dll+00041027 foundation::core::diag::thread_context::unspecified_bool+00000051
[  1] 0x00007ffa816a7338 C:\Program Files\MATLAB\R2021b\bin\win64\libmwfl.dll+00029496 foundation::core::diag::stacktrace_base::capture+00000024
[  2] 0x00007ffa816accee C:\Program Files\MATLAB\R2021b\bin\win64\libmwfl.dll+00052462 foundation::core::diag::symbols::getSymbolAddress+00010334
[  3] 0x00007ffa816abe91 C:\Program Files\MATLAB\R2021b\bin\win64\libmwfl.dll+00048785 foundation::core::diag::symbols::getSymbolAddress+00006657
[  4] 0x00007ffa816b0cfd C:\Program Files\MATLAB\R2021b\bin\win64\libmwfl.dll+00068861 foundation::core::diag::terminate+00000077
[  5] 0x00007ffa4a9e44d8   C:\Program Files\MATLAB\R2021b\bin\win64\iqm.dll+00804056 iqm::detail::initUserEvents+00001992
[  6] 0x00007ffa527f6c37 C:\Program Files\MATLAB\R2021b\bin\win64\libmwservices.dll+02518071 services::system_events::PpeCondition::wait+00000055
[  7] 0x00007ffa29cb6fd5   C:\Program Files\MATLAB\R2021b\bin\win64\uiw.dll+00421845 uiw::UIW_WindowProxy::`vbase destructor'+00014757
[  8] 0x00007ffa58e551fb C:\Program Files\MATLAB\R2021b\bin\win64\libmwbridge.dll+00086523 ioReadLine+00006411
[  9] 0x00007ffa58e53b3e C:\Program Files\MATLAB\R2021b\bin\win64\libmwbridge.dll+00080702 ioReadLine+00000590
[ 10] 0x00007ffa58e538b5 C:\Program Files\MATLAB\R2021b\bin\win64\libmwbridge.dll+00080053 ioReadLine+00000165
[ 11] 0x00007ffa58e64c50 C:\Program Files\MATLAB\R2021b\bin\win64\libmwbridge.dll+00150608 mnGetCommandLineBuffer+00000288
[ 12] 0x00007ffa58e650e2 C:\Program Files\MATLAB\R2021b\bin\win64\libmwbridge.dll+00151778 mnParser+00000466
[ 13] 0x00007ffa4ae5c666   C:\Program Files\MATLAB\R2021b\bin\win64\mcr.dll+00509542 mcr_set_enableReadingFromStdin+00013094
[ 14] 0x00007ffa4adffd03   C:\Program Files\MATLAB\R2021b\bin\win64\mcr.dll+00130307 mcrFunctionSignature::set_signature+00109267
[ 15] 0x00007ffa4ae1cd60   C:\Program Files\MATLAB\R2021b\bin\win64\mcr.dll+00249184 mcrFunctionSignature::set_signature+00228144
[ 16] 0x00007ffa4a9ddc9a   C:\Program Files\MATLAB\R2021b\bin\win64\iqm.dll+00777370 iqm::PackagedTaskPlugin::execute+00000074
[ 17] 0x00007ffa4ae5709d   C:\Program Files\MATLAB\R2021b\bin\win64\mcr.dll+00487581 services::lmgr::exception::LicensingStartupException::~LicensingStartupException+00007213
[ 18] 0x00007ffa4a9b5282   C:\Program Files\MATLAB\R2021b\bin\win64\iqm.dll+00610946 iqm::Iqm::setupIqmFcnPtrs+00094562
[ 19] 0x00007ffa4a9896f6   C:\Program Files\MATLAB\R2021b\bin\win64\iqm.dll+00431862 iqm::Iqm::create+00012454
[ 20] 0x00007ffa4a988de2   C:\Program Files\MATLAB\R2021b\bin\win64\iqm.dll+00429538 iqm::Iqm::create+00010130
[ 21] 0x00007ffa4ae36b9c   C:\Program Files\MATLAB\R2021b\bin\win64\mcr.dll+00355228 mcrInstantiationError::operator=+00010444
[ 22] 0x00007ffa4ae375f5   C:\Program Files\MATLAB\R2021b\bin\win64\mcr.dll+00357877 mcrInstantiationError::operator=+00013093
[ 23] 0x00007ffa4ae34d70   C:\Program Files\MATLAB\R2021b\bin\win64\mcr.dll+00347504 mcrInstantiationError::operator=+00002720
[ 24] 0x00007ffa9be89baa C:\Program Files\MATLAB\R2021b\bin\win64\mwboost_thread-vc142-mt-x64-1_72.dll+00039850 mwboost::thread::swap+00000074
[ 25] 0x00007ffaada11bb2                   C:\Windows\System32\ucrtbase.dll+00138162 configthreadlocale+00000146
[ 26] 0x00007ffaaf977034                   C:\Windows\System32\KERNEL32.DLL+00094260 BaseThreadInitThunk+00000020
[ 27] 0x00007ffaafdc2651                      C:\Windows\SYSTEM32\ntdll.dll+00337489 RtlUserThreadStart+00000033

</details>

You can try to reproduce it in this way:

``` matlab
viz = iDynTree.Visualizer()
viz.init()
viz.run()
viz.close()
```
matlab crashes on `viz.close()` call

traversaro, 2022-02-22:
Thanks for reporting the problem! 

I investigate a bit and the problem is similar to the one reported in https://sourceforge.net/p/irrlicht/bugs/449/ (to clarify, I reached this point as I found that the error is "Received WM_QUIT on the interpreter thread.", and so I looked for WM_QUIT  in Irrlicht\iDynTree code and google).

However, the problem in our case would not by solved by the patch in https://sourceforge.net/p/irrlicht/bugs/449/ . In a nutshell, the problem is that when you close a Irrlicht device on Windows (and we do that in https://github.com/robotology/idyntree/blob/7bcd1754105201bd17b7896ad8c907c353ae9917/src/visualization/src/Visualizer.cpp#L710) the `WM_QUIT   ` signal is invoked (see https://github.com/zaki/irrlicht/blob/97472da9c22ae4a49dfcefb9da5156601fa6a82a/source/Irrlicht/CIrrDeviceWin32.cpp#L1357) but this message is basically "received" by the MATLAB main thread instead of just the received.

As a first workaround, I just tried to comment out https://github.com/robotology/idyntree/blob/7bcd1754105201bd17b7896ad8c907c353ae9917/src/visualization/src/Visualizer.cpp#L710 . 
With that modification, I am able to correctly call the following code:
~~~
viz = iDynTree.Visualizer()
for i=1:20
  viz.init()
  viz.run()
  viz.draw()
  pause(1)
  viz.close()
end
~~~

However, the `viz.close`  does not close the window, so in the end 20 visualizers windows are created:
[image]

As soon as I close one, then a `WM_QUIT` message is created and MATLAB crashes.

traversaro, 2022-02-22:
Note that this kind of problems would be avoided if we had a backend of `iDynTree.Visualizer`  that would open the visualizer in a different process, as discussed in https://github.com/robotology/idyntree/issues/221#issuecomment-592952120 .

traversaro, 2022-02-22:
> As soon as I close one windows with the X button, then a `WM_QUIT` message is created and MATLAB crashes, so I do not think that commenting `closeDevice` is not an option.

Note that this happens due to https://github.com/zaki/irrlicht/blob/97472da9c22ae4a49dfcefb9da5156601fa6a82a/source/Irrlicht/CIrrDeviceWin32.cpp#L918 .

traversaro, 2022-02-23:
I think an alternative solution. Basically when you create a window with Irrlicht, you are creating a class called "device", and Irrlicht supports several "devices". The default one used in Windows is `EIDT_WIN32` (see https://github.com/zaki/irrlicht/blob/97472da9c22ae4a49dfcefb9da5156601fa6a82a/source/Irrlicht/Irrlicht.cpp#L69), that however has the problem described in this issue. However, an alternative one is `EIDT_SDL`, that instead uses the [multiplatform SDL library](https://www.libsdl.org/) (1.2). The SDL is not enabled by most irrlicht builds, but I enabled it for test in conda build (https://github.com/conda-forge/irrlicht-feedstock/pull/8), and after testing it I was able to run:
~~~
viz = iDynTree.Visualizer()
for i=1:20
  viz.init()
  viz.run()
  viz.draw()
  pause(1)
  viz.close()
end
~~~
without any problem. 
So I think we can switch to use `EIDT_SDL` on Windows to solve this bug.

closed 2022-02-23 by traversaro