gbionics/idyntree #985: Simulink controller not running on Windows

antonellopaolino, 2022-04-29:
Trying to run a Simulink controller on Windows, using `iDynTree::Visualizer`, the compilation is successfull but the controller crashes during starting with the following error: `SDL_SetWindowResizable missing in SDL2 library`

<details>
<summary>Simulink error message</summary>
[image]
</details>

<details>
<summary>Mamba env packages</summary>

```
# Name                    Version                   Build  Channel
ace                       7.0.6                h0e60522_0    conda-forge
assimp                    5.2.3                hc2aa0de_1    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
blockfactory              0.8.3               h0e60522_50    robotology
blosc                     1.21.1               hcbbf2c4_0    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
bullet-cpp                3.21                 h2e25243_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.4.2                h8ffe710_0    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cairo                     1.16.0            h15b3021_1010    conda-forge
cfitsio                   4.1.0                h5a969a9_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.23.1               h39d44d4_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
compilers                 1.4.2                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppzmq                    4.8.1                h4324990_0    conda-forge
curl                      7.83.0               h789b8ee_0    conda-forge
cxx-compiler              1.4.2                h2d74725_0    conda-forge
dartsim                   6.12.1               h9f8fb06_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
executing                 0.8.3              pyhd8ed1ab_0    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
fcl                       0.7.0                hbf27b1e_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
flann                     1.9.1             h075982a_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hce3cb01_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fortran-compiler          1.4.2                h9655429_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freeimage                 3.18.0               h6676e37_8    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
gazebo                    11.10.2              hc81694e_2    conda-forge
gazebo-yarp-plugins       4.3.0               h6b60dcb_50    robotology
geos                      3.10.2               h39d44d4_0    conda-forge
geotiff                   1.7.1                h38b14a8_1    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.12.1          nompi_h2a0e4a3_104    conda-forge
icu                       69.1                 h0e60522_0    conda-forge
icub-firmware-shared      1.24.0              h0e60522_50    robotology
[... 136 lines cut ...]
vc                        14.2                 hb210afc_6    conda-forge
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
vs2017_win-64             19.16.27033          hb90652a_6    conda-forge
vswhere                   3.0.1                h57928b3_0    conda-forge
wb-toolbox                5.5.0               h0e60522_50    robotology
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
whole-body-controllers    2.5.6               h0e60522_50    robotology
whole-body-estimators     0.6.1               h0e60522_50    robotology
xerces-c                  3.2.3                h0e60522_4    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yarp                      3.6.0           py39h2bb6a00_50    robotology
ycm-cmake-modules         0.14.0               h0e60522_0    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.11            h8ffe710_1014    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

</details>

Me and @gabrielenava also tried to use `iDynTree` commands from _Matlab Command Window_ and they are not working, returning the same error related to the iDynTree MEX file.

traversaro, 2022-04-29:
> Me and @gabrielenava also tried to use `iDynTree` commands from _Matlab Command Window_ and they are not working, returning the same error related to the iDynTree MEX file.

Which commands did you tried? For sure those are much easier to use to reproduce the problem.

traversaro, 2022-04-29:
I created a test env with sdl 1.2.52 to test the problem.

<details>
<summary>See details of test environment</summary>

~~~
C:\>mamba create -n test985 idyntree icub-models
C:\>mamba activate test985 
(test985) C:\>mamba list
# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\test985:
#
# Name                    Version                   Build  Channel
assimp                    5.2.3                hc2aa0de_1    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
executing                 0.8.3              pyhd8ed1ab_0    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
icub-models               1.23.3          py310h8a704f9_1    conda-forge
idyntree                  5.1.0           py310ha9eedfe_2    conda-forge
intel-openmp              2022.0.0          h57928b3_3663    conda-forge
ipopt                     3.14.5               hf6be2e5_0    conda-forge
ipython                   8.2.0           py310h5588dad_0    conda-forge
irrlicht                  1.8.5                h739eaf8_2    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jedi                      0.18.1          py310h5588dad_1    conda-forge
jpeg                      9e                   h8ffe710_1    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      3.0                  h0e60522_0    conda-forge
libblas                   3.9.0              14_win64_mkl    conda-forge
libcblas                  3.9.0              14_win64_mkl    conda-forge
libdeflate                1.10                 h8ffe710_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
liblapack                 3.9.0              14_win64_mkl    conda-forge
libosqp                   0.6.2                h0e60522_3    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libsodium                 1.0.18               h8d14728_1    conda-forge
libtiff                   4.3.0                hc4061b1_3    conda-forge
libwebp                   1.2.2                h57928b3_0    conda-forge
libwebp-base              1.2.2                h8ffe710_1    conda-forge
libxcb                    1.13              hcd874cb_1004    conda-forge
libxml2                   2.9.13               hf5bbc77_0    conda-forge
libzlib                   1.2.11            h8ffe710_1014    conda-forge
llvm-meta                 5.0.0                         0    conda-forge
lz4-c                     1.9.3                h8ffe710_1    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6    conda-forge
m2w64-gcc-libs            5.3.0                         7    conda-forge
m2w64-gcc-libs-core       5.3.0                         7    conda-forge
m2w64-gmp                 6.1.0                         2    conda-forge
m2w64-libwinpthread-git   5.0.0.4634.697f757               2    conda-forge
matplotlib-inline         0.1.3              pyhd8ed1ab_0    conda-forge
meshcat-python            0.3.2              pyhd8ed1ab_0    conda-forge
metis                     5.1.0             h0e60522_1006    conda-forge
mkl                       2022.0.0           h0e2418a_796    conda-forge
msys2-conda-epoch         20160418                      1    conda-forge
mumps-seq                 5.2.1               hb3f9cae_11    conda-forge
numpy                     1.22.3          py310hed7ac4c_2    conda-forge
openjpeg                  2.4.0                hb211442_1    conda-forge
openmp                    5.0.0                    vc14_1    conda-forge
openssl                   3.0.2                h8ffe710_1    conda-forge
osqp-eigen                0.7.0                h0e60522_0    conda-forge
[... 22 lines cut ...]
traitlets                 5.1.1              pyhd8ed1ab_0    conda-forge
tzdata                    2022a                h191b570_0    conda-forge
u-msgpack-python          2.7.1              pyh9f0ad1d_0    conda-forge
ucrt                      10.0.20348.0         h57928b3_0    conda-forge
vc                        14.2                 hb210afc_6    conda-forge
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
[... cut ...]
I tried with `idyntree-model-view` and it is working fine:
~~~
idyntree-model-view -m %CONDA_PREFIX%\Library\share\iCub\robots\iCubGazeboV3\model.urdf
~~~

So, this is actually a MATLAB-related problem or something related to your setup @antonellopaolino (perhaps a strange SDL2 dll found somewhere). If you are able to write/provide a self-contained MATLAB script that reproduces the problem that would simplifies the reproducing and eventual resolution of the problem. See https://stackoverflow.com/help/minimal-reproducible-example .

traversaro, 2022-04-29:
I installed `idyntree-matlab-bindings` in the test environments to check the problem with MATLAB:
<details>
<summary>Details after idyntree-matlab-bindings installation.</summary>

~~~
(test985) C:\Users\STraversaro>mamba install -c conda-forge -c robotology idyntree-matlab-bindings
(test985) C:\Users\STraversaro>mamba list
# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\test985:
#
# Name                    Version                   Build  Channel
assimp                    5.2.3                hc2aa0de_1    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
executing                 0.8.3              pyhd8ed1ab_0    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
icub-models               1.23.3          py310h8a704f9_1    conda-forge
idyntree                  5.1.0           py310ha9eedfe_2    conda-forge
idyntree-matlab-bindings  5.1.0               h0e60522_50    robotology
intel-openmp              2022.0.0          h57928b3_3663    conda-forge
ipopt                     3.14.5               hf6be2e5_0    conda-forge
ipython                   8.2.0           py310h5588dad_0    conda-forge
irrlicht                  1.8.5                h739eaf8_2    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jedi                      0.18.1          py310h5588dad_1    conda-forge
jpeg                      9e                   h8ffe710_1    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      3.0                  h0e60522_0    conda-forge
libblas                   3.9.0              14_win64_mkl    conda-forge
libcblas                  3.9.0              14_win64_mkl    conda-forge
libdeflate                1.10                 h8ffe710_0    conda-forge
libffi                    3.4.2                h8ffe710_5    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
liblapack                 3.9.0              14_win64_mkl    conda-forge
libosqp                   0.6.2                h0e60522_3    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libsodium                 1.0.18               h8d14728_1    conda-forge
libtiff                   4.3.0                hc4061b1_3    conda-forge
libwebp                   1.2.2                h57928b3_0    conda-forge
libwebp-base              1.2.2                h8ffe710_1    conda-forge
libxcb                    1.13              hcd874cb_1004    conda-forge
libxml2                   2.9.13               hf5bbc77_0    conda-forge
libzlib                   1.2.11            h8ffe710_1014    conda-forge
llvm-meta                 5.0.0                         0    conda-forge
lz4-c                     1.9.3                h8ffe710_1    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6    conda-forge
m2w64-gcc-libs            5.3.0                         7    conda-forge
m2w64-gcc-libs-core       5.3.0                         7    conda-forge
m2w64-gmp                 6.1.0                         2    conda-forge
m2w64-libwinpthread-git   5.0.0.4634.697f757               2    conda-forge
matplotlib-inline         0.1.3              pyhd8ed1ab_0    conda-forge
meshcat-python            0.3.2              pyhd8ed1ab_0    conda-forge
metis                     5.1.0             h0e60522_1006    conda-forge
mkl                       2022.0.0           h0e2418a_796    conda-forge
msys2-conda-epoch         20160418                      1    conda-forge
mumps-seq                 5.2.1               hb3f9cae_11    conda-forge
numpy                     1.22.3          py310hed7ac4c_2    conda-forge
openjpeg                  2.4.0                hb211442_1    conda-forge
openmp                    5.0.0                    vc14_1    conda-forge
openssl                   3.0.2                h8ffe710_1    conda-forge
osqp-eigen                0.7.0                h0e60522_0    conda-forge
parso                     0.8.3              pyhd8ed1ab_0    conda-forge
[... 14 lines cut ...]
setuptools                62.1.0          py310h5588dad_0    conda-forge
six                       1.16.0             pyh6c4a22f_0    conda-forge
sqlite                    3.38.3               h8ffe710_0    conda-forge
stack_data                0.2.0              pyhd8ed1ab_0    conda-forge
tbb                       2021.5.0             h2d74725_1    conda-forge
tk                        8.6.12               h8ffe710_0    conda-forge
tornado                   6.1             py310he2412df_3    conda-forge
traitlets                 5.1.1              pyhd8ed1ab_0    conda-forge
[... cut ...]
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.11            h8ffe710_1014    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge
~~~

</details>

antonellopaolino, 2022-04-29:
> I tried with `idyntree-model-view` and it is working fine:
>
> ```
> idyntree-model-view -m %CONDA_PREFIX%\Library\share\iCub\robots\iCubGazeboV3\model.urdf
> ```
>
> So, this is actually a MATLAB-related [...]

I think it's Matlab-related, since @gabrielenava reproduced the same error on his laptop following the mamba installation, moreover `iDynTree` works from windows command line (I reproduced your test and it works), but if I try to run a simple script with your same configuration

``` matlab
a = iDynTree.StringVector()
```

the output is

```
Invalid MEX-file 'C:\Users\apaolino\AppData\Local\mambaforge\envs\test\Library\mex\iDynTreeMEX.mexw64': A dynamic link library (DLL) initialization routine failed.

Error in iDynTree.StringVector (line 57)
        tmp = iDynTreeMEX(36, varargin{:});

Error in testscript (line 1)
a = iDynTree.StringVector()
```

Which is actually happening with `sdl=1.2.52` but it doesn't happen when running the same code installing `sdl=1.2.15` (I changed just this package).

traversaro, 2022-04-29:
The following code:
~~~
a = iDynTree.StringVector()
~~~
works fine for me on MATLAB R2022a . Which MATLAB version are you using?

antonellopaolino, 2022-04-29:
I'm using MATLAB R2021b.

traversaro, 2022-04-29:
I tested the problem also with MATLAB R2020b and R2021a and the behaviour is the same. 

I have a different problem with sdl 1.2.52 that I do not have with sdl 1.2.15: if I close:
~~~
viz = iDynTree.Visualizer()
viz.init()
viz.draw()
viz.close()
viz.init()
viz.draw()
viz.close()
~~~
the window is opened only once. 

@gabrielenava which MATLAB version did you used instead?

traversaro, 2022-04-30:
I installed MATLAB R2021b, and I still do not have the problem, so there is something else that is causing the problem on both @antonellopaolino and @gabrielenava systems. Perhaps something related to the Toolboxes you have installed?

traversaro, 2022-04-30:
> I have a different problem with sdl 1.2.52 that I do not have with sdl 1.2.15: if I close:
> 
> ```
> viz = iDynTree.Visualizer()
> viz.init()
> viz.draw()
> viz.close()
> viz.init()
> viz.draw()
> viz.close()
> ```
> 
> the window is opened only once.

I experienced the problem also with Python, so this problem is a different problem.

traversaro, 2022-04-30:
From https://it.mathworks.com/matlabcentral/answers/394532-incorrectly-linked-libsdl2-so?s_tid=srchtitle I would expect that some MATLAB toolbox installs a vendored SDL library, but I do not have anything like that in my system.

To debug the situation on systems where we have this problem, I think I found a simple way. 

@antonellopaolino @gabrielenava when you can, can you create a file called `testsdl.sh` in a directory that contains the following:
~~~
echo "Location of SDL.dll : "
which SDL.dll
echo "Location of SDL2.dll : "
which SDL2.dll
~~~

and then execute in the MATLAB Command Line:
~~~
system('"C:\Program Files\Git\bin\bash.exe" testsdl.sh')
~~~
In my system and environment, the output is:
~~~
>> system('"C:\Program Files\Git\bin\bash.exe" testsdl.sh')
Location of SDL.dll : 
/c/Users/STraversaro/AppData/Local/mambaforge/envs/test985/Library/bin/SDL.dll
Location of SDL2.dll : 
/c/Users/STraversaro/AppData/Local/mambaforge/envs/test985/Library/bin/SDL2.dll
~~~

I suspect that in the system where we have a problem a `SDL2.dll` different from the one in the environment is found.

antonellopaolino, 2022-05-02:
> @antonellopaolino @gabrielenava when you can execute in the MATLAB Command Line the lines:
> 
> ```
> system('"C:\Program Files\Git\usr\bin\which.exe" SDL.dll');
> system('"C:\Program Files\Git\usr\bin\which.exe" SDL2.dll');
> ```
> 
> In my system and environment, the output is:
> 
> ```
> >> system('"C:\Program Files\Git\usr\bin\which.exe" SDL.dll');
> system('"C:\Program Files\Git\usr\bin\which.exe" SDL2.dll');
> /c/Users/STraversaro/AppData/Local/mambaforge/envs/test985/Library/bin/SDL.dll
> /c/Users/STraversaro/AppData/Local/mambaforge/envs/test985/Library/bin/SDL2.dll
> ```
> 
> I suspect that in the system where we have a problem a `SDL2.dll` different from the one in the environment is found.

My output is:

```
>> system('"C:\Program Files\Git\usr\bin\which.exe" SDL.dll');
system('"C:\Program Files\Git\usr\bin\which.exe" SDL2.dll');
/c/Users/apaolino/AppData/Local/mambaforge/envs/robotologyenv/Library/bin/SDL.dll
/c/Program Files/MATLAB/R2021b/bin/win64/SDL2.dll
```

It seems to me that this is the cause.

traversaro, 2022-05-02:
Bingo, I wonder how we can discover which toolbox installed it. Which toolbox have you installed?

antonellopaolino, 2022-05-02:
Executing the command:

```
ver
```

in the MATLAB command line I obtained this:

```
MATLAB                                                Version 9.11        (R2021b)
Simulink                                              Version 10.4        (R2021b)
5G Toolbox                                            Version 2.3         (R2021b)
AUTOSAR Blockset                                      Version 2.5         (R2021b)
Aerospace Blockset                                    Version 5.1         (R2021b)
Aerospace Toolbox                                     Version 4.1         (R2021b)
Antenna Toolbox                                       Version 5.1         (R2021b)
Audio Toolbox                                         Version 3.1         (R2021b)
Automated Driving Toolbox                             Version 3.4         (R2021b)
Bioinformatics Toolbox                                Version 4.15.2      (R2021b)
Communications Toolbox                                Version 7.6         (R2021b)
Computer Vision Toolbox                               Version 10.1        (R2021b)
Control System Toolbox                                Version 10.11       (R2021b)
Curve Fitting Toolbox                                 Version 3.6         (R2021b)
DDS Blockset                                          Version 1.1         (R2021b)
DSP System Toolbox                                    Version 9.13        (R2021b)
Data Acquisition Toolbox                              Version 4.4         (R2021b)
Database Toolbox                                      Version 10.2        (R2021b)
Datafeed Toolbox                                      Version 6.1         (R2021b)
Deep Learning HDL Toolbox                             Version 1.2         (R2021b)
Deep Learning Toolbox                                 Version 14.3        (R2021b)
Econometrics Toolbox                                  Version 5.7         (R2021b)
Embedded Coder                                        Version 7.7         (R2021b)
Filter Design HDL Coder                               Version 3.1.10      (R2021b)
Financial Instruments Toolbox                         Version 3.3         (R2021b)
Financial Toolbox                                     Version 6.2         (R2021b)
Fixed-Point Designer                                  Version 7.3         (R2021b)
Fuzzy Logic Toolbox                                   Version 2.8.2       (R2021b)
GPU Coder                                             Version 2.2         (R2021b)
Global Optimization Toolbox                           Version 4.6         (R2021b)
HDL Coder                                             Version 3.19        (R2021b)
HDL Verifier                                          Version 6.4         (R2021b)
Image Acquisition Toolbox                             Version 6.5         (R2021b)
Image Processing Toolbox                              Version 11.4        (R2021b)
Instrument Control Toolbox                            Version 4.5         (R2021b)
LTE Toolbox                                           Version 3.6         (R2021b)
Lidar Toolbox                                         Version 2.0         (R2021b)
MATLAB Coder                                          Version 5.3         (R2021b)
MATLAB Compiler                                       Version 8.3         (R2021b)
MATLAB Compiler SDK                                   Version 6.11        (R2021b)
MATLAB Report Generator                               Version 5.11        (R2021b)
Mapping Toolbox                                       Version 5.2         (R2021b)
Mixed-Signal Blockset                                 Version 2.1         (R2021b)
Model Predictive Control Toolbox                      Version 7.2         (R2021b)
Model-Based Calibration Toolbox                       Version 5.11        (R2021b)
Motor Control Blockset                                Version 1.3         (R2021b)
Navigation Toolbox                                    Version 2.1         (R2021b)
OPC Toolbox                                           Version 5.0.3       (R2021b)
Optimization Toolbox                                  Version 9.2         (R2021b)
Parallel Computing Toolbox                            Version 7.5         (R2021b)
Partial Differential Equation Toolbox                 Version 3.7         (R2021b)
Phased Array System Toolbox                           Version 4.6         (R2021b)
Powertrain Blockset                                   Version 1.10        (R2021b)
Predictive Maintenance Toolbox                        Version 2.4         (R2021b)
RF Blockset                                           Version 8.2         (R2021b)
RF PCB Toolbox                                        Version 1.0         (R2021b)
RF Toolbox                                            Version 4.2         (R2021b)
ROS Toolbox                                           Version 1.4         (R2021b)
Radar Toolbox                                         Version 1.1         (R2021b)
Reinforcement Learning Toolbox                        Version 2.1         (R2021b)
Risk Management Toolbox                               Version 1.10        (R2021b)
[... 20 lines cut ...]
Simulink Coverage                                     Version 5.3         (R2021b)
Simulink Design Optimization                          Version 3.10        (R2021b)
Simulink Design Verifier                              Version 4.6         (R2021b)
[... cut ...]
Vehicle Dynamics Blockset                             Version 1.7         (R2021b)
Vehicle Network Toolbox                               Version 5.1         (R2021b)
Vision HDL Toolbox                                    Version 2.4         (R2021b)
WLAN Toolbox                                          Version 3.3         (R2021b)
Wavelet Toolbox                                       Version 6.0         (R2021b)
Wireless HDL Toolbox                                  Version 2.3         (R2021b)
```

traversaro, 2022-05-02:
Too many to be able to understand on our own, asked for help in https://mathworks.com/matlabcentral/answers/1709785-it-is-possible-to-find-which-toolbox-installed-a-given-file . @gabrielenava just in case, do you happen to have less toolboxes installed?

traversaro, 2022-05-02:
By the way, just to make a test you can try if everything works if you rename `/c/Program Files/MATLAB/R2021b/bin/win64/SDL2.dll` in `/c/Program Files/MATLAB/R2021b/bin/win64/SDL2.dll.backup`, even if I am afraid that then you will experience https://github.com/robotology/idyntree/issues/986 .

antonellopaolino, 2022-05-02:
> By the way, just to make a test you can try if everything works if you rename `/c/Program Files/MATLAB/R2021b/bin/win64/SDL2.dll` in `/c/Program Files/MATLAB/R2021b/bin/win64/SDL2.dll.backup`, even if I am afraid that then you will experience #986 .

After renaming it's using the correct library:
```
>> system('"C:\Program Files\Git\usr\bin\which.exe" SDL.dll');
system('"C:\Program Files\Git\usr\bin\which.exe" SDL2.dll');
/c/Users/apaolino/AppData/Local/mambaforge/envs/robotologyenv/Library/bin/SDL.dll
/c/Users/apaolino/AppData/Local/mambaforge/envs/robotologyenv/Library/bin/SDL2.dll
```

antonellopaolino, 2022-05-02:
Still have to try, for now I can confirm that executing this code:

```
viz = iDynTree.Visualizer()
viz.init()
viz.draw()
viz.close()
viz.init()
viz.draw()
viz.close()
```

the visualizer opens only the first time when using sdl 1.2.52, while it opens and closes twice when using sdl 1.2.15.

antonellopaolino, 2022-05-02:
> And the controller is running correctly?

Now it crashes with this log file:

<details>
<summary>Error log</summary>
MATLAB Log File: C:\Users\apaolino\AppData\Local\Temp\matlab_crash_dump.11448-1

------------------------------------------------
MATLAB Log File
------------------------------------------------ 

--------------------------------------------------------------------------------
             Access violation detected at 2022-05-02 17:04:18 +0200
--------------------------------------------------------------------------------

Configuration:
  Crash Decoding           : Disabled - No sandbox or build area path
  Crash Mode               : continue (default)
  Default Encoding         : windows-1252
  Deployed                 : false
  Graphics Driver          : Intel Intel(R) UHD Graphics 630 Version 4.5.0 - Build 27.20.100.9664
  Graphics Text Issue Info : xmin=0.000000, ymin=-2.400000, xmax=14.400000, ymax=10.200000
  Graphics card 1          : Intel Corporation ( 0x8086 ) Intel(R) UHD Graphics 630 Version 27.20.100.9664 (2021-6-1)
  Graphics card 2          : Not Started NVIDIA ( 0x10de ) NVIDIA Quadro RTX 3000 Version 27.21.14.6192 (2021-3-10)
  Java Version             : Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  MATLAB Architecture      : win64
  MATLAB Entitlement ID    : 6829974
  MATLAB Root              : C:\Program Files\MATLAB\R2021b
  MATLAB Version           : 9.11.0.1873467 (R2021b) Update 3
  OpenGL                   : hardware
  Operating System         : Microsoft Windows 10 Enterprise
  Process ID               : 11448
  Processor ID             : x86 Family 6 Model 158 Stepping 10, GenuineIntel
  Session Key              : 0a485f2b-002f-4311-b6c1-4f17775a314b
  Window System            : Version 10.0 (Build 19042)

Fault Count: 1

Abnormal termination:
Access violation

Current Thread: 'MCR 0 interpreter thread' id 7992

Register State (from fault):
  RAX = 0000000000000000  RBX = 000001d98d1c4e60
  RCX = 000001d98d1c4f70  RDX = 0000000000000000
  RSP = 000000268b9eda88  RBP = 000000268b9edb90
  RSI = 00000000ffffffff  RDI = 000001d98d1c51b0
 
   R8 = 000001d98d1c5050   R9 = 000001d98d1c5138
  R10 = 0000000000008000  R11 = 000000268b9ede70
  R12 = 0000000000000006  R13 = 000001d98d1c51b0
  R14 = 000001d98d1c4f70  R15 = 000001d98d1c5138
 
  RIP = 00007ff810653733  EFL = 00010246
 
   CS = 0033   FS = 0053   GS = 002b

Stack Trace (from fault):
[  0] 0x00007ff810653733 C:\Users\apaolino\AppData\Local\mambaforge\envs\robotologyenv\Library\bin\idyntree-model.dll+00538419 iDynTree::SubModelDecomposition::getTraversal+00000003
[  1] 0x00007ff8105e659b C:\Users\apaolino\AppData\Local\mambaforge\envs\robotologyenv\Library\bin\idyntree-model.dll+00091547 iDynTree::CompositeRigidBodyAlgorithm+00001579
[  2] 0x00007ff8113e4091 C:\Users\apaolino\AppData\Local\mambaforge\envs\robotologyenv\Library\bin\idyntree-high-level.dll+00082065 iDynTree::KinDynComputations::computeRawMassMatrixAndTotalMomentum+00000097
[  3] 0x00007ff8113e6f14 C:\Users\apaolino\AppData\Local\mambaforge\envs\robotologyenv\Library\bin\idyntree-high-level.dll+00093972 iDynTree::KinDynComputations::getCenterOfMassPosition+00000036
[  4] 0x00007ff81192e4c9 C:\Users\apaolino\AppData\Local\mambaforge\envs\robotologyenv\Library\mex\iDynTreeMEX.mexw64+00779465
[  5] 0x00007ff81192e13f C:\Users\apaolino\AppData\Local\mambaforge\envs\robotologyenv\Library\mex\iDynTreeMEX.mexw64+00778559
[  6] 0x00007ff811a75311 C:\Users\apaolino\AppData\Local\mambaforge\envs\robotologyenv\Library\mex\iDynTreeMEX.mexw64+02118417 mexFunction+00054497
[  7] 0x00007ff84a205c99 C:\Program Files\MATLAB\R2021b\bin\win64\libmex.dll+00547993 mexfile::Inspector::needs_upgrade+00030057
[  8] 0x00007ff84a209bf2 C:\Program Files\MATLAB\R2021b\bin\win64\libmex.dll+00564210 MexRetrieveVersion+00011378
[  9] 0x00007ff84a209d67 C:\Program Files\MATLAB\R2021b\bin\win64\libmex.dll+00564583 MexRetrieveVersion+00011751
[... 107 lines cut ...]
[117] 0x00007fffcb8f959c C:\Program Files\MATLAB\R2021b\bin\win64\systemobject_mcos.dll+01021340 somcos::callProcessTunedPropertiesImpl+00002220
[118] 0x00007fffcb8e0ac9 C:\Program Files\MATLAB\R2021b\bin\win64\systemobject_mcos.dll+00920265 somcos::callSetupImpl+00000089
[119] 0x00007fffcbbb9675 C:\Program Files\MATLAB\R2021b\bin\win64\matlabsystemblock.dll+01087093 sysobjblock::MSysObjDispatcher::invokeSysObjMethodImpl+00000533
[120] 0x00007fffcbbacb59 C:\Program Files\MATLAB\R2021b\bin\win64\matlabsystemblock.dll+01035097 sysobjblock::MSysObjDispatcher::callSetupImpl+00000377
[121] 0x00007fffcbbad090 C:\Program Files\MATLAB\R2021b\bin\win64\matlabsystemblock.dll+01036432 sysobjblock::MSysObjDispatcher::callStart+00001088
[122] 0x00007fffcbb70b04 C:\Program Files\MATLAB\R2021b\bin\win64\matlabsystemblock.dll+00789252 SysBlockBase::CreateExecutionBlock+00006324
[123] 0x00007fffedf76533 C:\Program Files\MATLAB\R2021b\bin\win64\libmwsimulink.dll+13395251 BlockSetLocation+00000451
[124] 0x00007fffedbd7623 C:\Program Files\MATLAB\R2021b\bin\win64\libmwsimulink.dll+09598499 ouRequestBlockReduction+00041603
[125] 0x00007fffeb57493a C:\Program Files\MATLAB\R2021b\bin\win64\sl_engine_classes.dll+02246970 SLExecBD::executeInAccelForNormalModeBasicCheck+00003434
[126] 0x00007fffeb63e0e9 C:\Program Files\MATLAB\R2021b\bin\win64\sl_engine_classes.dll+03072233 SlSystemDefaultMethods::SetupRuntimeResources+00000297
[... cut ...] rces+00000828

Program State:

Most Recent Simulink Activity:
playSimulationAction : OK in editor 1 at Mon May  2 17:04:12 2022
		updateDiagramAction : OK in editor 1 at Mon May  2 17:03:52 2022
		Ctrl+D : OK in editor 1 at Mon May  2 17:03:52 2022
		

This error was detected while a MEX-file was running. If the MEX-file
is not an official MathWorks function, please examine its source code
for errors. Please consult the External Interfaces Guide for information
on debugging MEX-files.
<\details>

traversaro, 2022-05-02:
Interesting, this does not seems related at all to SDL2 or to the problem we had.

antonellopaolino, 2022-05-02:
Yes, the problem with SDL2 for now it's solved.

traversaro, 2022-05-02:
Yes, but the controller is working if sdl 1.2.15 is used, right? So in some way or another the problem is related.

antonellopaolino, 2022-05-03:
I found out that there was a wrong path, now it runs with both the environments (sdl 1.2.15 and 1.2.52).

traversaro, 2022-05-03:
Ok, great! Are you able also to launch it two times in a row? Or the visualizer is not launched in the second run, as I would expect due to https://github.com/robotology/idyntree/issues/986 .

antonellopaolino, 2022-05-05:
As expected, when running the controller a second consecutive time after stopping it, the controller terminates with this error:

```
An error occurred while running the simulation and the simulation was terminated

Caused by: Closing visualizer.
```

traversaro, 2022-05-05:
Interesting, so that means that `viz.run` is returning false?

antonellopaolino, 2022-05-05:
Exactly, the error is thrown by the `viz.run()` returning false in the robot Visualizer used by the Simulink controller.

gabrielenava, 2022-05-09:
a bit too late but:

>@gabrielenava which MATLAB version did you used instead?

R2021a

>@gabrielenava just in case, do you happen to have less toolboxes installed?

no I have all possible toolbox installed

LorenzoFiori, 2022-12-23:
@traversaro

```matlab 
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.13.0.2126072 (R2022b) Update 3
Operating System: Microsoft Windows 10 Pro Version 10.0 (Build 19045)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.13        (R2022b)
Simulink                                              Version 10.6        (R2022b)
5G Toolbox                                            Version 2.5         (R2022b)
AUTOSAR Blockset                                      Version 3.0         (R2022b)
Aerospace Blockset                                    Version 5.3         (R2022b)
Aerospace Toolbox                                     Version 4.3         (R2022b)
Antenna Toolbox                                       Version 5.3         (R2022b)
Audio Toolbox                                         Version 3.3         (R2022b)
Automated Driving Toolbox                             Version 3.6         (R2022b)
Bioinformatics Toolbox                                Version 4.16.1      (R2022b)
Bluetooth Toolbox                                     Version 1.1         (R2022b)
Communications Toolbox                                Version 7.8         (R2022b)
Computer Vision Toolbox                               Version 10.3        (R2022b)
Control System Toolbox                                Version 10.12       (R2022b)
Curve Fitting Toolbox                                 Version 3.8         (R2022b)
DDS Blockset                                          Version 1.3         (R2022b)
DSP HDL Toolbox                                       Version 1.1         (R2022b)
DSP System Toolbox                                    Version 9.15        (R2022b)
Data Acquisition Toolbox                              Version 4.6         (R2022b)
Database Toolbox                                      Version 10.4        (R2022b)
Datafeed Toolbox                                      Version 6.3         (R2022b)
Deep Learning HDL Toolbox                             Version 1.4         (R2022b)
Deep Learning Toolbox                                 Version 14.5        (R2022b)
Econometrics Toolbox                                  Version 6.1         (R2022b)
Embedded Coder                                        Version 7.9         (R2022b)
Filter Design HDL Coder                               Version 3.1.12      (R2022b)
Financial Instruments Toolbox                         Version 3.5         (R2022b)
Financial Toolbox                                     Version 6.4         (R2022b)
Fixed-Point Designer                                  Version 7.5         (R2022b)
Fuzzy Logic Toolbox                                   Version 3.0         (R2022b)
GPU Coder                                             Version 2.4         (R2022b)
Global Optimization Toolbox                           Version 4.8         (R2022b)
HDL Coder                                             Version 4.0         (R2022b)
HDL Verifier                                          Version 7.0         (R2022b)
Image Acquisition Toolbox                             Version 6.7         (R2022b)
Image Processing Toolbox                              Version 11.6        (R2022b)
Industrial Communication Toolbox                      Version 6.1         (R2022b)
Instrument Control Toolbox                            Version 4.7         (R2022b)
LTE Toolbox                                           Version 3.8         (R2022b)
Lidar Toolbox                                         Version 2.2         (R2022b)
MATLAB Coder                                          Version 5.5         (R2022b)
MATLAB Compiler                                       Version 8.5         (R2022b)
MATLAB Compiler SDK                                   Version 7.1         (R2022b)
MATLAB Report Generator                               Version 5.13        (R2022b)
Mapping Toolbox                                       Version 5.4         (R2022b)
Medical Imaging Toolbox                               Version 1.0         (R2022b)
Mixed-Signal Blockset                                 Version 2.3         (R2022b)
Model Predictive Control Toolbox                      Version 8.0         (R2022b)
Model-Based Calibration Toolbox                       Version 5.13        (R2022b)
Motor Control Blockset                                Version 1.5         (R2022b)
Navigation Toolbox                                    Version 2.3         (R2022b)
Optimization Toolbox                                  Version 9.4         (R2022b)
Parallel Computing Toolbox                            Version 7.7         (R2022b)
Partial Differential Equation Toolbox                 Version 3.9         (R2022b)
Phased Array System Toolbox                           Version 4.8         (R2022b)
Powertrain Blockset                                   Version 1.12        (R2022b)
Predictive Maintenance Toolbox                        Version 2.6         (R2022b)
RF Blockset                                           Version 8.4         (R2022b)
RF PCB Toolbox                                        Version 1.2         (R2022b)
RF Toolbox                                            Version 4.4         (R2022b)
[... cut ...]
Vehicle Network Toolbox                               Version 5.3         (R2022b)
Vision HDL Toolbox                                    Version 2.6         (R2022b)
WLAN Toolbox                                          Version 3.5         (R2022b)
Wavelet Toolbox                                       Version 6.2         (R2022b)
Wireless HDL Toolbox                                  Version 2.5         (R2022b)
Wireless Testbench                                    Version 1.1         (R2022b)
```
sorry but I cant attach a zip file with SDL2.ddl

vpunithreddy, 2023-09-07:
I also experience this error:
~~~
>> ver

-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.11.0.1873467 (R2021b) Update 3
MATLAB License Number: 40875934
Operating System: Microsoft Windows 11 Enterprise Version 10.0 (Build 22000)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.11        (R2021b)
Simulink                                              Version 10.4        (R2021b)
5G Toolbox                                            Version 2.3         (R2021b)
AUTOSAR Blockset                                      Version 2.5         (R2021b)
Aerospace Blockset                                    Version 5.1         (R2021b)
Aerospace Toolbox                                     Version 4.1         (R2021b)
Antenna Toolbox                                       Version 5.1         (R2021b)
Audio Toolbox                                         Version 3.1         (R2021b)
Automated Driving Toolbox                             Version 3.4         (R2021b)
Bioinformatics Toolbox                                Version 4.15.2      (R2021b)
Communications Toolbox                                Version 7.6         (R2021b)
Computer Vision Toolbox                               Version 10.1        (R2021b)
Control System Toolbox                                Version 10.11       (R2021b)
Curve Fitting Toolbox                                 Version 3.6         (R2021b)
DDS Blockset                                          Version 1.1         (R2021b)
DSP System Toolbox                                    Version 9.13        (R2021b)
Data Acquisition Toolbox                              Version 4.4         (R2021b)
Database Toolbox                                      Version 10.2        (R2021b)
Datafeed Toolbox                                      Version 6.1         (R2021b)
Deep Learning HDL Toolbox                             Version 1.2         (R2021b)
Deep Learning Toolbox                                 Version 14.3        (R2021b)
Econometrics Toolbox                                  Version 5.7         (R2021b)
Embedded Coder                                        Version 7.7         (R2021b)
Filter Design HDL Coder                               Version 3.1.10      (R2021b)
Financial Instruments Toolbox                         Version 3.3         (R2021b)
Financial Toolbox                                     Version 6.2         (R2021b)
Fixed-Point Designer                                  Version 7.3         (R2021b)
Fuzzy Logic Toolbox                                   Version 2.8.2       (R2021b)
GPU Coder                                             Version 2.2         (R2021b)
Global Optimization Toolbox                           Version 4.6         (R2021b)
HDL Coder                                             Version 3.19        (R2021b)
HDL Verifier                                          Version 6.4         (R2021b)
Image Acquisition Toolbox                             Version 6.5         (R2021b)
Image Processing Toolbox                              Version 11.4        (R2021b)
Instrument Control Toolbox                            Version 4.5         (R2021b)
LTE Toolbox                                           Version 3.6         (R2021b)
Lidar Toolbox                                         Version 2.0         (R2021b)
MATLAB Coder                                          Version 5.3         (R2021b)
MATLAB Compiler                                       Version 8.3         (R2021b)
MATLAB Compiler SDK                                   Version 6.11        (R2021b)
MATLAB Report Generator                               Version 5.11        (R2021b)
Mapping Toolbox                                       Version 5.2         (R2021b)
Mixed-Signal Blockset                                 Version 2.1         (R2021b)
Model Predictive Control Toolbox                      Version 7.2         (R2021b)
Model-Based Calibration Toolbox                       Version 5.11        (R2021b)
Motor Control Blockset                                Version 1.3         (R2021b)
Navigation Toolbox                                    Version 2.1         (R2021b)
OPC Toolbox                                           Version 5.0.3       (R2021b)
Optimization Toolbox                                  Version 9.2         (R2021b)
Parallel Computing Toolbox                            Version 7.5         (R2021b)
Partial Differential Equation Toolbox                 Version 3.7         (R2021b)
Phased Array System Toolbox                           Version 4.6         (R2021b)
Powertrain Blockset                                   Version 1.10        (R2021b)
Predictive Maintenance Toolbox                        Version 2.4         (R2021b)
RF Blockset                                           Version 8.2         (R2021b)
RF PCB Toolbox                                        Version 1.0         (R2021b)
RF Toolbox                                            Version 4.2         (R2021b)
ROS Toolbox                                           Version 1.4         (R2021b)
Radar Toolbox                                         Version 1.1         (R2021b)
Reinforcement Learning Toolbox                        Version 2.1         (R2021b)
[... cut ...]
Vehicle Dynamics Blockset                             Version 1.7         (R2021b)
Vehicle Network Toolbox                               Version 5.1         (R2021b)
Vision HDL Toolbox                                    Version 2.4         (R2021b)
WLAN Toolbox                                          Version 3.3         (R2021b)
Wavelet Toolbox                                       Version 6.0         (R2021b)
Wireless HDL Toolbox                                  Version 2.3         (R2021b)
~~~

DavideBarbone, 2023-11-21:
I have the same problem on MATLAB R2023b with all toolbox installed, the SDL2 version installed there is: 
[SDL2.zip](https://github.com/robotology/idyntree/files/13425198/SDL2.zip) .

traversaro, 2023-11-21:
> I have the same problem on MATLAB R2023b with all toolbox installed, the SDL2 version installed there is: [SDL2.zip](https://github.com/robotology/idyntree/files/13425198/SDL2.zip) .

I just tested this .dll, and even on MATLAB R2023b the SDL2 version installed is really old:
~~~
SDL version:
Compiled: 2.28.4
Linked: 2.0.3
~~~

traversaro, 2024-05-09:
@Gianlucamilani experienced the same problem, again has it had all the toolbox installed. However, he was also using R2021a.

traversaro, 2024-05-09:
> @Gianlucamilani experienced the same problem, again has it had all the toolbox installed. However, he was also using R2021a.

Just tested on R2024a, and the problem is still there.

akhilsathuluri, 2024-07-01:
I just faced the same problem, but is fixed by the workaround! Thanks!

> Too many to be able to understand on our own, asked for help in https://mathworks.com/matlabcentral/answers/1709785-it-is-possible-to-find-which-toolbox-installed-a-given-file . @gabrielenava just in case, do you happen to have less toolboxes installed?

I dont have a lot of toolboxes installed, so heres the output of `ver` if it helps

```
-----------------------------------------------------------------------------------------------------
MATLAB Version: 23.2.0.2515942 (R2023b) Update 7
MATLAB License Number: 722433
Operating System: Microsoft Windows 11 Pro Version 10.0 (Build 22631)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 23.2        (R2023b)
Simulink                                              Version 23.2        (R2023b)
Control System Toolbox                                Version 23.2        (R2023b)
Global Optimization Toolbox                           Version 23.2        (R2023b)
Image Processing Toolbox                              Version 23.2        (R2023b)
MATLAB Compiler                                       Version 23.2        (R2023b)
Optimization Toolbox                                  Version 23.2        (R2023b)
ROS Toolbox                                           Version 23.2        (R2023b)
Robotics System Toolbox                               Version 23.2        (R2023b)
Symbolic Math Toolbox                                 Version 23.2        (R2023b)
```

traversaro, 2024-09-20:
@mfussi66 was impacted by this today. :(

mfussi66, 2024-09-20:
> @mfussi66 was impacted by this today. :(

This is my list of toolboxes, I'm highlighting which ones might be the culprits:

```diff
MATLAB                                                Version 24.1        (R2024a)
Simulink                                              Version 24.1        (R2024a)
Aerospace Blockset                                    Version 24.1        (R2024a)
+ Aerospace Toolbox                                     Version 24.1        (R2024a)
Antenna Toolbox                                       Version 24.1        (R2024a)
+ Automated Driving Toolbox                             Version 24.1        (R2024a)
+ Computer Vision Toolbox                               Version 24.1        (R2024a)
Control System Toolbox                                Version 24.1        (R2024a)
Curve Fitting Toolbox                                 Version 24.1        (R2024a)
DSP HDL Toolbox                                       Version 24.1        (R2024a)
DSP System Toolbox                                    Version 24.1        (R2024a)
Embedded Coder                                        Version 24.1        (R2024a)
Filter Design HDL Coder                               Version 24.1        (R2024a)
Fixed-Point Designer                                  Version 24.1        (R2024a)
Global Optimization Toolbox                           Version 24.1        (R2024a)
Image Processing Toolbox                              Version 24.1        (R2024a)
Lidar Toolbox                                         Version 24.1        (R2024a)
MATLAB Coder                                          Version 24.1        (R2024a)
MATLAB Compiler                                       Version 24.1        (R2024a)
MATLAB Compiler SDK                                   Version 24.1        (R2024a)
MATLAB Report Generator                               Version 24.1        (R2024a)
MATLAB Test                                           Version 24.1        (R2024a)
Mapping Toolbox                                       Version 24.1        (R2024a)
Model Predictive Control Toolbox                      Version 24.1        (R2024a)
Model-Based Calibration Toolbox                       Version 24.1        (R2024a)
Motor Control Blockset                                Version 24.1        (R2024a)
Navigation Toolbox                                    Version 24.1        (R2024a)
Optimization Toolbox                                  Version 24.1        (R2024a)
Parallel Computing Toolbox                            Version 24.1        (R2024a)
Phased Array System Toolbox                           Version 24.1        (R2024a)
Powertrain Blockset                                   Version 24.1        (R2024a)
Predictive Maintenance Toolbox                        Version 24.1        (R2024a)
ROS Toolbox                                           Version 24.1        (R2024a)
Radar Toolbox                                         Version 24.1        (R2024a)
Requirements Toolbox                                  Version 24.1        (R2024a)
+ Robotics System Toolbox                               Version 24.1        (R2024a)
Robust Control Toolbox                                Version 24.1        (R2024a)
Sensor Fusion and Tracking Toolbox                    Version 24.1        (R2024a)
Signal Processing Toolbox                             Version 24.1        (R2024a)
Simscape                                              Version 24.1        (R2024a)
Simscape Battery                                      Version 24.1        (R2024a)
+ Simscape Driveline                                    Version 24.1        (R2024a)
Simscape Electrical                                   Version 24.1        (R2024a)
Simscape Fluids                                       Version 24.1        (R2024a)
+ Simscape Multibody                                    Version 24.1        (R2024a)
Simscape Multibody Link                               Version 24.1        (R2024a)
Simulink 3D Animation                                 Version 24.1        (R2024a)
Simulink Check                                        Version 24.1        (R2024a)
Simulink Code Inspector                               Version 24.1        (R2024a)
Simulink Coder                                        Version 24.1        (R2024a)
Simulink Compiler                                     Version 24.1        (R2024a)
Simulink Control Design                               Version 24.1        (R2024a)
Simulink Coverage                                     Version 24.1        (R2024a)
Simulink Design Optimization                          Version 24.1        (R2024a)
Simulink Design Verifier                              Version 24.1        (R2024a)
Simulink Desktop Real-Time                            Version 24.1        (R2024a)
Simulink Fault Analyzer                               Version 24.1        (R2024a)
Simulink PLC Coder                                    Version 24.1        (R2024a)
Simulink Real-Time                                    Version 24.1        (R2024a)
Simulink Report Generator                             Version 24.1        (R2024a)
Simulink Test                                         Version 24.1        (R2024a)
Speedgoat I/O Blockset                                Version 9.8.1.1     (R2024a)
Stateflow                                             Version 24.1        (R2024a)
Statistics and Machine Learning Toolbox               Version 24.1        (R2024a)
Symbolic Math Toolbox                                 Version 24.1        (R2024a)
System Composer                                       Version 24.1        (R2024a)
System Identification Toolbox                         Version 24.1        (R2024a)
+ Vehicle Dynamics Blockset                             Version 24.1        (R2024a)
Vehicle Network Toolbox                               Version 24.1        (R2024a)
```

traversaro, 2024-09-20:
Could be! It is also possible that multiple toolboxes install that files.

traversaro, 2025-01-17:
@FabioBergonti is affected by this:

~~~
>> ver
-----------------------------------------------------------------------------------------------------
MATLAB Version: 24.2.0.2806996 (R2024b) Update 3
MATLAB License Number: 40875934
Operating System: Microsoft Windows 11 Enterprise Version 10.0 (Build 22631)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 24.2        (R2024b)
Simulink                                              Version 24.2        (R2024b)
Control System Toolbox                                Version 24.2        (R2024b)
Curve Fitting Toolbox                                 Version 24.2        (R2024b)
Data Acquisition Toolbox                              Version 24.2        (R2024b)
Global Optimization Toolbox                           Version 24.2        (R2024b)
MATLAB Compiler                                       Version 24.2        (R2024b)
MATLAB Report Generator                               Version 24.2        (R2024b)
MATLAB Test                                           Version 24.2        (R2024b)
Motor Control Blockset                                Version 24.2        (R2024b)
Optimization Toolbox                                  Version 24.2        (R2024b)
Requirements Toolbox                                  Version 24.2        (R2024b)
Robotics System Toolbox                               Version 24.2        (R2024b)
Signal Processing Toolbox                             Version 24.2        (R2024b)
Simscape                                              Version 24.2        (R2024b)
Simscape Electrical                                   Version 24.2        (R2024b)
Simscape Multibody                                    Version 24.2        (R2024b)
Simulink Compiler                                     Version 24.2        (R2024b)
Simulink Control Design                               Version 24.2        (R2024b)
Simulink Design Optimization                          Version 24.2        (R2024b)
Simulink Test                                         Version 24.2        (R2024b)
Statistics and Machine Learning Toolbox               Version 24.2        (R2024b)
Symbolic Math Toolbox                                 Version 24.2        (R2024b)
System Composer                                       Version 24.2        (R2024b)
~~~

intersecating the other lists, it is possible that the culprit is "Robotics System Toolbox".