gbionics/idyntree #994: Problem when  asking to reach a specific target with the right hand and the left hand

marcello-goccia, 2022-05-27:
@Nicogene and I are using iDynTree within Blender to allow to the robot to reach a cartesian target (xyz-rpy). We define the base frame and the end-effector frame and we make use of the inverse kinematic from iDynTree to move to that target.
We found a problem when we ask for a specific target with the right and the left hand, the resulting pose is in two completely different positions.

For example, when we want to reach the target xyz=0,0,0 and rpy=0,0,0, with respect to the root link this is the pose for the right hand:

[image]

and the pose for the left hand  

[image]

Alongside @traversaro we found out some interesting things, that makes us suspect that this "left-hand issue" is due to the rpy singularity.

We printed the relative rotation between r_hand and l_hand

```python
print(IkVariables.dynComp.getRelativeTransform("r_hand", "l_hand").getRotation().asRPY().toString())
```

And it was

```
-2.29128e-16 5.55112e-17 3.14159
```

There are 180 degrees difference on the yaw component.
@traversaro suggested us to attach a fake frame to the `l_hand rotated` of 180 and ask the solver to find a solution to that frame, which we called `l_hand_rotated`.

```python
    pos = iDynTree.Position(0,0,0)
    rot = iDynTree.Rotation.RPY(0,0,math.pi)
    transf = iDynTree.Transform(rot,pos)
    ikv.iDynTreeModel.addAdditionalFrameToLink("l_hand","l_hand_rotated",transf)
```
	
(using this [function](https://robotology.github.io/idyntree/classiDynTree_1_1Model.html#ab67ec3bb2074412c1002befff50e2f57))

And it then worked fine

[image]

traversaro, 2022-05-27:
Thanks for opening the issue! Can you clarify:
* the iCub model you are using
* the iDynTree version you are using?

With this info it should be easy to write a [mre](https://stackoverflow.com/help/minimal-reproducible-example).

marcello-goccia, 2022-05-30:
We tried with `iCubGazeboV2_5` and `iCubGazeboV3`

traversaro, 2022-05-30:
Sorry, I meant which version of the icub-models repo/package, like 1.23.4, 1.23.3, etc etc.

marcello-goccia, 2022-05-30:
It should be `master` (I could ask @Nicogene to confirm).
Thank you.

traversaro, 2022-05-30:
Thanks, so I guess https://github.com/robotology/icub-models/commit/f546203781a209e57d1441b5d5facdee9caf90e3 (I mark down the commit as the master branch could change in the future).