gbionics/idyntree #999: Wrong type returned when calling FromPython function

isorrentino, 2022-06-11:
Here an example. If I run this code:

```python
x_py = numpy.zeros(6)
x_idyn = idyn.JointPosDoubleArray(6)
x_idyn = x_idyn.FromPython(x_py)
```
`x_idyn` becomes a `VectorDynSize` object. Maybe, this happens because the `FromPython` function that is actually called is the one defined in https://github.com/robotology/idyntree/blob/163cff7899f7a660ed78fe550e4774553a4ced60/bindings/python/python.i#L184.

traversaro, 2022-06-13:
Good point. This is a nasty effect of the fact that `.FromPython` method is a function that returns a new instance of the type, instead of populating the instance on which it is called (as `.fromMatlab` does). I am not sure why this happened back in time, but the quickest solution for your use case is to add a `FromMatlab` overload for `JointPosDoubleArray`, as done for example in https://github.com/robotology/idyntree/pull/959 for `iDynTree::Position`.

traversaro, 2022-06-13:
An alternative may be to define a `.FromPythonInPlace` method for `iDynTree::VectorDynSize` (better name suggestions are welcome) that populate the existing vector instead of returning a new one (a bit like `.fromMatlab` do), and then use this new method in `JointPosDoubleArray`.

traversaro, 2025-04-04:
@Gianlucamilani today was also confused by this.

isorrentino, 2025-04-04:
I have opened a PR https://github.com/robotology/idyntree/pull/1234 implementing the first solution proposed by @traversaro (https://github.com/robotology/idyntree/issues/999#issuecomment-1153615036).

closed 2025-04-08 by traversaro