gbionics/idyntree #1009: Unable to open idyntree meshcat visualizer

GiulioRomualdi, 2022-07-29:
We installed the superbuild by enabling the python profile on a pc of the iCub3 setup. when I try to open the  the `MeshcatVisualizer` the web server seems correctly running but when I try to access to the URL `http://127.0.0.1:7000/static/` I got `404: Not Found`

To replicate the problem open a python3 interpreter and type
```python
from idyntree.visualize import MeshcatVisualizer                        viz = MeshcatVisualizer()                    
```

GiulioRomualdi, 2022-07-29:
I noticed that the problem can be replicated directly using meshcat. Indeed running the following command in a python shell
```python
import meshcat
viz = meshcat.Visualizer()
```
I got 
```
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7000/static/
```
but 

[image]

I also noticed that 127.0.0.1 points to `nginx webserver`

[image]

I think this is required to use [`jitsi meet`](https://meet.jit.si/)

traversaro, 2022-07-29:
Can you check which process are listening on which ports? I guess `netstat -ltnp` should give you this info.

GiulioRomualdi, 2022-07-29:
So I run the following command in a python shell
```python
import meshcat
viz = meshcat.Visualizer()
```
I got 
```
You can open the visualizer by visiting the following URL:
http://127.0.0.1:7000/static/
```

and `netstat -ltnp` gave me the following 
```
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.1:6000          0.0.0.0:*               LISTEN      12525/python3       
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      -                   
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:7000            0.0.0.0:*               LISTEN      12525/python3       
tcp        0      0 0.0.0.0:5269            0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:5280            0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:5222            0.0.0.0:*               LISTEN      -                   
tcp6       0      0 127.0.0.1:8080          :::*                    LISTEN      -                   
tcp6       0      0 :::8888                 :::*                    LISTEN      -                   
tcp6       0      0 :::9090                 :::*                    LISTEN      -                   
tcp6       0      0 ::1:631                 :::*                    LISTEN      -                   
tcp6       0      0 :::80                   :::*                    LISTEN      -                   
tcp6       0      0 :::22                   :::*                    LISTEN      -                   
tcp6       0      0 :::443                  :::*                    LISTEN      -                   
tcp6       0      0 :::7000                 :::*                    LISTEN      12525/python3       
tcp6       0      0 :::5269                 :::*                    LISTEN      -                   
tcp6       0      0 :::5280                 :::*                    LISTEN      -                   
tcp6       0      0 :::5222                 :::*                    LISTEN      -       
```

traversaro, 2022-07-29:
> (Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)

Given this, it may be useful to run `sudo netstat -ltnp`.

GiulioRomualdi, 2022-07-29:
```
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.1:6000          0.0.0.0:*               LISTEN      15469/python3       
tcp        0      0 127.0.0.1:631           0.0.0.0:*               LISTEN      1188/cupsd          
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      951/systemd-resolve 
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      1172/nginx: master  
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1145/sshd: /usr/sbi 
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      1172/nginx: master  
tcp        0      0 0.0.0.0:7000            0.0.0.0:*               LISTEN      15469/python3       
tcp        0      0 0.0.0.0:5269            0.0.0.0:*               LISTEN      1202/lua5.2         
tcp        0      0 0.0.0.0:5280            0.0.0.0:*               LISTEN      1202/lua5.2         
tcp        0      0 0.0.0.0:5222            0.0.0.0:*               LISTEN      1202/lua5.2         
tcp6       0      0 127.0.0.1:8080          :::*                    LISTEN      1607/java           
tcp6       0      0 :::8888                 :::*                    LISTEN      1353/java           
tcp6       0      0 :::9090                 :::*                    LISTEN      1607/java           
tcp6       0      0 ::1:631                 :::*                    LISTEN      1188/cupsd          
tcp6       0      0 :::80                   :::*                    LISTEN      1172/nginx: master  
tcp6       0      0 :::22                   :::*                    LISTEN      1145/sshd: /usr/sbi 
tcp6       0      0 :::443                  :::*                    LISTEN      1172/nginx: master  
tcp6       0      0 :::7000                 :::*                    LISTEN      15469/python3       
tcp6       0      0 :::5269                 :::*                    LISTEN      1202/lua5.2         
tcp6       0      0 :::5280                 :::*                    LISTEN      1202/lua5.2         
tcp6       0      0 :::5222                 :::*                    LISTEN      1202/lua5.2     
```

I'm afraid that the connection we are looking for is this one 
```
tcp        0      0 0.0.0.0:7000            0.0.0.0:*               LISTEN      15469/python3       
```

S-Dafarra, 2022-07-29:
If it is the ``icub-console-cuda`` machine, you can try to uninstall ``jitsi`` with 
```
sudo apt remove --purge jitsi-meet jitsi-meet-prosody jitsi-meet-web-config jitsi-meet-web jicofo jitsi-videobridge jitsi-videobridge2  nginx nginx-common nginx-full prosody
```

GiulioRomualdi, 2022-08-24:
@S-Dafarra and I run `git submodule update --init --recursive` in the `meshcat-python` repo cloned by the superbuild following what is written in https://github.com/rdeits/meshcat-python/issues/30#issuecomment-1040079875 after this the 404 error disappeared

GiulioRomualdi, 2022-08-24:
@traversaro, I think this should be automatically handled by the superbuild

traversaro, 2022-08-24:
> @traversaro, I think this should be automatically handled by the superbuild

I just tested on a clean superbuild in Ubuntu 22.04 with apt deps, and apparently the submodule is correctly initialized:
~~~
straversaro@IITICUBLAP257:~/test/robotology-superbuild$ ls ./src/meshcat-python/src/meshcat/viewer/
data  LICENSE       Readme.md  test               yarn.lock
dist  package.json  src        webpack.config.js
~~~

Are you sure that nothing strange happened to the `robotology-superbuild/src/meshcat-python` directory? On which OS and which CMake version are you using?

traversaro, 2023-10-16:
@GiulioRomualdi @S-Dafarra did you experienced again this problem? Can we close?

S-Dafarra, 2023-10-16:
> @GiulioRomualdi @S-Dafarra did you experienced again this problem? Can we close?

Not on my side

GiulioRomualdi, 2023-10-16:
me neither

closed 2023-10-16 by traversaro