gbionics/idyntree #1028: Singular Inertia Matrix?

Woolfrey, 2022-09-06:
I encountered an unusual problem where for some configuration(s) of iCub3 the inertia matrix is singular.

For context, as part of a Cartesian impedance controller, I solve the following feedforward + feedback control:
[image]
(I use this equation to avoid inverting the Jacobian in case of singularity).

In code I use Cholesky decomposition in Eigen to invert the inertia matrix:
```
xddot += J*M.llt().solve(J.transpose()*feedback);
```
I was moving around the robot randomly to rest the robustness of a joint limit avoidance method, and sometimes the Cholesky decomposition would fail:
[image]
The determinant is -2.76e-22.

I know that with many DoF the inertia matrix can be ill-conditioned, but it should still be invertible in theory. All I can think of is that the condition number is so small that the determinant cannot be computed accurately.

I have switched to using LU decomposition for now since it is robust to singularities, but Cholesky is slightly faster.

DanielePucci, 2022-09-06:
Hi @Woolfrey, the result you are getting seems in accordance to previously documented similar issues https://github.com/robotology-playground/WBI-Toolbox/issues/47

Woolfrey, 2022-09-06:
Hi @DanielePucci, thanks for the information!

 I will continue using LU decomposition then since it has "fixed" the problem for me.

closed 2022-09-06 by Woolfrey