gbionics/idyntree #1034: iDynTree Usage in MATLAB

yuying334, 2022-11-03:
Hi, I am confused about the description " If this is not working, make sure that you are launching matlab after having activated the conda environment (if you installed iDynTree via conda)". which environment do you mean here? When I use IDynTree in Matlab by just add the folder path "D:\conda\conda\Library\mex" and it has problem with running "iDynTreeMEX".  The error message is "Invalid MEX-file mexfilename: The specified module could not be found." Could you help me or give a more detailed guidance for matalab usage. I am really appreciate for that.

traversaro, 2022-11-05:
Hello @yuying334 !

That phrase means that everything works if you launch matlab from the Command Prompt in which you run `conda activate <nameofenv>`. If you prefer not to launch matlab from the Command Prompt, an alternative is to add to the path the necessary libraries directly via MATLAB, as we do for example in https://github.com/robotology/robotology-superbuild/blob/master/doc/matlab-one-line-install.md :

~~~
setenv("PATH",strcat("D:\conda\conda\Library\bin", ";", getenv("PATH")));
add_path("D:\conda\conda\Library\mex")
~~~

yuying334, 2022-11-06:
> Hello @yuying334 !
>
> That phrase means that everything works if you launch matlab from the Command Prompt in which you run `conda activate <nameofenv>`. If you prefer not to launch matlab from the [...]

Hello, 
Thanks a lot! It really helps me a lot.

traversaro, 2022-11-06:
I see, probably we could add this to the docs, even if I am afraid it would become too complex.

traversaro, 2022-12-13:
Hello @yuying334, do you think we can close this issue?

yuying334, 2022-12-13:
Hello, yes of course, thank you

closed 2022-12-13 by traversaro