gbionics/idyntree #1035: Support `.obj` loading in meshcat iDynTree Visualizer

antonellopaolino, 2022-11-23:
When I try to import the iRonCub model in the `robot-log-visualizer` the robot is loaded showing the colors of the `model.urdf` file which are randomly assigned during the urdf generation:

[image]

while the real colors of the robot are already in the `.obj` files and are correctly loaded when using `idyntree-model-view`:

[image]

Is there a possibility to load the `obj` meshes also in the meshcat visualizer?

GiulioRomualdi, 2022-11-23:
Good question! The color that you see are taken from idyntree that reads it from the urdf.

antonellopaolino, 2022-11-24:
Yes, I was thinking the same!

traversaro, 2022-11-24:
The `<material>` tag in the URDF (see http://wiki.ros.org/urdf/XML/link) is meant to be used only for meshes format that do not carry color information on their own (as the `.stl` format used originally in iCub models). Once you switch to use manually edited `.obj` meshes, could it make sense to also remove the `<color>` tag of the visual element. For example, I think your model has a link decribed by:
~~~  
<link name="l_upper_arm">
    <inertial>
      <origin xyz="-0.0031401005252317538 -0.0005800001394788284 0.03001409994235449" rpy="0 -1.3089969894682865 -3.141592653589722"/>
      <mass value="0.713864"/>
      <inertia ixx="0.000455006" ixy="-9.1018e-07" ixz="-3.2008e-05" iyy="0.000615652" iyz="1.30798e-05" izz="0.000613922"/>
    </inertial>
    <visual>
      <origin xyz="-0.014921196617346573 -1.2999101713595529e-05 -0.19330000026117397" rpy="0 -1.3089969894682865 -3.141592653589722"/>
      <geometry>
        <mesh filename="package://iRonCub-Mk1_1/iRonCub/meshes/obj/sim_sea_2-5_l_elbow_prt.obj" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="green">
        <color rgba="0 1 0 1"/>
      </material>
    </visual>
    <collision>
      <origin xyz="-0.014921196617346573 -1.2999101713595529e-05 -0.19330000026117397" rpy="0 -1.3089969894682865 -3.141592653589722"/>
      <geometry>
        <mesh filename="package://iRonCub-Mk1_1/iRonCub/meshes/obj/sim_sea_2-5_l_elbow_prt.obj" scale="0.001 0.001 0.001"/>
      </geometry>
    </collision>
  </link>
~~~

Probably we can just remove the:
~~~
      <material name="green">
        <color rgba="0 1 0 1"/>
      </material>
~~~
from the visual tag?

An alternative option is for iDynTree to ignore the `material` tag on meshes loaded from format that support colors, but it is not trivial to specify "format that support colors" as unfortunatly the URDF format literally accepts "Any geometry format is acceptable".

antonellopaolino, 2022-11-24:
> Probably we can just remove the:
> 
> ```
>       <material name="green">
>         <color rgba="0 1 0 1"/>
>       </material>
> ```
> 
> from the visual tag?

With @Giulero we already tried to remove that info from the urdf but then the robot link was not visible in the idyntree visualizer with meshcat.

antonellopaolino, 2022-11-24:
Yes

traversaro, 2022-11-24:
I suspect that the bug is in the robot-log-visualizer then. In particular, in https://github.com/ami-iit/robot-log-visualizer/blob/82aa5ba1c92599672365b5393d520a049e28a3ac/robot_log_visualizer/robot_visualizer/meshcat_visualizer.py#L226 the code should call `solid_shape.isMaterialSet()`  (see https://robotology.github.io/idyntree/classiDynTree_1_1SolidShape.html#a79698fa4e4ec3278dcb1f22dfaee0c04) before calling `solid_shape.getMaterial()` . If  `solid_shape.isMaterialSet()` is false, then `None` should be passed to second argument of set_object in line https://github.com/ami-iit/robot-log-visualizer/blob/82aa5ba1c92599672365b5393d520a049e28a3ac/robot_log_visualizer/robot_visualizer/meshcat_visualizer.py#L243 .