gbionics/idyntree #1043: Wrong constructor called when a MatrixView<const T> is built from MatrixView<T>

GiulioRomualdi, 2023-01-09:
```cpp
#include <iDynTree/Core/MatrixView.h>
#include <array>
#include <iostream>

iDynTree::MatrixView<double> array_to_matrix_view(std::array<double, 16>& array)
{
    constexpr iDynTree::MatrixView<double>::index_type rows = 4;
    constexpr iDynTree::MatrixView<double>::index_type cols = 4;
    return iDynTree::make_matrix_view(array.data(),
                                      rows,
                                      cols,
                                      iDynTree::ColumnMajor);
}

int main(int argc, char *argv[])
{
    std::array<double, 16> array = {1, 0, 0, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 0, 1};
    iDynTree::MatrixView<double> tmp1 = array_to_matrix_view(array);
    std::cerr << "tmp1: " << (tmp1.storageOrder() == iDynTree::ColumnMajor ? "column major" : "row major") << std::endl;

    iDynTree::MatrixView<const double> tmp2(tmp1);
    std::cerr << "tmp2: " << (tmp2.storageOrder() == iDynTree::ColumnMajor ? "column major" : "row major") << std::endl;

    iDynTree::MatrixView<double> tmp3(tmp1);
    std::cerr << "tmp3: " << (tmp3.storageOrder() == iDynTree::ColumnMajor ? "column major" : "row major") << std::endl;

    return 0;
}
```

I got
```
tmp1: column major
tmp2: row major
tmp3: column major
```

that is not what I'm expecting since tmp2 should share the same `storage order` as `tmp1`.

I dug into the code and I noticed that this happens because the following constructor is called 

https://github.com/robotology/idyntree/blob/a7fdef001d38e47aabcdc541607a255101212dd9/src/core/include/iDynTree/Core/MatrixView.h#L131-L142

while it should call

https://github.com/robotology/idyntree/blob/a7fdef001d38e47aabcdc541607a255101212dd9/src/core/include/iDynTree/Core/MatrixView.h#L105-L112

@traversaro

traversaro, 2023-01-09:
Given that this is a C++ issue, which compiler (and version of it) are you using?

GiulioRomualdi, 2023-01-09:
> Given that this is a C++ issue, which compiler (and version of it) are you using?

yes sorry. I'm on ubuntu 20.04
`gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1)`

traversaro, 2023-01-09:
It is not clear to me why the constructor selected should be the one at line 105 instead of the one at line 137 .

traversaro, 2023-01-09:
There are some failures in CI, but they are unrelated and related to https://github.com/actions/runner-images/issues/675 instead.

PR #1044 (2023-01-10): Fix MatrixView constructor when MatrixView<const T> is built from MatrixView<T>
This fixes #1043
closed 2023-01-10 by traversaro