gbionics/idyntree #1052: Error - SDL_SetWindowResizable missing in SDL2 library

G-Cervettini, 2023-02-22:
I'm trying to run these MATLAB lines of code on Windows:

```matlab
robotName='iCubGenova09'; %% Name of the robot

meshFilePrefix = [getenv('ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX') '/share']; %% Path to the model meshes
modelPath = [getenv('ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX') '/share/iCub/robots/' robotName '/'];  %% Path to the robot model

fileName='model.urdf'; %% Name of the urdf file
gravityAcceleration = 9.80665;
% 
world_H_base = eye(4);
KinDynModel = iDynTreeWrappers.loadReducedModel(robot_logger_device.description_list, 'root_link', modelPath, fileName, false);
```

and i'm getting the following error:

[image]

This is the Command Window output:
```
[loadReducedModel]: loading the following model: C:\Users\gcervettini\robotology-superbuild\build\install/share/iCub/robots/iCubGenova09/model.urdf
Invalid MEX-file 'C:\Users\gcervettini\robotology-superbuild\build\install\mex\iDynTreeMEX.mexw64': A dynamic
link library (DLL) initialization routine failed.

Error in iDynTree.StringVector (line 57)
        tmp = iDynTreeMEX(36, varargin{:});

Error in iDynTreeWrappers.loadReducedModel (line 37)
    jointList_idyntree     = iDynTree.StringVector();
```

The dependencies have been installed with conda.
The robotology-superbuild has been built again today.

this is my conda environment:

```
(robsub) C:\Users\gcervettini>mamba list
packages in environment at C:\Users\gcervettini\AppData\Local\mambaforge\envs\robsub:

Name                    Version                   Build  Channel
ace                       7.0.8                h0e60522_0    conda-forge
asio                      1.22.1               h0e60522_0    conda-forge
assimp                    5.2.4                hc2aa0de_0    conda-forge
blosc                     1.21.1               h74325e0_3    conda-forge
boost                     1.74.0          py310hc781a3c_5    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.4.2                h8ffe710_0    conda-forge
ca-certificates           2022.6.15            h5b45459_0    conda-forge
cairo                     1.16.0            h0ac17fb_1011    conda-forge
cfitsio                   4.1.0                h5a969a9_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.23.2               h39d44d4_0    conda-forge
compilers                 1.4.2                h57928b3_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppzmq                    4.8.1                h4324990_0    conda-forge
curl                      7.83.1               h789b8ee_0    conda-forge
cxx-compiler              1.4.2                h2d74725_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
[... 152 lines cut ...]
vc                        14.2                 hb210afc_6    conda-forge
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
vs2017_win-64             19.16.27033          hb90652a_6    conda-forge
vswhere                   3.0.3                h57928b3_0    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
xerces-c                  3.2.3                h0e60522_5    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_1    conda-forge
xorg-libxpm               3.5.13               hcd874cb_0    conda-forge
xorg-libxt                1.2.1                hcd874cb_2    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1002    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.12               h8ffe710_1    conda-forge
zstd                      1.5.2                h6255e5f_2    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```
sdl2 is present. What could be the problem?

traversaro, 2023-02-22:
See https://github.com/robotology/idyntree/issues/985 for a debug of the issue and a workaround. Can you provide the output of ver command in your system?

G-Cervettini, 2023-02-23:
@traversaro here the output:

```
>> ver
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.12.0.1975300 (R2022a) Update 3
MATLAB License Number: 40875934
Operating System: Microsoft Windows 10 Enterprise Version 10.0 (Build 19042)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.12        (R2022a)
Simulink                                              Version 10.5        (R2022a)
5G Toolbox                                            Version 2.4         (R2022a)
AUTOSAR Blockset                                      Version 2.6         (R2022a)
Aerospace Blockset                                    Version 5.2         (R2022a)
Aerospace Toolbox                                     Version 4.2         (R2022a)
Antenna Toolbox                                       Version 5.2         (R2022a)
Audio Toolbox                                         Version 3.2         (R2022a)
Automated Driving Toolbox                             Version 3.5         (R2022a)
Bioinformatics Toolbox                                Version 4.16        (R2022a)
Bluetooth Toolbox                                     Version 1.0         (R2022a)
Communications Toolbox                                Version 7.7         (R2022a)
Computer Vision Toolbox                               Version 10.2        (R2022a)
Control System Toolbox                                Version 10.11.1     (R2022a)
Curve Fitting Toolbox                                 Version 3.7         (R2022a)
DDS Blockset                                          Version 1.2         (R2022a)
DSP HDL Toolbox                                       Version 1.0         (R2022a)
DSP System Toolbox                                    Version 9.14        (R2022a)
Data Acquisition Toolbox                              Version 4.5         (R2022a)
Database Toolbox                                      Version 10.3        (R2022a)
Datafeed Toolbox                                      Version 6.2         (R2022a)
Deep Learning HDL Toolbox                             Version 1.3         (R2022a)
Deep Learning Toolbox                                 Version 14.4        (R2022a)
Econometrics Toolbox                                  Version 6.0         (R2022a)
Embedded Coder                                        Version 7.8         (R2022a)
Filter Design HDL Coder                               Version 3.1.11      (R2022a)
Financial Instruments Toolbox                         Version 3.4         (R2022a)
Financial Toolbox                                     Version 6.3         (R2022a)
Fixed-Point Designer                                  Version 7.4         (R2022a)
Fuzzy Logic Toolbox                                   Version 2.9         (R2022a)
GPU Coder                                             Version 2.3         (R2022a)
Global Optimization Toolbox                           Version 4.7         (R2022a)
HDL Coder                                             Version 3.20        (R2022a)
HDL Verifier                                          Version 6.5         (R2022a)
Image Acquisition Toolbox                             Version 6.6         (R2022a)
Image Processing Toolbox                              Version 11.5        (R2022a)
Industrial Communication Toolbox                      Version 6.0         (R2022a)
Instrument Control Toolbox                            Version 4.6         (R2022a)
LTE Toolbox                                           Version 3.7         (R2022a)
Lidar Toolbox                                         Version 2.1         (R2022a)
MATLAB Coder                                          Version 5.4         (R2022a)
MATLAB Compiler                                       Version 8.4         (R2022a)
MATLAB Compiler SDK                                   Version 7.0         (R2022a)
MATLAB Report Generator                               Version 5.12        (R2022a)
Mapping Toolbox                                       Version 5.3         (R2022a)
Mixed-Signal Blockset                                 Version 2.2         (R2022a)
Model Predictive Control Toolbox                      Version 7.3         (R2022a)
Model-Based Calibration Toolbox                       Version 5.12        (R2022a)
Motor Control Blockset                                Version 1.4         (R2022a)
Navigation Toolbox                                    Version 2.2         (R2022a)
Optimization Toolbox                                  Version 9.3         (R2022a)
Parallel Computing Toolbox                            Version 7.6         (R2022a)
Partial Differential Equation Toolbox                 Version 3.8         (R2022a)
Phased Array System Toolbox                           Version 4.7         (R2022a)
Powertrain Blockset                                   Version 1.11        (R2022a)
Predictive Maintenance Toolbox                        Version 2.5         (R2022a)
RF Blockset                                           Version 8.3         (R2022a)
RF PCB Toolbox                                        Version 1.1         (R2022a)
RF Toolbox                                            Version 4.3         (R2022a)
ROS Toolbox                                           Version 1.5         (R2022a)
[... cut ...]
Vehicle Network Toolbox                               Version 5.2         (R2022a)
Vision HDL Toolbox                                    Version 2.5         (R2022a)
WLAN Toolbox                                          Version 3.4         (R2022a)
Wavelet Toolbox                                       Version 6.1         (R2022a)
Wireless HDL Toolbox                                  Version 2.4         (R2022a)
Wireless Testbench                                    Version 1.0         (R2022a)
```

G-Cervettini, 2023-03-07:
> See #985 for a debug of the issue and a workaround.
This solved the issue, thank you.
Closing as completed.

closed 2023-03-07 by G-Cervettini